arendst/Tasmota #6845: Gosund SP111 15A version - no power measuring

FUSiiiiON, 2019-11-05:
It seems, that the Gosund SP111 got a new Revision. I got recently the 15A version (3450W). I already have the Gosund SP111 10A version, where everything is working fine with the BlitzWolf SHP (45) template.

The BlitzWolf template does not work completely with the 15A version, because it does only show the Energy Today / Energy Yesterday and Energy Total. Voltage, Current, Power and all the other power measuring values are 0:

On the left is the 15A version .. on the right the 10A version of the SP111
[image]

Does anybody has already got the 15A version and now the correct GPIO values to get the missing power informations?

cavac, 2019-11-05:
I just got mine an hour ago. Where did you find IO0 and GND? Or did you flash OTA?

FUSiiiiON, 2019-11-05:
I flashed the device via OTA

cavac, 2019-11-05:
Tuya convert?

FUSiiiiON, 2019-11-05:
Yes

FUSiiiiON, 2019-11-05:
BTW: I just recognized, that this topic would be better posted in the forum instead as an issue for Tasmota itself. I hope it is not a problem. Next time i open up a topic in the forum.

Jason2866, 2019-11-05:
@FUSiiiiON  Exactly😀 please close
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-05 by FUSiiiiON

Viech83, 2019-11-07:
I had the same problem.
Insert this as Template
{"NAME":"Gosund SP111 V","GPIO":[57,255,56,255,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45}

komw, 2020-01-28:
> I had the same problem.
> Insert this as Template
> {"NAME":"Gosund SP111 V","GPIO":[57,255,56,255,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45}

Are you sure that is working for SP111? I still have: 
"
Voltage | 0 V
-- | --
0.000 A
0 W
0 VA
0 VAr
0.00
0.033 kWh
0.000 kWh
0.033 kWh
"

taste66, 2020-01-28:
I use this for the BW-SHP6 3450W which is internally the same
Red LED when on
`{"NAME":"BW-SP6 15A","GPIO":[56,255,158,255,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45}`
Blue LED when on
`{"NAME":"BW-SP6 15A","GPIO":[158,255,56,255,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45}`

Viech83, 2020-01-28:
Hi !yes, it is working well at all my 8 SP111.Can you send me a screenshot from your template (how you Insert this in the device?)Kind Regards,Philipp LenzVon meinem Samsung Galaxy Smartphone gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/arendst/Tasmota/issues/6845?email_source=notifications\u0026email_token=ANWWYEYSTDRD54CEPEG2AOLRACKXDA5CNFSM4JJIISJ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKE24NY#issuecomment-579448375",
"url": "https://github.com/arendst/Tasmota/issues/6845?email_source=notifications\u0026email_token=ANWWYEYSTDRD54CEPEG2AOLRACKXDA5CNFSM4JJIISJ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEKE24NY#issuecomment-579448375",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Viech83, 2020-01-28:
Just copied from one of my device:{"NAME":"Gosund SP111 V","GPIO":[57,255,56,255,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45}Von meinem Samsung Galaxy Smartphone gesendet.

komw, 2020-01-28:
Ah,, I forgot to tick "activate"... but still I need to calibrate it, because it shows me a 280V (instead od 240V)

Viech83, 2020-01-29:
Yes, calibration is needed.Best you first calibrate the voltage and then the power.    Von meinem Samsung Galaxy Smartphone gesendet.