arendst/Tasmota #6846: RF Bridge will send full codes when first digit of code is not a zero

peteshoard, 2019-11-05:
Using both the MQTT command and the RFCode command in the console, i am able to send codes where the first digit is **not** zero successfully - for example "87F282" but if the first digit of the code is a zero - for example "07F282" the console shows the code sent is "7F282"

if anyone can help identify the cause of this - it would be most appreciated.

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used ( Sonoff RF Bridge)
- [x ] Tasmota binary firmware version number used: 6.6.0(release-sonoff)

s-hadinger, 2019-11-05:
I looked at the code, and it is just a logging artifact. The code sent is always 24 bits. What you see in the console is the 24 bits without the leading zeroes.

To my understanding:`RFCode #07F282` and `RFCode #7F282` are sending exactly the same commands.

If it makes more sense to always report the 24 bits, it would just need to replace:
`snprintf_P(stemp, sizeof(stemp), PSTR("\"#%X\""), code);`
with
`snprintf_P(stemp, sizeof(stemp), PSTR("\"#%06X\""), code);`

peteshoard, 2019-11-05:
Thanks for the fast reply - the trouble i have is that all of the devices i have that have the zero as the first digits are passive door sensors, i therefore am struggling to verify that the codes are received in the same way. I have some plug switches i can test on, but need to wait until the house is empty to stop them being coded into the motion sensors.

s-hadinger, 2019-11-06:
Ok. I will make the change in the code to report all 24 bits every time.

PR #6855 (2019-11-06): Display 6 digits hex for RFCode #6846
## Description:

Display all 24 bits (6 hex digits) code with `RFCode`

**Related issue (if applicable):** fixes #6846 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core pre-2.6
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-c [... cut]
closed 2019-11-06 by arendst