arendst/Tasmota #6848: Highlight the major positions on the clock

robin13, 2019-11-06:
**Is your feature request related to a problem? Please describe.**  

Making a minimalist clock with only WS2812 LEDs but no external markings, it can be difficult to identify where 12, 3, 6, 9 are to get a reference for what the actual time is.

Image of my clock for illustration:
[image]

**Describe the solution you'd like**  
The ability to highlight (defining colour, brightness, and the interval (each hour, only the quarters, or only 12) the hour markers of the clock

**Describe alternatives you've considered**  
Painting numbers onto the clock... but this would take away from the minimalist design

arendst, 2019-11-06:
Try command ``color6 808000``

robin13, 2019-11-06:
Damn - I missed that in the documentation!  Thank you! :heart:

closed 2019-11-06 by robin13