arendst/Tasmota #6850: Can't get rule to set color of bulb on boot

g-off-white, 2019-11-06:
> **GUIDE**
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>
> Please DO NOT OPEN AN ISSUE:
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
I am trying to get a bulb to power on with a default color.  At present it will boot up and assume the last state.  I have tried using rules on switch1#boot and system#boot to change the color but seems to have no effect.  Perhaps the color command is being overwritten by the remembered color setting?
I never even see the rule firing in the console.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): rgbw bulb using pwm_____
- [ x] Tasmota binary firmware version number used: __6.7.1___
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: _tuya_convert____
- [ x] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
14:20:31 MQT: stat/B_Lounge1/RESULT = {"NAME":"Cocoon RGBW","GPIO":[0,0,0,0,37,40,0,0,38,0,39,0,0],"FLAG":0,"BASE":18}
14:20:31 MQT: stat/B_Lounge1/RESULT = {"Module":{"0":"Cocoon RGBW"}}
14:20:32 MQT: stat/B_Lounge1/RESULT = {"GPIO":"Not supported"}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
14:21:29 MQT: stat/B_Lounge2/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":472,"Rules":"on system#boot do color #000000ff enddo"}

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
14:22:31 MQT: stat/B_Lounge1/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"B_Lounge1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
14:22:31 MQT: stat/B_Lounge1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:05:22","StartupUTC":"2019-11-06T13:17:09","Sleep":50,"CfgHolder":4617,"BootCount":29,"SaveCount":92,"SaveAddress":"F4000"}}
14:22:31 MQT: stat/B_Lounge1/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
14:22:31 MQT: stat/B_Lounge1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["fibreking","WhatWouldCharlieDo?"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005AFF000000000000","00000000"]}}
14:22:31 MQT: stat/B_Lounge1/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
14:22:31 MQT: stat/B_Lounge1/STATUS5 = {"StatusNET":{"Hostname":"B_Lounge1-2228","IPAddress":"10.1.1.145","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"EC:FA:BC:B0:08:B4","Webserver":2,"WifiConfig":2}}
14:22:31 MQT: stat/B_Lounge1/STATUS6 = {"StatusMQT":{"MqttHost":"eastst.synology.me","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B008B4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
14:22:31 MQT: stat/B_Lounge1/STATUS7 = {"StatusTIM":{"UTC":"Wed Nov 06 13:22:31 2019","Local":"Wed Nov 06 14:22:31 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:44","Sunset":"17:22"}}
14:22:31 MQT: stat/B_Lounge1/STATUS10 = {"StatusSNS":{"Time":"2019-11-06T14:22:31"}}
14:22:31 MQT: stat/B_Lounge1/STATUS11 = {"StatusSTS":{"Time":"2019-11-06T14:22:31","Uptime":"0T00:05:22","UptimeSec":322,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":631,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","Channel":[0,0,0,100],"Scheme":0,"Fade":"OFF","Speed":2,"LedTable":"ON","Wifi":{"AP":2,"SSId":"WhatWouldCharlieDo?","BSSId":"70:8B:CD:C2:31:50","Channel":8,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:09"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-11-06:
In the last part of the rule, it says ENDDO. It should say ENDON

Please, change your rule from

```
Rule1 on system#boot do color #000000ff enddo
```

To

```
Rule1 on system#boot do color #000000ff endon
```

And try again.

ascillato, 2019-11-06:
Another question, if you set the white value of your bulb, that value is set to 0 when restarting but the rest of the RGB is retained?

g-off-white, 2019-11-06:
I have tried it both ways.  It never even shows up in the console after booting

Yes to the RGB.  Only the RGB is retained.  Using white only also is retained.

ascillato, 2019-11-06:
Please, can you post the output of the console when booting? Please, set weblog to 4.

s-hadinger, 2019-11-06:
I tried to reproduce but it runs fine. However `System#Boot` doesn't happen immediately: it is triggered once MQTT is initialized, it happened 5 seconds after boot when I tried.

You can try `Power1#Boot` to run the command right at start.

`Rule1 on power1#boot do color #000000ff endon`

g-off-white, 2019-11-06:
I was testing with 2 bulbs.  Same make and fw.  One works the other does not.
There may have been a typo with the trigger event but even cut and paste does not work on this one:

The other fires the rule after the project Sonoff line.

I suspect it is time to reset the configuration and start again.

00:00:00 CFG: Loaded from flash at F6, Count 114
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.7.1(sonoff)-STAGE
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 WhatWouldCharlieDo? in mode 11N as B_Lounge1-2228...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on B_Lounge1-2228 with IP address 10.1.1.145
21:42:00 NTP: Drift 0, (UTC) Wed Nov 06 20:42:00 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
21:42:01 MQT: Attempting connection...
21:42:01 MQT: Connected
21:42:01 MQT: tele/B_Lounge1/LWT = Online (retained)
21:42:01 MQT: cmnd/B_Lounge1/POWER = 
21:42:01 MQT: Subscribe to cmnd/B_Lounge1/#
21:42:01 MQT: Subscribe to cmnd/sonoffs/#
21:42:01 MQT: Subscribe to cmnd/DVES_B008B4_fb/#
21:42:01 MQT: tele/B_Lounge1/INFO1 = {"Module":"Cocoon RGBW","Version":"6.7.1(sonoff)","FallbackTopic":"cmnd/DVES_B008B4_fb/","GroupTopic":"sonoffs"}
21:42:01 MQT: tele/B_Lounge1/INFO2 = {"WebServerMode":"Admin","Hostname":"B_Lounge1-2228","IPAddress":"10.1.1.145"}
21:42:01 MQT: tele/B_Lounge1/INFO3 = {"RestartReason":"Power on"}
21:42:01 MQT: stat/B_Lounge1/RESULT = {"POWER":"ON"}
21:42:01 MQT: stat/B_Lounge1/POWER = ON (retained)
21:42:01 QPC: Reset
21:42:03 APP: Boot Count 45
21:42:04 CFG: Saved to flash at F5, Count 115, Bytes 4096
21:42:09 MQT: tele/B_Lounge1/STATE = {"Time":"2019-11-06T21:42:09","Uptime":"0T00:00:16","UptimeSec":16,"Heap":28,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":88,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"FF000000","HSBColor":"0,100,100","Channel":[100,0,0,0],"Scheme":0,"Fade":"OFF","Speed":2,"LedTable":"ON","Wifi":{"AP":2,"SSId":"WhatWouldCharlieDo?","BSSId":"70:8B:CD:C2:31:50","Channel":8,"RSSI":70,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:42:14 CMD: rule
21:42:14 SRC: WebConsole from 10.1.1.144
21:42:14 CMD: Group 0, Index 1, Command "RULE", Data ""
21:42:14 MQT: stat/B_Lounge1/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":472,"Rules":"on power1#boot do color #000000ff endon"}
21:42:18 CMD: status 0
21:42:18 SRC: WebConsole from 10.1.1.144
21:42:18 CMD: Group 0, Index 1, Command "STATUS", Data "0"
21:42:18 MQT: stat/B_Lounge1/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"B_Lounge1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
21:42:18 MQT: stat/B_Lounge1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:25","StartupUTC":"2019-11-06T20:41:53","Sleep":50,"CfgHolder":4617,"BootCount":45,"SaveCount":115,"SaveAddress":"F5000"}}
21:42:18 MQT: stat/B_Lounge1/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
21:42:18 MQT: stat/B_Lounge1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["fibreking","WhatWouldCharlieDo?"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005AFF000000000000","00000000"]}}
21:42:18 MQT: stat/B_Lounge1/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
21:42:18 MQT: stat/B_Lounge1/STATUS5 = {"StatusNET":{"Hostname":"B_Lounge1-2228","IPAddress":"10.1.1.145","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"EC:FA:BC:B0:08:B4","Webserver":2,"WifiConfig":2}}
21:42:18 MQT: stat/B_Lounge1/STATUS6 = {"StatusMQT":{"MqttHost":"eastst.synology.me","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B008B4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:42:18 MQT: stat/B_Lounge1/STATUS7 = {"StatusTIM":{"UTC":"Wed Nov 06 20:42:18 2019","Local":"Wed Nov 06 21:42:18 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:44","Sunset":"17:22"}}
21:42:18 MQT: stat/B_Lounge1/STATUS10 = {"StatusSNS":{"Time":"2019-11-06T21:42:18"}}
21:42:18 MQT: stat/B_Lounge1/STATUS11 = {"StatusSTS":{"Time":"2019-11-06T21:42:18","Uptime":"0T00:00:25","UptimeSec":25,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":115,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"FF000000","HSBColor":"0,100,100","Channel":[100,0,0,0],"Scheme":0,"Fade":"OFF","Speed":2,"LedTable":"ON","Wifi":{"AP":2,"SSId":"WhatWouldCharlieDo?","BSSId":"70:8B:CD:C2:31:50","Channel":8,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:05"}}}
21:42:19 WIF: Checking connection...
21:42:19 WIF: Connected

g-off-white, 2019-11-06:
Is have tried resetting the configuration and reflashing ota but the problem persists with this bulb.  Is there a procedure for nuking it without pulling it apart?

ascillato, 2019-11-06:
`reset 6`

g-off-white, 2019-11-06:
No change :(
This seems bizarre.

ascillato, 2019-11-06:
As you have two identical devices, and in one works, but in the other does not work, could you please share status 0 from both devices and also the output of the console at boot time? Thanks

g-off-white, 2019-11-07:
Here is a working one
This log is from a different network.

00:00:00 CFG: Loaded from flash at FA, Count 189
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.7.1(sonoff)-STAGE
00:00:00 RUL: POWER1#BOOT performs "backlog power on;color #000000ff"
00:00:00 SRC: Rule
00:00:00 CMD: Group 0, Index 1, Command "BACKLOG", Data "power on;color #000000ff"
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 fibreking in mode 11N as B_Lounge2-4582...
00:00:00 SRC: Backlog
00:00:00 CMD: Group 0, Index 1, Command "POWER", Data "on"
00:00:00 RSL: stat/B_Lounge2/RESULT = {"POWER":"ON"}
00:00:00 RSL: stat/B_Lounge2/POWER = ON
00:00:00 SRC: Backlog
00:00:00 CMD: Group 0, Index 1, Command "COLOR", Data "#000000ff"
00:00:00 RSL: stat/B_Lounge2/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","Channel":[0,0,0,100]}
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on B_Lounge2-4582 with IP address 192.168.9.226
00:00:08 APP: Boot Count 99
00:00:09 CFG: Saved to flash at F9, Count 190, Bytes 4096
00:00:10 MQT: Attempting connection...
00:00:14 MQT: Connected
00:00:14 MQT: tele/B_Lounge2/LWT = Online (retained)
00:00:14 MQT: cmnd/B_Lounge2/POWER = 
00:00:14 MQT: Subscribe to cmnd/B_Lounge2/#
00:00:14 MQT: Subscribe to cmnd/sonoffs/#
00:00:14 MQT: Subscribe to cmnd/DVES_1671E6_fb/#
00:00:14 MQT: tele/B_Lounge2/INFO1 = {"Module":"Cocoon RGBW","Version":"6.7.1(sonoff)","FallbackTopic":"cmnd/DVES_1671E6_fb/","GroupTopic":"sonoffs"}
00:00:14 MQT: tele/B_Lounge2/INFO2 = {"WebServerMode":"Admin","Hostname":"B_Lounge2-4582","IPAddress":"192.168.9.226"}
00:00:14 MQT: tele/B_Lounge2/INFO3 = {"RestartReason":"Power on"}
00:00:14 MQT: stat/B_Lounge2/RESULT = {"POWER":"ON"}
00:00:14 MQT: stat/B_Lounge2/POWER = ON (retained)
00:00:22 MQT: tele/B_Lounge2/STATE = {"Time":"1970-01-01T00:00:22","Uptime":"0T00:00:20","UptimeSec":20,"Heap":29,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":453,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","Channel":[0,0,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"fibreking","BSSId":"F0:9F:C2:2A:4B:1B","Channel":11,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:09"}}
00:56:41 NTP: Drift 0, (UTC) Wed Nov 06 23:56:41 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
00:56:43 WIF: Checking connection...
00:56:43 WIF: Connected
00:57:03 WIF: Checking connection...
00:57:03 WIF: Connected
00:57:23 WIF: Checking connection...
00:57:23 WIF: Connected
00:57:26 HTP: Main Menu
00:57:26 HTP: Main Menu
00:57:42 HTP: Console
00:57:43 WIF: Checking connection...
00:57:43 WIF: Connected
00:58:03 WIF: Checking connection...
00:58:03 WIF: Connected
00:58:07 CMD: status 0
00:58:07 SRC: WebConsole from 192.168.9.250
00:58:07 CMD: Group 0, Index 1, Command "STATUS", Data "0"
00:58:07 MQT: stat/B_Lounge2/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"B_Lounge2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
00:58:07 MQT: stat/B_Lounge2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:01:54","StartupUTC":"2019-11-06T23:56:13","Sleep":50,"CfgHolder":4617,"BootCount":99,"SaveCount":190,"SaveAddress":"F9000"}}
00:58:07 MQT: stat/B_Lounge2/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
00:58:07 MQT: stat/B_Lounge2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["fibreking","WhatWouldCharlieDo?"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005AFF000000000000","00000000"]}}
00:58:07 MQT: stat/B_Lounge2/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
00:58:07 MQT: stat/B_Lounge2/STATUS5 = {"StatusNET":{"Hostname":"B_Lounge2-4582","IPAddress":"192.168.9.226","Gateway":"192.168.9.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.9.1","Mac":"4C:11:AE:16:71:E6","Webserver":2,"WifiConfig":2}}
00:58:07 MQT: stat/B_Lounge2/STATUS6 = {"StatusMQT":{"MqttHost":"Eastst.synology.me","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1671E6","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
[... cut ...]
00:58:07 MQT: stat/B_Lounge2/STATUS10 = {"StatusSNS":{"Time":"2019-11-07T00:58:07"}}
00:58:07 MQT: stat/B_Lounge2/STATUS11 = {"StatusSTS":{"Time":"2019-11-07T00:58:07","Uptime":"0T00:01:54","UptimeSec":114,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":545,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","Channel":[0,0,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"fibreking","BSSId":"F0:9F:C2:2A:4B:1B","Channel":11,"RSSI":56,"LinkCount":1,"Downtime":"0T00:00:09"}}}

ascillato2, 2019-11-07:
Found the difference.

In the NOT working device, you have rule1 OFF

Please do **RULE1 1** in the console. Thanks.

closed 2019-11-07 by ascillato2