arendst/Tasmota #6852: v7.0.0.3 : pow r2 : no power values displayed

ncelectronico, 2019-11-06:
> **GUIDE**
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>
> Please DO NOT OPEN AN ISSUE:
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POW R2
- [x] Tasmota binary firmware version number used: 7.0.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: platformio visual studio
- [x] Flashing tools used: platformio visual studio
- [x] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
`05:59:15 SRC: WebConsole from 192.168.1.2
05:59:15 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO"
05:59:15 SRC: Backlog
05:59:15 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
05:59:15 MQT: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[17,145,0,146,0,0,0,0,21,56,0,0,0],"FLAG":0,"BASE":43}
05:59:15 SRC: Backlog
05:59:15 CMD: Group 0, Index 1, Command "MODULE", Data ""
05:59:15 MQT: stat/sonoff/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
05:59:15 SRC: Backlog
05:59:15 CMD: Group 0, Index 1, Command "GPIO", Data ""
05:59:15 MQT: stat/sonoff/RESULT = {"GPIO":"Not supported"}
05:59:17 WIF: Checking connection...
05:59:17 WIF: Connected`

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [x] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
`06:01:14 SRC: WebConsole from 192.168.1.2
06:01:14 CMD: Group 0, Index 1, Command "STATUS", Data "0"
06:01:14 MQT: stat/sonoff/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:01:14 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:17:29","StartupUTC":"2019-11-06T18:43:45","Sleep":50,"CfgHolder":6422,"BootCount":6,"SaveCount":21,"SaveAddress":"F7000"}}
06:01:14 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"7.0.0.3(tasmota)","BuildDateTime":"2019-11-06T19:54:12","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
06:01:14 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":4,"SysLog":4,"LogHost":"192.168.1.21","LogPort":514,"SSId":["electronico_EXT","electronico1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005A00000000000000","00000000","00000000"]}}
06:01:14 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":555,"Free":448,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8FDAE397","043683A1","22B617CD","01001BC0","00000081"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
06:01:14 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5318","IPAddress":"192.168.1.215","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:92:34:C6","Webserver":2,"WifiConfig":4}}
06:01:14 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.21","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9234C6","MqttUser":"dave","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
06:01:14 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Nov 06 19:01:14 2019","Local":"Thu Nov 07 06:01:14 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+11:00","Sunrise":"05:05","Sunset":"18:08"}}
06:01:14 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
06:01:14 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-11-07T06:01:14","ENERGY":{"TotalStartTime":"2019-11-06T19:55:47","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
06:01:14 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-11-07T06:01:14","Uptime":"0T00:17:29","UptimeSec":1049,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":2,"SSId":"electronico1","BSSId":"18:F1:45:52:F2:3D","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}}
06:01:18 WIF: Checking connection...
06:01:18 WIF: Connected
06:01:22 NTP: Drift 0, (UTC) Wed Nov 06 19:01:22 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
06:01:38 WIF: Checking connection...`

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-11-06:
Please try a precompiled firmware http://thehackbox.org/tasmota/tasmota.bin

ncelectronico, 2019-11-06:
How can I insert user_config_override with this precompiled firmware ?

meingraham, 2019-11-06:
You have to compile your own firmware binary for your `user_config_override.h` to take effect. See https://github.com/arendst/Tasmota/wiki/Compile-your-build

Jason2866, 2019-11-06:
I suggested to use a precompiled firmware to test if there is a fault in Tasmota.
If it works (i think so) then your selfcompiled firmware is faulty

ncelectronico, 2019-11-06:
Results are identical using the pre-compiled firmware (except a black background on device UI)

Jason2866, 2019-11-06:
Do a `reset 6` erases flash and settings. Only mqtt and wifi settings do survive.
Configure device manually. Do not use a backup

ncelectronico, 2019-11-06:
`reset 6` done
No backup used, device configured manually : 
Module 
[image]

Template
[image]

Still same result
[image]

ncelectronico, 2019-11-06:
I downloaded precompiled 6.7.1 (that is supposed to fix this issue) [https://github.com/arendst/Tasmota/releases/download/v6.7.1/sonoff.bin](url)
uploaded firmware

> ~/Sonoff-Tasmota_esptool/tasmota/6.7.1$ esptool --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
> esptool.py v2.1
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266
> Enabling default SPI flash mode...
> Configuring flash size...
> Erasing flash...
> Took 2.60s to erase flash block
> Wrote 577536 bytes at 0x00000000 in 56.3 seconds (82.1 kbit/s)...
> Leaving...
> Hard resetting...

Still same issue : 
[image]

The 4 POW R2 I have are all doing the same ...

blakadder, 2019-11-06:
Don't know if its related but I have a device that was confirmed working before the 6.7.1 update (its screenshot is on the front page of the wiki) and now its not showing power measurements but its measuring energy daily, yesterday and total. Its a template based on `module 45`. Relay and RGB operate like before.

Tried all the resets and different builds from 6.7 till 7.0.0.3 but nothing helped. No way to serial flash it since its a closed shell.
[image]

[image]

status 0
```
00:41:40 CMD: status 0
00:41:40 MQT: stat/rgb_pow/STATUS = {"Status":{"Module":0,"FriendlyName":["RGB Pow","Sonoff2"],"Topic":"rgb_pow","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:41:40 MQT: stat/rgb_pow/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Power on","Uptime":"0T02:43:14","StartupUTC":"2019-11-06T20:58:26","Sleep":50,"CfgHolder":4617,"BootCount":28,"SaveCount":107,"SaveAddress":"F9000"}}
00:41:40 MQT: stat/rgb_pow/STATUS2 = {"StatusFWR":{"Version":"7.0.0.3(tasmota)","BuildDateTime":"2019-11-05T17:29:55","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(5ab15d1)"}}
00:41:40 MQT: stat/rgb_pow/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HTPC","numero"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
00:41:40 MQT: stat/rgb_pow/STATUS4 = {"StatusMEM":{"ProgramSize":573,"Free":428,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,91","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
00:41:40 MQT: stat/rgb_pow/STATUS5 = {"StatusNET":{"Hostname":"rgb_pow-6614","IPAddress":"10.1.1.215","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"80:7D:3A:23:19:D6","Webserver":2,"WifiConfig":2}}
00:41:40 MQT: stat/rgb_pow/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2319D6","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:41:40 MQT: stat/rgb_pow/STATUS7 = {"StatusTIM":{"UTC":"Wed Nov 06 23:41:40 2019","Local":"Thu Nov 07 00:41:40 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:46","Sunset":"17:21"}}
00:41:40 MQT: stat/rgb_pow/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:41:40 MQT: stat/rgb_pow/STATUS10 = {"StatusSNS":{"Time":"2019-11-07T00:41:40","ENERGY":{"TotalStartTime":"2019-11-05T22:29:27","Total":0.073,"Yesterday":0.065,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
00:41:40 MQT: stat/rgb_pow/STATUS11 = {"StatusSTS":{"Time":"2019-11-07T00:41:40","Uptime":"0T02:43:14","UptimeSec":9794,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Dimmer":100,"Color":"FF00AA","HSBColor":"320,100,100","Channel":[100,0,67],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"HTPC","BSSId":"50:64:2B:5B:41:59","Channel":6,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

ascillato2, 2019-11-06:
@ncelectronico 

Just to confirm. Are you 100% sure that your device is a **Sonoff Pow R2** and not a **Sonoff Pow**?

Have you tried just the Sonoff Pow Template?

ncelectronico, 2019-11-07:
@ascillato2 
[image]
I have just tried the POW (without R2) module and template : Power is displayed !!!
[image]

So, a POW ver 2.0 is NOT a POW R2 ... ?!?!

ascillato2, 2019-11-07:
They are different devices.

Great to know that now it is working for you :+1:

closed 2019-11-07 by ascillato2