arendst/Tasmota #6853: Request for DS18B20 sensors to always add ID in MQTT message

mariobrx, 2019-11-06:
When having multiple DS18B20 sensors, the MQTT message provides info of each sensor ID like this: ..."DS18B20-1":{"Id":"011543E249BA","Temperature":55.2},"DS18B20-2":{"Id":"0115340334BA","Temperature":45.3}, ...
When there is only one DS18B20 sensor, the ID is not published: ..."DS18B20":{"Temperature":22.19}...
So when you have 10 sensors, you can doublecheck that you attribute the correct temperature to it's corresponding sensor. When one of the sensors fails, you know exactly which sensor you should check. Now, when you only have 2 sensors and one of them fails, you don't know which sensor is failing as you no longer receive the ID of the working sensor, 

So my request is very simple: please always send the ID of the DS18B20 sensor, even when there's only one connected.

Additional context: I send temperatures of DS18B20 sensors, connected on Wemos D1 mini modules via MQTT to another device that consolidates all changes and attributes each sensor to the right device or room .

arendst, 2019-11-07:
Done

```
"DS18B20":{"Id":"0000048EC44C","Temperature":22.2}
```

blakadder, 2019-11-07:
Fixed and merged in dev.

closed 2019-11-07 by blakadder

mariobrx, 2019-11-07:
Thank you very much! I really appreciate this fast implementation of my request. Now I can continue my scripts as I intended to.

Thanks again !!