arendst/Tasmota #6854: Backlog not working anymore after upgrade to 6.7.1

paolo9785, 2019-11-06:
After upgrading from 6.6 to 6.7.1 on all my Sonoff Dual devices Backlog http command does not work anymore.

Sample command tested:
http://192.168.1.114/cm?cmnd=Backlog%20Power2%20On;%20Delay%20240;%20Power2%20Off 

Expected behavior: Power 2 ON --> Wait 24 seconds --> Power 2 OFF

Actual behavior: Power 2 ON --> No further action

meingraham, 2019-11-06:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

P.S. Consider setting `PulseTime2 124` one time and then controlling the device with a single command `Power2 ON`. This eliminates the need for `Backlog` for the moment.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-06 by ascillato2

localhost61, 2019-11-06:
@paolo9785 I confirm this behavior with v7.0.0.3 too.
The following works as intended in the WebUI console, activating the relay during 5s :
```
backlog pulsetime1 0; pulsetime2 0
backlog power1 1;delay 50;power1 0
```
but ` http://192.168.1.114/cm?cmnd=backlog power1 1;delay 50;power1 0` doesn't close the relay as you tell.

ascillato, 2019-11-06:
you need to replace spaces and ; with the correspondent http code

paolo9785, 2019-11-07:
@ascillato even putting the correc %20 space code Backlog does not work properly

localhost61, 2019-11-07:
@ascillato you are right, it works. I didn't know for the ';' to be replaced by %3B, I thought Firefox would do the conversion if necessary like it does with the space character into %20.
`http://192.168.1.114/cm?cmnd=backlog power1 1%3Bdelay 50%3Bpower1 0` works here.