arendst/Tasmota #6856: rules enabled caused compiler failure

stefanbode, 2019-11-07:
Compiling .pioenvs\tasmota\libaae\bearssl-esp8266\codec\enc32be.c.o
C:/Users/stbode/Documents/GitHub/Sonoff-Tasmota/tasmota/xdrv_10_rules.ino: In function 'void RulesVarReplace(String&, const String&, const String&)':
C:/Users/stbode/Documents/GitHub/Sonoff-Tasmota/tasmota/xdrv_10_rules.ino:383:35: error: **'memmove_P' was not declared in this scope**
     memmove_P(write_to, find, flen);                      // Make variable Uppercase

localhost61, 2019-11-07:
Weird, no compilation issue here.

stefanbode, 2019-11-07:
Ok, I will double check. Can you add a typo mistake at this point to see, if your compiler really compile this part of the code. In this case you Must get an error.

localhost61, 2019-11-07:
I did it as you suggest and the typo was detected. It's a good trick as I could compile minimal.bin without triggering the error ;-)

```
Compiling .pioenvs\tasmota-FR\lib152\bearssl-esp8266\ec\ec_pubkey.c.o
Compiling .pioenvs\tasmota-FR\lib152\bearssl-esp8266\ec\ec_secp256r1.c.o
D:/RnD/gitHub/Sonoff-Tasmota/arendst/development/tasmota/xdrv_10_rules.ino: In function 'void RulesVarReplace(String&, const String&, const String&)':
D:/RnD/gitHub/Sonoff-Tasmota/arendst/development/tasmota/xdrv_10_rules.ino:383:37: error: expected ';' before 'Make'
     memmove_P(write_to, find, flen)                      Make variable Uppercase
                                     ^
```

localhost61, 2019-11-07:
This false alarm happens to me sometimes and disappears after running the PIO clean rule.
Erasing the .cache folder in the project may help too.

localhost61, 2019-11-07:
Maybe an artifact of the project renamed into Tasmota. I did it locally, I see you didn't.

Jason2866, 2019-11-07:
@stefanbode  Replace hidden folder `.platformio` in C:\users\....
Your PlatformIO setup is messed up.
PlafformIO will be reinstalled from VSC

ascillato, 2019-11-07:
No compilation issue here. 

@stefanbode Did you solve your issue?

localhost61, 2019-11-07:
@ascillato Considering [PR#6858](https://github.com/arendst/Tasmota/pull/6858) I think he did

philbarclay, 2019-11-07:
I have seen errors like this in the past that are resolved by re-running the pio run command.  I think it has something to do with the .ino -> .cpp conversion process.

Jason2866, 2019-11-08:
@philbarclay  No, that doesent help. Something is messed up in folder `.platformio`
Only solution is to delete...

ascillato, 2019-11-08:
@stefanbode Did you solve your issue?

stefanbode, 2019-11-08:
Just now solved 2min ago: solution was to delete the .platform.ini

localhost61, 2019-11-08:
@stefanbode You mean the .plaformio folder or the platformio.ini project file?  In that later case you restored from mainstream?

stefanbode, 2019-11-08:
I just deleted the folder

closed 2019-11-08 by stefanbode

byte4geek, 2019-11-12:
I have the same issue with the new renamed project to tasmota and delete .platformio dir or clean and delete cache do not solve it.
i compiled fine up to the version 6.6.0.12

ascillato, 2019-11-12:
You need to delete the folders:

`C:\Users\user\.platformio`

And in the compiling folder:

`.pio`
`.pioenvs`
`.cache`

byte4geek, 2019-11-13:
I done all of these steps but the error remains

ascillato, 2019-11-13:
So, your VSC brokes. You will need to uninstall everything and delete all folders PIO, Tasmota and VSC folders.

localhost61, 2019-11-13:
and even format your hard disc I fear ...  XD

localhost61, 2019-11-13:
- Can't you rely on the error messages you get?
- Did you try to run the clean task in PIO?

byte4geek, 2019-11-14:
i tryed on a new PC, the issue is present only if i use core_2_3_0.platform and core_2_3_0.build_flags and tasmota evironment, is not present if i build a minimal.

localhost61, 2019-11-14:
>the issue is not present if i build a tasmota-minimal with the same core_2_3_0.

Yes, this part is not compiled in tasmota-minimal, whatever the core.

Jason2866, 2019-11-14:
Core support for old cores will be droped beginning with release v.7.x
Every change is NOT tested anymore to run with core < 2.6.0.

byte4geek, 2019-11-14:
i solved with commenting the line 383 "memmove_P(write_to, find, flen);                      // Make variable Uppercase"

for me the core 2.6 is very unstable, for example when i do the ota upgrade, on the sonoff mini or ESP board too, all of these device do not reconnect to wifi automatically untill i reset or power off and power on manually.
On serial log i can see unable to connect to wifi.
Core 2.3 is very more stable.

ascillato, 2019-11-14:
* Core 2.3 and 2.4.2 have security issues. It is very easy to hack your device and take control of it.

* When changing cores or just changing SDK (the low level routines of the core) it is recommended to perform a `reset 3` to delete SDK RF calibration and let the SDK start its calibrations from skratch.

If that don't make your device stable, you will need to backup your settings and make a `reset 5`

Or as last option, Erase all flash by serial with esptool.py as explained in the wiki, and then flash the precompiled bins.

ascillato, 2019-11-14:
@byte4geek 

```
memmove_P(write_to, find, flen); // Make variable Uppercase
```
Was introduced on Nov,1 with commit 49e8f52a43b282144a230952b1f44d7943edda06

So, it is very recent and not compatible with core 2.3.0. Sorry.

You will need to revert that commit in your setup, or rules won't work.

localhost61, 2019-11-15:
> i solved with commenting the line 383 "memmove_P(write_to, find, flen); // Make variable Uppercase"

@byte4geek 
Considering [this](https://github.com/esp8266/Arduino/pull/6368), as a workaround  to compile with core 2.3.0 you could add the missing code near line `#180` instead of removing the offending call.

```
#if defined(ARDUINO_ESP8266_RELEASE_2_3_0) || defined(ARDUINO_ESP8266_RELEASE_2_4_2) 
void *memmove_P(void *dest, const void *src, size_t n) {
  if (src > (void*)0x40000000) return memcpy_P(dest, src, n);
  else return memmove(dest, src, n);
}
#endif
```