arendst/Tasmota #6857: Tasmoda to Finder 26.04

asterix4you, 2019-11-07:
Hi, is it possible to configure the Dual R2 sonoff (tasmoda) in "Finder 26.04" relay mode? I have not found any information on the WEB.
MB.

Jason2866, 2019-11-07:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

blakadder, 2019-11-07:
Closing issue. Redirected to proper support channel.

closed 2019-11-07 by blakadder