arendst/Tasmota #6859: More Arilux devices (Modules)

0n3-70uch, 2019-11-07:
Hello, I think it would be nice if we add more Arilux devices. I added some informations on this page:
https://github.com/arendst/Tasmota/wiki/MagicHome-with-ESP8285-(Arilux)

You can buy these devices:
| Model | Color Support | Voltages (sometimes) | Remote | Link                      |
|-------|---------------|----------|--------|---------------------------|
| LC01  | RGB           | 5-28V    | None   | [Banggood][LC01-banggood] |
| LC02  | RGBW          | 9-12V (5-28V)    | None   | [Banggood][LC02-banggood] |
| LC03  | RGB           | 5-28V    | IR     | [Banggood][LC03-banggood] |
| LC04  | RGBW          | 9-12V (5-28V)    | IR     | [Banggood][LC04-banggood] |
| LC08  | RGBWW         | 5-28V    | None   | [Banggood][LC08-banggood] |
| LC09  | RGB           | 5-28V    | RF     | [Banggood][LC09-banggood] |
| LC10  | RGBW          | 9-28V    | RF     | [Banggood][LC10-banggood] |
| LC11  | RGBWW         | 9-28V    | RF     | [Banggood][LC11-banggood] |

Two more support pages from Tasmota:
https://github.com/arendst/Tasmota/wiki/Arilux-LC04
https://github.com/arendst/Tasmota/wiki/Arilux-LC02

I can change the three support pages to one next weekend.

[LC01-banggood]: http://www.banggood.com/ARILUX-AL-LC01-Super-Mini-LED-WIFI-Smart-RGB-Controller-For-RGB-LED-Strip-Light-DC-9-12V-p-1058603.html?rmmds=search
[LC02-banggood]: http://www.banggood.com/ARILUX-AL-LC02-Super-Mini-LED-WIFI-APP-Controller-Dimmer-for-RGBW-LED-Strip-Light-DC-9-12V-p-1060222.html
[LC03-banggood]: http://www.banggood.com/ARILUX-AL-LC03-Super-Mini-LED-WIFI-APP-Controller-Remote-Control-For-RGB-LED-Strip-DC-9-12V-p-1060223.html
[LC04-banggood]: http://www.banggood.com/ARILUX-AL-LC04-Super-Mini-LED-WIFI-APP-Controller-Remote-Control-For-RGBW-LED-Strip-DC-9-12V-p-1060231.html
[LC08-banggood]: http://www.banggood.com/ARILUX-AL-LC08-Super-Mini-LED-WIFI-APP-Controller-Dimmer-for-RGBWW-LED-Strip-Light-DC-5-28V-p-1081241.html
[LC09-banggood]: http://www.banggood.com/ARILUX-AL-LC09-Super-Mini-LED-WIFI-APP-Controller-RF-Remote-Control-For-RGB-LED-Strip-DC9-28V-p-1081344.html
[LC10-banggood]: http://www.banggood.com/ARILUX-AL-LC10-Super-Mini-LED-WIFI-APP-Controller-RF-Remote-Control-For-RGBW-LED-Strip-DC9-28V-p-1085111.html
[LC11-banggood]: http://www.banggood.com/ARILUX-AL-LC11-Super-Mini-LED-WIFI-APP-Controller-RF-Remote-Control-For-RGBWW-LED-Strip-DC9-28V-p-1085112.html

blakadder, 2019-11-07:
That way of configuring devices is outdated. Use [Templates ](https://github.com/arendst/Tasmota/wiki/Templates) instead and check the [Template Repo](https://blakadder.github.io/templates/light.html) for already submitted Arilux devices.
Don't add devices you don't own and didn't verify they're working

0n3-70uch, 2019-11-07:
> That way of configuring devices is outdated. Use [Templates ](https://github.com/arendst/Tasmota/wiki/Templates) instead and check the [Template [...]

Oh thats cool. Thanks for the tip.

closed 2019-11-07 by blakadder

Tech-TX, 2020-03-21:
@0n3-70uch Fabian, I realize this is a closed issue, but here's some additional documentation you might find useful.

I just finished reverse-engineering the schematic of the [MagicHome RGBW strip-LED controller](https://github.com/arendst/Tasmota/wiki/MagicHome-with-ESP8285), and during the process was sorely confused at the EXT_RSTB pin as I only saw a single capacitor hanging on it.  The ESP8285 chip seems to have an _internal pull-up resistor_ on RST that's not mentioned anywhere in the docs from Espressif, or from anyone else that I've googled today.

I buzzed the RST pin to every pad on every resistor on the board, and it only has an (unknown) cap to ground on the RST pin, yet the 'scope is showing a solid 3,3V on the pin.  Hanging a 10K resistor on RST the voltage dropped to 0,7V, so the equivalent internal resistance is ~ 37K ohms.   It's a weak P-FET, not a 'resistor' that you can measure with an ohmmeter.

[image]

The JoulWatt JW5017S regulator on the board *works*, but the implementation is marginal.  There's a footprint for an A-size tantalum cap on the 3.3V output that's not populated, so there's only a few ceramic caps to keep the voltage steady while WiFi is hitting the current.  There's ~ 100mV worth of noise on the 3,3V supply while everything is running, so it's within spec for the CPU.

I couldn't measure any of the capacitors without removing them from the board, so most are unknown in the schematic.  The schematic is released to the public domain, if you folks in the Tasmota group want to add it to your wiki.  _I'll just leave this here..._ :smile:

0n3-70uch, 2020-03-21:
> @0n3-70uch Fabian, I realize this is a closed issue, but here's some additional documentation you might find useful.
>
> I just finished reverse-engineering the schematic of the [MagicHome RGBW [...]

Hi, thats really really nice. Thanks for your work with this shematic. How to find out if the ESP8285 comes with an internal resistor... or is it just a bad wiring?

Tech-TX, 2020-03-21:
I measured it by connecting a 10K ohm resistor to ground on the RST pin.  The voltage dropped to 0.7V, so the internal 'resistor' is roughly 37K ohms.

0n3-70uch, 2020-03-21:
> I measured it by connecting a 10K ohm resistor to ground on the RST pin. The voltage dropped to 0.7V, so the internal 'resistor' is roughly 37K ohms.

Ah great.

Tech-TX, 2020-03-21:
Here's an idea that I saw in git memory for the programming.  It uses a 5 pin header so you can connect a jumper block / shunt to the GPIO0 pin for programming with an FTDI board and 3 Dupont wires.  I need to add a small spot of epoxy as the two ground pins (the green wire) are floating in the header.  It fits in the original plastic case.

[image]

0n3-70uch, 2020-03-21:
This is nice for programming... I solder some wires to the PCB and remove it after flashing.