arendst/Tasmota #6862: Flickering H801 low brightness

hawkeyexp, 2019-11-07:
Hi, im using a h801 controller with one channel led panel - it is flickering with brightness < 30% - is there a way to set the min value to 30% to prevent flickering? For example all values from 1-29% are auto set to 30%?

ascillato, 2019-11-07:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-11-08:
When you have time, please, complete the template or address this to the Tasmota Support Chat. Thanks.

closed 2019-11-08 by ascillato2