arendst/Tasmota #6863: H801 forgets PWM settings CW WW

urmelchen, 2019-11-07:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): H801_____
- [x ] Tasmota binary firmware version number used: 6.7.1_____
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ x] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

18:05:11 RSL: RESULT = {"NAME":"Generic","GPIO":[255,52,255,255,41,57,0,0,39,38,40,37,0],"FLAG":0,"BASE":20}
18:05:11 RSL: RESULT = {"Module":{"20":"H801"}}
18:05:11 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"}}

  ```
- [ x] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

18:08:41 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":471,"Rules":"on Dimmer#Boot do color c030000050 endon"}
18:08:41 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
18:08:41 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

  ```
- [x ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
18:09:42 RSL: STATUS = {"Status":{"Module":20,"FriendlyName":["WZ-Licht1"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:09:42 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"External System","Uptime":"0T00:15:37","StartupUTC":"2019-11-07T16:54:05","Sleep":0,"CfgHolder":4617,"BootCount":318,"SaveCount":364,"SaveAddress":"F8000"}}
18:09:42 RSL: STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019.10.26 13:25:04","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
18:09:42 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005AFF000000000000","00000000"]}}
18:09:42 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":436,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:09:42 RSL: STATUS5 = {"StatusNET":{"Hostname":"licht4","IPAddress":"x.x.x.x","Gateway":"x.x.x.x","Subnetmask":"255.255.255.0","DNSServer":"x.x.x.x","Mac":"xxx","Webserver":2,"WifiConfig":4}}
18:09:42 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 07 17:09:42 2019","Local":"Thu Nov 07 18:09:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:46","Sunset":"17:21"}}
18:09:42 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-11-07T18:09:42"}}
18:09:42 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-11-07T18:09:42","Uptime":"0T00:15:37","UptimeSec":937,"Heap":28,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":0,"POWER":"ON","Dimmer":75,"Color":"C030000050","HSBColor":"15,100,75","Channel":[75,19,0,0,31],"CT":500,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":12,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:07"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
color c030005050
Turn Power OFF (wallswitch)
Turn Power ON
color = c030000000
(without Rule1)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I upgraded from 6.6.0 to 6.7.1
before i set color once and turned off and on power by switch
color did never change

now with 6.7.1
RGB stays same color but CW WW reset to 0

(workaround via rule)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-11-07:
Duplicate of #6764 

Can you tell us more about your use case? Did you try `SetOption37 128` ?

urmelchen, 2019-11-07:
SetOption37 128 does not seem to do it.
setoption68 1 does, but i don't like it.
I prefer the workaround with rule "on Dimmer#Boot do color c030000050 endon"

use case:
H801 controls RGBW downlights in the living room, white=5W, RGB=1W each.
turn on and off by wallswitch.
pure white is too cold so i added some orange.
color usualy stays the same all the time, but i can toogle with a brighter setting .

now after reading #6764 i understand that this change is by intent.
hope you will find another way to avoid overloading and restore rgbww at the same time. 
Why not set 8080808080 or whatever in firmware defaults?

closed 2019-11-07 by urmelchen

s-hadinger, 2019-11-07:
I understand your problem. Changing the default value wouldn't really solve it because there are many firmwares included in tuya-convert out there.

I could integrate another protection. The default value is #FFFFFFFFFF with Dimmer 10 (which is displayed as #1A1A1AFFFF), so I could only deactivate CW/WW in this case, and keep it in all other cases.

That would mean that after a reboot, you get back your previous color levels, except if you set `Color 1A1A1AFFFF` that would reload as `Color 1A1A1A0000`.

Would that be ok with you?

urmelchen, 2019-11-07:
This would be ok.

Would it be possible to do a soft on?
Dimm all channels from 0 to last value in a couple of seconds.

Thank you

______________________________________________
Dieter Neumann, Koppler Str. 14, 90513 Zirndorf

s-hadinger, 2019-11-07:
Ok.
I never looked it the soft on/off feature and will not have time to do so in the near future.

PR #6870 (2019-11-08): Fix restore all 5 channels at boot
## Description:

Restores all 5 channels at reboot, reversing partially the behavior introduced in #6612 
The only case where White is disabled is when `Color 1A1A1AFFFF` which is transformed to `Color 1A1A1A0000` at reboot. This is to avoid overloading a freshly tasmotized device, as reported in #6534

**Related issue (if applicable):** fixes #6863 #6764 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and wo [... cut]