arendst/Tasmota #6865: Sonoff RF Bridge publishes only when the sensor stops emitting the signal

terba, 2019-11-07:
### BUG DESCRIPTION
Sonoff RF Bridge with original RF firmware does not publishes an MQTT message until the sensor stops emitting its signal. For example a water leakage sensor's message appears when the water is gone, or the smoke detector's when the house is burned to ashes :)

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff RF Bridge
- [X] Tasmota binary firmware version number used: 6.6.0.12
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: platformio
- [X] Flashing tools used: platformio direct upload
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:
```
stat/sonoff-bridge/RESULT {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
stat/sonoff-bridge/RESULT {"Module":"25 (Sonoff Bridge)"}
stat/sonoff-bridge/RESULT {"GPIO2":"0 (None)","GPIO4":"0 (None)","GPIO5":"0 (None)","GPIO12":"0 (None)","GPIO14":"0 (None)"}
```
- [X] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
```
no rules
```
- [X] Provide the output of this command: ``Status 0``:
```
stat/sonoff-bridge/STATUS {"Status":{"Module":25,"FriendlyName":["Sonoff-Bridge"],"Topic":"sonoff-bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
stat/sonoff-bridge/STATUS1 {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"************","RestartReason":"Power on","Uptime":"3T12:43:33","StartupUTC":"2019-11-04T06:25:02","Sleep":50,"CfgHolder":4617,"BootCount":34,"SaveCount":49,"SaveAddress":"FB000"}}
stat/sonoff-bridge/STATUS2 {"StatusFWR":{"Version":"6.6.0.12(sonoff)","BuildDateTime":"2019-09-11T19:54:11","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
stat/sonoff-bridge/STATUS3 {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":1,"LogHost":"*************","LogPort":514,"SSId":["************",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008028","0C0500000200060000005A64000000000000","00000800"]}}
stat/sonoff-bridge/STATUS4 {"StatusMEM":{"ProgramSize":413,"Free":588,"Heap":31,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","80880046","042283A0","000004C7","000000C0","00000001"],"Drivers":"2,3,4,6,8,16,18,19,22,24","Sensors":"1,3,4,5,6,9"}}
stat/sonoff-bridge/STATUS5 {"StatusNET":{"Hostname":"*************","IPAddress":"*************","Gateway":"*************","Subnetmask":"*************","DNSServer":"*************","Mac":"*************","Webserver":0,"WifiConfig":5}}
stat/sonoff-bridge/STATUS6 {"StatusMQT":{"MqttHost":"*************","MqttPort":*************,"MqttClientMask":"DVES_%06X","MqttClient":"*************","MqttUser":"*************","MqttCount":5,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
stat/sonoff-bridge/STATUS7 {"StatusTIM":{"UTC":"Thu Nov 07 19:08:35 2019","Local":"Thu Nov 07 20:08:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99}}
stat/sonoff-bridge/STATUS10 {"StatusSNS":{"Time":"2019-11-07T20:08:35"}}
stat/sonoff-bridge/STATUS11 {"StatusSTS":{"Time":"2019-11-07T20:08:35","Uptime":"3T12:43:33","UptimeSec":305013,"Vcc":3.447,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"Wifi":{"AP":1,"SSId":"*************","BSSId":"*************","Channel":*************,"RSSI":70,"LinkCount":5,"Downtime":"0T00:05:26"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
```
  I have no console, only mqtt
```

### TO REPRODUCE

Put a water leakage sensor into water, it starts emitting the signal. Take it out of the water later, and the only time when az MQTT message is genetated is at this point. So the message appears when the problem is over. Same with smoke detector of an other brand.

The published message is this:
```
tele/sonoff-bridge/RESULT {"Time":"2019-11-07T20:33:01","RfReceived":{"Sync":10100,"Low":320,"High":920,"Data":"******","RfKey":"None"}}
```
### EXPECTED BEHAVIOUR

The message in the queue should be presented as soon as the first signal is re

### ADDITIONAL CONTEXT

This is the same problem as described in #3450 and #2434.

My suspect is **SFB_TIME_AVOID_DUPLICATE**, so this line in **xdrv_06_snfbridge.ino**:
```c
      if (!((received_id == SnfBridge.last_received_id) && (now - SnfBridge.last_time < SFB_TIME_AVOID_DUPLICATE))) {
```

Jason2866, 2019-11-07:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-11-08:
Hi,

Please, address this to the support chat because seems that there is a misconfiguration in your device. I can't reproduce your issue on latest Tasmota (7.0.0.4). If in a sonoff bridge, a code is received by RF repeatedly, the same code is being published to mqtt every second until the RF code receiving stops.

For example, I have a RF button that if I press once, it will send one mqtt message, but it I keep it pressing for more than 2 seconds, it will send another mqtt message:

```
rule1
  on rfreceived#data=7ffff2 do backlog publish home/button1 PRESSED; add2 1; ruletimer2 3 endon
  on rules#timer=2 do var2 0 endon
  on var2#state>1 do backlog var2 0; publish home/button1 HOLD endon
```

closed 2019-11-08 by ascillato2

terba, 2019-11-08:
I flashed the EFM8BB1 with the Portisch firmware and the problems have gone. It looks like the stock (Sonoff) firmware of this chip is wrong:
https://support.itead.cc/support/discussions/topics/11000027187
https://support.itead.cc/support/discussions/topics/11000019077
Sorry for the false bug report!

ascillato, 2019-11-08:
You are right! I have Portisch firmware installed on my sonoff bridge of the example. Thanks for pointing this out. Will be useful for others finding this issue.