arendst/Tasmota #6871: Rollershutter switches on two relays instead of one

Gu213, 2019-11-08:
### BUG DESCRIPTION
When ShutterClose command is issued Sonoff is switching on a pair of relays instead of one relay. This would drive roller shutter motor in two directions at one time.
When PulseTime and Interlock configured for InterlockMode 1, the module does not drive relays at ShutterOpen command.

- [v ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ v] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [v ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ v] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ v] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [v ] Device used (e.g., Sonoff Basic): Eachen 4ch DC (Sonoff 4ch clone)
- [ v] Tasmota binary firmware version number used: 6.7.1
  - [ ] Pre-compiled
  - [ v] Self-compiled
    - [ v] IDE / Compiler used: Github
- [v ] Flashing tools used: device web page
- [ v] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
18:58:13 MQT: stat/sonoff/RESULT = {"NAME":"Eachen 4ch","GPIO":[17,0,0,0,23,22,18,19,21,52,20,24,0],"FLAG":0,"BASE":18}
18:58:13 MQT: stat/sonoff/RESULT = {"Module":{"0":"Eachen 4ch"}}
18:58:13 MQT: stat/sonoff/RESULT = {"GPIO":"Not supported"}

  ```
- [ -] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
18:57:04 MQT: stat/sonoff/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
18:57:04 MQT: stat/sonoff/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
18:57:05 MQT: stat/sonoff/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

  ```
- [v ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
19:15:51 CMD: Group 0, Index 1, Command "STATUS", Data "0"
19:15:51 MQT: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:15:51 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:23:23","StartupUTC":"2019-11-08T17:52:28","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":24,"SaveAddress":"FA000"}}
19:15:51 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-11-01T22:01:30","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
19:15:51 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000000"]}}
19:15:51 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":549,"Free":452,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE393","043683A1","0280008D","01001BC0","0000F181"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,30","Sensors":"1,2,3,4,6,20,22,34"}}
19:15:51 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6260","IPAddress":"192.168.0.106","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"62.179.1.62","Mac":"C4:4F:33:8C:D8:74","Webserver":2,"WifiConfig":4}}
19:15:51 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.247","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8CD874","MqttUser":"openhab","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
19:15:51 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Nov 08 18:15:51 2019","Local":"Fri Nov 08 19:15:51 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:47","Sunset":"17:19"}}
19:15:51 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-11-08T19:15:51","Shutter1":{"Position":100,"direction":0}}}
19:15:51 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-11-08T19:15:51","Uptime":"0T00:23:23","UptimeSec":1403,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"IOT","BSSId":"C2:4A:00:5D:AE:D4","Channel":1,"RSSI":28,"LinkCount":1,"Downtime":"0T00:00:05"}}}

  ```
- [v ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

CASE 1 - right after configuring the shutter mode in the following way:
18:56:15 CMD: Backlog SetOption80 on; ShutterRelay1 1
18:56:16 MQT: stat/sonoff/RESULT = {"SetOption80":"ON"}
18:56:16 SHT: Relay 1 is 1
18:56:16 SHT: Accuracy digits: 0
18:56:16 SHT: RESET/INIT CALIBRATION MATRIX DIV 0
18:56:16 SHT: Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1
18:56:16 MQT: stat/sonoff/RESULT = {"ShutterRelay1":1}

At close command two relays are energised (1 and 2):
18:57:20 CMD: shutterclose1
18:57:20 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
18:57:20 MQT: stat/sonoff/POWER1 = OFF
18:57:20 SHT: Wait for Motorstop 500
18:57:20 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
18:57:20 MQT: stat/sonoff/POWER2 = ON
18:57:20 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
18:57:20 MQT: stat/sonoff/POWER1 = ON
[... 57 lines cut ...]
19:07:55 SHT: Switched relay: 2 by I
19:07:55 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
19:07:55 MQT: stat/sonoff/POWER2 = OFF
19:07:55 SRC: Shutter
19:07:55 SHT: Switched relay: 1 by Shutter
19:07:55 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
19:07:55 MQT: stat/sonoff/POWER1 = ON
19:07:55 MQT: stat/sonoff/RESULT = {"ShutterClose1":0}
19:07:55 CFG: Saved to flash at FA, Count 24, Bytes 4096

  ```

### TO REPRODUCE
as above

### EXPECTED BEHAVIOUR
Relays 1 and 2  should interlock themselves. After the duration time both relays should switch off.

### ADDITIONAL CONTEXT

When I noted the strange behaviour, made few attempts to reset, refresh, reconfigure. Finally for this demonstration it was reset 5, minimal flash, target flash, configured.
When inverted shutter logic, same behaviour.
When Mqqt broker alive - same behaviour

**(Please, remember to close the issue when the problem has been addressed)**

Gu213, 2019-11-08:
I went through the logs one more time and found that my heading description is wrong when saying that at ShutterOpen the module does not energise. As per pasted logs - it should be described that ShutterClose command does not initiate the count down timer.

stefanbode, 2019-11-09:
You have. Or defined an interlock. Therefore the shutter assumes you have a fail save configuration where relay is power and relay 2 is up/down. The behavior is intended. You can add a rule to set relay 2 to off if Relay1 goes off. If interlock is set relay 1 is down and relay2 is up. In this case there is never power on both relays

stefanbode, 2019-11-09:
Interlock 1,2
Should solve your problem

Gu213, 2019-11-09:
Thank you Stefan for responding to my problem.
It seems that setting shutter mode to 0 has fixed the behaviour. You had a good feeling when you wrote "relay 2 is up/down". This led me to check the shutter mode again. It was 1.
I will attach the motors and hope they work.
Anywhere thank you so much for wonderful job on the roller shutter control :)

ascillato2, 2019-11-09:
Great to know that now it is working for you.

Closing.

closed 2019-11-09 by ascillato2

stefanbode, 2019-11-10:
Take a look at the video in the wiki. Calibration is awesome if you do it right. Take a little time but is fun to see how accurate it can be

Gu213, 2019-11-11:
Thanks Stefan,
my shutters have mechanical limit switches so I do not care so much about calibration of the controls. I give ca. 2 seconds of overhead to each way drive time to be sure that irrespective of changing torsion they will arrive to the end position. 
I have flashed all my sonoffs to the rollershuter software. They work well with OpenHab2. I'm a beginner in IOT programming. Don't know yet how to program roller shutter item with mqqt topic without using of rules. Item gives command UP, DOWN and STOP. However Sonoff expects command ShutterOpen1 etc. I convert the commands from UP to ShutterOpen1 within rules but there is probably a simpler and more elegant way.
What is bothering me now is how to flash the Sonoff RF and allow controls from the remote. Will keep reading your wiki on that new topic :)

stefanbode, 2019-11-12:
Hi, regarding the UP/DOWN/STOP I have an idea to bundle this into the shutterposition command. Let me do a try. Even if you have end-stops it is a good idea to have a proper calibration. Otherwise, you will have problems if you move to a designated position e.g. 40% you will get different results if coming from open or from close. As long as you only fully open and close I agree. No need for calibration. I have sunblinds in the garden that moves with the sun :-)

stefanbode, 2019-11-12:
I have added this functionality and created a pull request against the master project. You can use shutterposition and then UP/DOWN/STOP to control

Gu213, 2019-11-13:
Thanks Stefan. Will keep an eye on this function in the GitHub and compile when it is available. I bet OpenHab community will like it :)