arendst/Tasmota #6875: Hostname i ESP-xxxxx in DHCP

blakadder, 2019-11-09:
### BUG DESCRIPTION
Device hostname is not tasmota-xxxx but rather ESP-xxxxxx. Problem reported by a user then reproduced on two different network setups. Happens with commit ae3e45b builds, precompiled and self compiled even after clean serial flash. 7.0.0.3 build does not exhibit such behaviour!

this is logs from selfcompiled binary:
```
00:00:00 CFG: Loaded from flash at F5, Count 21
00:00:00 Project tasmota Bathroom Light Version 7.0.0.4(tasmota)-STAGE
00:00:00 WIF: Connecting to AP2 HTPCext in mode 11N as bathroom_switch-1102...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on bathroom_switch-1102 with IP address 10.1.1.228
```
```
Sat Nov  9 02:37:47 2019 daemon.info dnsmasq-dhcp[2403]: DHCPDISCOVER(br-lan) 80:7d:3a:61:24:4e
Sat Nov  9 02:37:47 2019 daemon.info dnsmasq-dhcp[2403]: DHCPOFFER(br-lan) 10.1.1.228 80:7d:3a:61:24:4e
Sat Nov  9 02:37:47 2019 daemon.info dnsmasq-dhcp[2403]: DHCPREQUEST(br-lan) 10.1.1.228 80:7d:3a:61:24:4e
Sat Nov  9 02:37:47 2019 daemon.info dnsmasq-dhcp[2403]: DHCPACK(br-lan) 10.1.1.228 80:7d:3a:61:24:4e ESP-61244E
```

precompiled:
```
00:00:00 CFG: Loaded from flash at FA, Count 14
00:00:00 Project tasmota Tasmota Version 7.0.0.4(ae3e45b-tasmota)-STAGE
00:00:00 WIF: Connecting to AP2 HTPC in mode 11N as tasmota-7718...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-7718 with IP address 10.1.1.150
```

```
Sat Nov  9 02:41:15 2019 daemon.info dnsmasq-dhcp[2403]: DHCPDISCOVER(br-lan) 80:7d:3a:3d:5e:26
Sat Nov  9 02:41:15 2019 daemon.info dnsmasq-dhcp[2403]: DHCPOFFER(br-lan) 10.1.1.150 80:7d:3a:3d:5e:26
Sat Nov  9 02:41:15 2019 daemon.info dnsmasq-dhcp[2403]: DHCPREQUEST(br-lan) 10.1.1.150 80:7d:3a:3d:5e:26
Sat Nov  9 02:41:15 2019 daemon.info dnsmasq-dhcp[2403]: DHCPACK(br-lan) 10.1.1.150 80:7d:3a:3d:5e:26 ESP-3D5E26
```

### TO REPRODUCE
use binary from commit ae3e45b

ascillato, 2019-11-09:
Hi, I could also reproduce this by:
* Flashing the precompiled bins.
* Selfcompiling with PIO with core pre2.6.0

commit b944166cd502a6fdf35e243170e5a89e40da58e8 (7.0.0.3) Don't have this issue
commit e099ebe7b62b313cf7ad127b8ed1165c210194e5 (7.0.0.4) (new Wifipower command) Introduce this problem.

arendst, 2019-11-09:
Verified. Nice issue. Working on it.

blakadder, 2019-11-09:
Works on my side!

ascillato2, 2019-11-09:
Closing as this issue has been fixed. Thanks!!!

closed 2019-11-09 by ascillato2