arendst/Tasmota #6878: Power Meter - SDM630 & additional values

altbau9944, 2019-11-09:
Hello,

I have successfully integrated the EASTRON SDM630 via ESP32 and a Modbus RS485 adapter into my Openhab. The communication is handled via MQTT and everything works very well.

Now I want to read additional values from the energy meter.
1. frequency
2. phase_angle
3. import_active
4. export_active
5. import_reactive
6. export_reactive

Unfortunately I have problems to see these values in the Tasmota web interface. 
Can someone tell me how to adjust the "xnrg_10_sdm630.ino" to see these values?

Many thanks in advance...

ascillato2, 2019-11-09:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-09 by ascillato2

pablozg, 2019-11-09:
Hi, try this code and if works well I will send a PR to integrate if Theo are agree.

You must copy all code and replace it in xnrg_10_sdm630.ino

````
/*
  xnrg_10_sdm630.ino - Eastron SDM630-Modbus energy meter support for Tasmota

  Copyright (C) 2019  Gennaro Tortone, Theo Arends and Pablo Zerón

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifdef USE_ENERGY_SENSOR
#ifdef USE_SDM630
/*********************************************************************************************\
 * Eastron SDM630-Modbus energy meter
 *
 * Based on: https://github.com/reaper7/SDM_Energy_Meter
\*********************************************************************************************/

#define XNRG_10             10

// can be user defined in my_user_config.h
#ifndef SDM630_SPEED
  #define SDM630_SPEED      9600    // default SDM630 Modbus address
#endif
// can be user defined in my_user_config.h
#ifndef SDM630_ADDR
  #define SDM630_ADDR       1       // default SDM630 Modbus address
#endif

#include <TasmotaModbus.h>
TasmotaModbus *Sdm630Modbus;

const uint16_t sdm630_start_addresses[] {
  0x0000,  // L1 - SDM630_VOLTAGE [V]
  0x0002,  // L2 - SDM630_VOLTAGE [V]
  0x0004,  // L3 - SDM630_VOLTAGE [V]
  0x0006,  // L1 - SDM630_CURRENT [A]
  0x0008,  // L2 - SDM630_CURRENT [A]
  0x000A,  // L3 - SDM630_CURRENT [A]
  0x000C,  // L1 - SDM630_POWER [W]
  0x000E,  // L2 - SDM630_POWER [W]
  0x0010,  // L3 - SDM630_POWER [W]
  0x0018,  // L1 - SDM630_REACTIVE_POWER [VAR]
  0x001A,  // L2 - SDM630_REACTIVE_POWER [VAR]
  0x001C,  // L3 - SDM630_REACTIVE_POWER [VAR]
  0x001E,  // L1 - SDM630_POWER_FACTOR
  0x0020,  // L2 - SDM630_POWER_FACTOR
  0x0022,  // L3 - SDM630_POWER_FACTOR
  0x0024,  // L1 - SD630_FASE_ANGLE [DEGREES]
  0x0026,  // L2 - SD630_FASE_ANGLE [DEGREES]
  0x0028,  // L3 - SD630_FASE_ANGLE [DEGREES]
  0x0042,  // TOTAL FASE ANGLE [DEGREES]
  0x0046,  // FREQUENCY [HZ]
  0x0048,  // IMPORT ACTIVE [W]
  0x004A,  // EXPORT ACTIVE [W]
  0x004C,  // IMPORT REACTIVE [VAR]
[... 242 lines cut ...]
      Sdm630Show(1);
      break;
#ifdef USE_WEBSERVER
    case FUNC_WEB_SENSOR:
      Sdm630Show(0);
      break;
#endif  // USE_WEBSERVER
    case FUNC_ENERGY_RESET:
      Sdm630Reset();
      break;
    case FUNC_INIT:
      Sdm630SnsInit();
      break;
    case FUNC_PRE_INIT:
      Sdm630DrvInit();
      break;
  }
  return result;
}

#endif  // USE_SDM630
#endif  // USE_ENERGY_SENSOR
````

If you need more info you can get the need address from https://github.com/reaper7/SDM_Energy_Meter, please test it and feedback.

altbau9944, 2019-11-23:
Hello,

@pablozg:
I did not manage to adjust the display with this INO file or any other adjustment in xnrg_10_sdm630.ino. No matter if I use dev 7.0.xx or the latest official release.

I also tried to get some support via Tasmota Support Chat. I no success. The search also takes a bit of getting used to.

Can somebody give me a hint how to display these values in the UI of the Tasmota interface?
- frequency
- phase_angle
- import_active
- export_active
- import_reactive
- export_reactive

Any feedback is welcome!!

arendst, 2019-11-23:
Have a look at xnrg_08_sdm120.ino

altbau9944, 2019-11-30:
Hello,

@arendst: I did but....

I'm afraid I have to come back to the subject. I only deal with the issue at the weekend, so I always get feedback later on.

I have used my time to try out some things. With the sensor.bin file an others as well.
If I select the SDM120 in the Tasmota settings, no frequency will be displayed in the web interface, although it is entered in xnrg_08_sdm120.ino.

[image]

[image]

From the file xnrg_08_sdm120.ino:
52: 0x0046, // SDM120C_FREQUENCY [Hz]
          Energy.frequency[0] = value; // 50.0 Hz

124: case 6:
125: Energy.frequency[0] = value; // 50.0 Hz
126: break;

Maybe I'm making a mistake, but at the moment I can't get any further here.
Would be great if I could get some help...

Regards

arendst, 2019-11-30:
Did you try the suggested code in the first answer from pablozg?

All he asks you to do is copy all the text and paste it in the file xnrg_10_sdm630.ino replacing everything.

Then select the SDM630 serial interfaces like you did the sdm120 above and see what happens.

altbau9944, 2019-12-01:
Hi Arendst,

Thank you very much for your effort and feedback!!
Yes, I did. To recheck this once again, I paste to code in again and upload this to my NodeMCU V3. Rebbot and ...

[image]

[image]

These values are not visable..
  0x0024,  // L1 - SD630_FASE_ANGLE [DEGREES]
  0x0026,  // L2 - SD630_FASE_ANGLE [DEGREES]
  0x0028,  // L3 - SD630_FASE_ANGLE [DEGREES]
  0x0042,  // TOTAL FASE ANGLE [DEGREES]
  0x0046,  // FREQUENCY [HZ]
  0x0048,  // IMPORT ACTIVE [W]
  0x004A,  // EXPORT ACTIVE [W]
  0x004C,  // IMPORT REACTIVE [VAR]
  0x004E,  // EXPORT REACTIVE [VAR]

Cheers altbau

arendst, 2019-12-01:
I'm afraid your connection to the SDM630 is not correct. There appears to be no communication at all.

With logging set to 4 you should see at least this:
```
12:08:21 CMD: weblog 4
12:08:21 SRC: WebConsole from 192.168.2.1
12:08:21 CMD: Group 0, Index 1, Command "WEBLOG", Data "4"
12:08:21 MQT: stat/sonoff3/RESULT = {"WebLog":4}
12:08:21 DMP: 01 04 04 3F B7 CE D9 D2 4C
12:08:21 CFG: Saved to flash at F6, Count 707, Bytes 4096
12:08:21 DMP: 01 04 04 00 00 00 00 FB 84
12:08:21 DMP: 01 04 04 00 00 00 00 FB 84
12:08:22 DMP: 01 04 04 41 3C E5 60 65 0C
12:08:22 DMP: 01 04 04 43 91 C9 75 28 5A
12:08:22 DMP: 01 04 04 43 60 B5 4B D8 B9
12:08:22 DMP: 01 04 04 3D 21 31 1A 32 79
12:08:23 DMP: 01 04 04 00 00 00 00 FB 84
12:08:23 DMP: 01 04 04 40 EA 6C FA 62 F3
12:08:23 DMP: 01 04 04 C0 DC C2 ED 97 53
12:08:23 DMP: 01 04 04 3E AC 1F D0 3F E1
12:08:24 DMP: 01 04 04 42 48 00 00 6F EA
12:08:24 DMP: 01 04 04 3F B7 CE D9 D2 4C
```
which represent the answers from the SDM630 when requesting to send each individual value.

The fact that screen only shows the default fields is a good sign there is no comms between the SDM630. To connect the SDM630 to an ESP8266 you will need a RS485 converter as documented here https://github.com/arendst/Tasmota/pull/2694#issuecomment-388309240

altbau9944, 2019-12-07:
Hi arendst,

First of all I would like to thank you for your support.

I think the communication works between the SDM630, via the RS485 adapter to the ESP8266.

Take a look at the logging:

> 09:05:23 DMP: 01 04 04 42 C6 FE 9D 8E 08
09:05:23 RSL: tele/tasmota/INFO1 = {"Module":"Generic","Version":"7.1.0(sensors)","FallbackTopic":"cmnd/DVES_174102_fb/","GroupTopic":"cmnd/tasmotas/"}
09:05:23 RSL: tele/tasmota/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-xxx","IPAddress":"xxx.xxx.xxx.xxx"}
09:05:23 RSL: tele/tasmota/INFO3 = {"RestartReason":"Software/System restart"}
09:05:23 DMP: 01 04 04 00 00 00 00 FB 84
09:05:23 NTP: Drift 0, (UTC) Fri Dec 06 08:05:23 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
09:05:23 QPC: Reset
09:05:23 DMP: 01 04 04 00 00 00 00 FB 84
09:05:23 HTP: Main Menu
09:05:23 DMP: 01 04 04 C1 9C 38 02 94 57
09:05:24 DMP: 01 04 04 00 00 00 00 FB 84
09:05:24 DMP: 01 04 04 00 00 00 00 FB 84
09:05:24 DMP: 01 04 04 3F 7B 64 98 AC E3
09:05:24 DMP: 01 04 04 3F 80 00 00 F6 78
09:05:25 DMP: 01 04 04 3F 80 00 00 F6 78
09:05:25 DMP: 01 04 04 42 1E F2 B0 CB 2E
09:05:25 APP: Boot Count 16
09:05:25 DMP: 01 04 04 43 65 7B 60 DC C7
09:05:25 CFG: Saved to flash at FA, Count 36, Bytes 4096
09:05:25 DMP: 01 04 04 00 00 00 00 FB 84
09:05:26 DMP: 01 04 04 00 00 00 00 FB 84
09:05:26 DMP: 01 04 04 3E D3 A0 E5 BE 1E
09:05:26 HTP: Console
09:05:26 DMP: 01 04 04 00 00 00 00 FB 84
09:05:26 DMP: 01 04 04 00 00 00 00 FB 84
09:05:27 DMP: 01 04 04 42 CB C9 E7 88 18
09:05:27 DMP: 01 04 04 00 00 00 00 FB 84
09:05:27 RSL: tele/tasmota/STATE = {"Time":"2019-12-06T09:05:27","Uptime":"0T00:00:12","UptimeSec":12,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":31,"MqttCount":0,"Wifi":{"AP":1,"SSId":"xxxxxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:05:27 RSL: tele/tasmota/SENSOR = {"Time":"2019-12-06T09:05:27","ENERGY":{"TotalStartTime":"2019-12-01T07:34:20","Total":0.011,"Yesterday":0.000,"Today":0.011,"Period":0,"Power":[95,0,0],"ApparentPower":[95,0,0],"ReactivePower":[-20,0,0],"Factor":[0.98,1.00,1.00],"Voltage":[229,0,0],"Current":[0.413,0.000,0.000]}}
09:05:27 DMP: 01 04 04 00 00 00 00 FB 84
09:05:27 DMP: 01 04 04 C1 95 DB 1F CD 6C
09:05:28 DMP: 01 04 04 00 00 00 00 FB 84
09:05:28 DMP: 01 04 04 00 00 00 00 FB 84
09:05:28 DMP: 01 04 04 3F 7B 44 02 35 48
09:05:28 DMP: 01 04 04 3F 80 00 00 F6 78
09:05:29 DMP: 01 04 04 3F 80 00 00 F6 78
09:05:29 DMP: 01 04 04 42 1E F2 B0 CB 2E
09:05:29 DMP: 01 04 04 43 65 C1 E8 AF C1
09:05:29 DMP: 01 04 04 00 00 00 00 FB 84
09:05:30 DMP: 01 04 04 00 00 00 00 FB 84
09:05:30 DMP: 01 04 04 3F 0D 3D 33 36 D6
09:05:30 DMP: 01 04 04 00 00 00 00 FB 84
09:05:30 DMP: 01 04 04 00 00 00 00 FB 84
09:05:31 DMP: 01 04 04 42 E5 A6 44 85 98
09:05:31 DMP: 01 04 04 00 00 00 00 FB 84
09:05:31 DMP: 01 04 04 00 00 00 00 FB 84
09:05:31 DMP: 01 04 04 C1 A3 E5 E9 BD 44
09:05:32 DMP: 01 04 04 00 00 00 00 FB 84
09:05:32 DMP: 01 04 04 00 00 00 00 FB 84
09:05:32 DMP: 01 04 04 3F 7B E9 55 08 26
09:05:32 DMP: 01 04 04 3F 80 00 00 F6 78
09:05:33 DMP: 01 04 04 3F 80 00 00 F6 78
09:05:33 DMP: 01 04 04 42 1E F2 B0 CB 2E
09:05:33 DMP: 01 04 04 43 65 DB A3 E4 96
09:05:33 DMP: 01 04 04 00 00 00 00 FB 84
09:05:34 DMP: 01 04 04 00 00 00 00 FB 84
09:05:34 DMP: 01 04 04 3E E2 B6 E4 20 71
09:05:34 DMP: 01 04 04 00 00 00 00 FB 84
09:05:34 DMP: 01 04 04 00 00 00 00 FB 84
09:05:35 DMP: 01 04 04 42 C9 AB 3D 81 23
09:05:35 DMP: 01 04 04 00 00 00 00 FB 84
09:05:35 DMP: 01 04 04 00 00 00 00 FB 84
09:05:35 DMP: 01 04 04 C1 9A 3E 7E 76 17
09:05:36 DMP: 01 04 04 00 00 00 00 FB 84
09:05:36 DMP: 01 04 04 00 00 00 00 FB 84
09:05:36 DMP: 01 04 04 3F 7B 8B 9A 61 12
09:05:36 DMP: 01 04 04 3F 80 00 00 F6 78
09:05:37 DMP: 01 04 04 3F 80 00 00 F6 78
09:05:37 DMP: 01 04 04 42 1E F2 B0 CB 2E
09:05:37 DMP: 01 04 04 43 65 FC 40 BE EF
09:05:37 DMP: 01 04 04 00 00 00 00 FB 84
09:05:38 DMP: 01 04 04 00 00 00 00 FB 84
09:05:38 DMP: 01 04 04 3E E4 F4 F2 71 1E

In general the needed values are not shown in the web-interface. As shown in my last picture.

By the way: I was able to add e.g. the Frequence in the Sonoff-Tasmota 6.6.0.11 3 month ago, but with the Tasmota 7.1.0 I have no success.

Nogmaals bedankt!!

altbau9944, 2019-12-14:
Good Morning,

How can I manage, or who can contect to get some support here. I've tried to solve this issue this morning again and with no success... 

I've adapted the xnrg_10_sdm630.ino file, as proposed by pablozg. But othing changed in the web UI...

@pablozg 
Can you support me??

[image]

altbau

panciunio, 2021-06-06:
Hi,

At first, thanks for great work for Tasmota OS.
 
I know that this topic is little bit old and some new features were added to **xnrg_10_sdm630.ino** code but I'm still looking for other measure parameters like 'Import Active' and 'Total Import'. I sow in code that a few variables are available but depends from SDM630_IMPORT define directive by default:

```
#ifdef SDM630_IMPORT
  0x015A,  //  +   +   +   kWh  Phase 1 import active energy
  0x015C,  //  +   +   +   kWh  Phase 2 import active energy
  0x015E,  //  +   +   +   kWh  Phase 3 import active energy
#endif  // SDM630_IMPORT
```
Is any serious reason why? I heard about Tasmota limitation regarding amount of cells but I don't believe...

Anybody know how to add missing variables to Tasmota-sensors?

seicento, 2022-05-10:
> #ifdef SDM630_IMPORT
>   0x015A,  //  +   +   +   kWh  Phase 1 import active energy
>   0x015C,  //  +   +   +   kWh  Phase 2 import active energy
>   0x015E,  //  +   +   +   kWh  Phase 3 import active energy
> #endif  // SDM630_IMPORT

@panciunio  Im strugling to get import kwh. I have tryed what you suggest but still no import?
Maybe you could post your entire file?

Thanks

funnyfrish, 2023-02-10:
#SDM630_IMPORT does not work. Im still getting the same view without SDM630_IMPORT....

arendst, 2023-02-11:
Energy Total equals SDM630 Import Active hence you won't see a difference.

panciunio, 2023-04-18:
Hi All,
I my country anybody saying "Better is the enemy of good one!". It's mean if someone working properly - don't touch it!

I decided to upgrade my all Tasmota's network and of course, my Wemos D1 Pro connected to SDM630 stopped provide an information regarding Import Energy at all. 

I started with official tasmota-sensors firmware and what I can see:

[image]

Okay... nothing happened... I had the same situation with Tasmota 8(9).x... 

I downloaded newest development source archive and checked   'xnrg_10_sdm630.ino' file which contains section:

`//#ifdef SDM630_IMPORT
  0x015A,  //  +   +   +   kWh  Phase 1 import active energy
  0x015C,  //  +   +   +   kWh  Phase 2 import active energy
  0x015E,  //  +   +   +   kWh  Phase 3 import active energy
//#endif  // SDM630_IMPORT`

I assume, that's ok... I checked cases into SDM630Every250ms():
`        case 19:
          Energy->import_active[0] = value;
          break;

        case 20:
          Energy->import_active[1] = value;
          break;

        case 21:
          Energy->import_active[2] = value;
          break;`

look's good as well...

Compiled, flashed... result? the same...
[image]

No Import Active visible...

I tried to dig deeper... found 'configurations.md' file into 'energy_modbus_config' branch. Very interesting file which contains JSON structure for each data from energy measurement device, I founf SDM630 section and discovered that ImportActive variable is hidden into "User" section:

`SDM630             {"Name":"SDM630","Baud":9600,"Config":"8N1","Address":1,"Function":4,"Voltage":[0,2,4],"Current":[6,8,10],"Power":[12,14,16],"ApparentPower":[18,20,22],"ReactivePower":[24,26,28],"Factor":[30,32,34],"Frequency":70,"Total":342,"ExportActive":[352,354,356],**"User":{"R":[346,348,350],"J":"ImportActive","G":"Import Active","U":"kWh","D":24}**}`

The question is... how to check data from table "User" under Tasmota console and how to provide these information to MQTT if these are available?

Best Regards,
panciunio

arendst, 2023-04-18:
SDM630 ImportActive =Energy Total.

If you want more registers you will need to start using the new user configurable modbus driver. See https://github.com/arendst/Tasmota/pull/18413#issuecomment-1510131402

This is also more future proof where dedicated modbus drivers will be removed someday.