arendst/Tasmota #6880: Configure Time under Configuration

une18, 2019-11-09:
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?** 
Time 14:24:17
Timezone +2
Worked these commands, but could not find in Wiki.
Would like to have it as other Configuration buttons instead of commands.
https://github.com/arendst/Tasmota/wiki/Commands#how-to-use-commands
[image]

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
It is difficult for many users to use commands. Would like to use them as less as possible

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Have a button and window for Time and Timezone setting
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-11-09:
The commands are documented in the wiki:
https://github.com/arendst/Tasmota/wiki/Commands#timezone
https://github.com/arendst/Tasmota/wiki/Commands#time

Adding UI dialogs consumes program memory that may be better used for new features or enhancements.

Jason2866, 2019-11-09:
The way you describe is loosing time after a power loss.
It is for users where no wifi network is available. 
If your device is connected to wifi, it will get the time from the NTP server.

What you suggest is not needed

closed 2019-11-09 by ascillato2