arendst/Tasmota #6882: Upgrade Decode Tools to Python 3

meingraham, 2019-11-09:
@curzon01 

Norbert,

The decode tools do not run under Windows 10 due to being unable to install the needed supporting packages (configargparse, pycurl, etc.). The installation throws warnings regarding deprecation of Python 2.7 commencing in 2020. The installation of pycurl fails altogether. I'm unsure if using Python 3 would resolve this issue... but the Python upgrade is probably soon requisite.

Mike

P.S. Apologies for making a request rather than performing the task... but my Python skills are bad at worse and knowing the necessary Python2 vs. Python3 syntax changes is a bridge too far :wink:

blakadder, 2019-11-09:
new docs explaining the Win 10 workaround are PRd #6884

curzon01, 2019-11-09:
Will have a look but currently I'm busy and also offline (not really but development related) so it have to wait 2+ weeks from my side.

ascillato2, 2019-11-09:
Closing this request has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-09 by ascillato2