arendst/Tasmota #6883: Add Command to change WEB_USERNAME

nightcat91, 2019-11-09:
**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  
No

**Describe the solution you'd like**  
I would like to be able to change the WEB_USERNAME somehow after flashing a precompiled binary.

So far I have mostly compiled my binaries myself with hardcoded IPs and lots of other stuff, but I am starting to use more and more the precompiled binaries. But now I cannot change the WEB_USERNAME.
So now my selfcompiled devices have a different username than the devices using precompiled binaries. And its hard to remember which device has which version (I have over 30 devices with tasmota)

Would it be possible to just have this as an optional command so that it can be used in the console or via MQTT?

IP Settings (static, dhcp) and password can be changed, why not the username?

Thanks

andrethomas, 2019-11-09:
> And its hard to remember which device has which version

Have a look at https://github.com/jziolkowski/tdm

nightcat91, 2019-11-09:
> > And its hard to remember which device has which version
> 
> Have a look at https://github.com/jziolkowski/tdm

Thats very nice, but requires a few things to install.
And I am away from home quiet often (due to working), and then I just use the tasmota web site over VPN to check on a few settings when needed...

Jason2866, 2019-11-09:
What is wrong using the same WEB_USERNAME for all devices?

nightcat91, 2019-11-09:
Like I said above, I have lots of devices with self compiled binaries where I changed the WEB_USERNAME in the my_userconfig.h before upload. 
Now I am adding more and more devices with pre compied binaries (mostly over tuya convert) and dont make special self compiled binaries anymore for the new devices...
So I have different WEB_USERNAME on different devices.
And I cant change them, not on my precompiled devices and not on the new ones

Jason2866, 2019-11-09:
I added info, didnt wrote what i had in mind

ascillato2, 2019-11-09:
Duplicated issue:

https://github.com/arendst/Tasmota/issues/2554

https://github.com/arendst/Tasmota/issues/2798#issuecomment-390699227

nightcat91, 2019-11-09:
Since the Yeti app supports username and password according to the comments, there should be no problem on that side. 
Both of the issues are over a year old (I did find them while searching), but maybe things have changed until then?

ascillato2, 2019-11-09:
Yeti App reached EOL and it is not available anymore :(

ascillato2, 2019-11-12:
Closing this issue as no one is working on this. Sorry.

Thanks for sharing your ideas.

closed 2019-11-12 by ascillato2