arendst/Tasmota #6887: PWM on more than one channel

Emmpunkt, 2019-11-09:
I´m trying to setup a Wemos Mini with two PWM-Outputs. Two different Lamps.
The problem is, that Lamp No2 only works if No1 is switched on.
If i switch No1 off, No2 is also switched off. (In the GUI still on)

My setup:
Wemos D1 Mini
Tasmota 7.0.0.4
Type: Generic (18)
D2 GPIO4 PWM2 (38)
D1 GPIO5 PWM1 (37)
SetOption68 1

Did i something wrong or missunderstood this function?
Regards M.

ascillato2, 2019-11-09:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-09 by ascillato2

s-hadinger, 2019-11-09:
This is a confirmed bug that appeared between 6.7.1 and 7.0.2. We are investigating.

PR #6894 (2019-11-10): Fix bad PWM Power control with SetOption68
## Description:

**Related issue (if applicable):** fixes #6887 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core pre-2.6
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).