arendst/Tasmota #6888: Rule execution break seems not to behave as documented.

Hello1024, 2019-11-09:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
`break` command in rules doesn't behave as documented if two or more matching triggers follow the break clause.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Dual R2
- [X] Tasmota binary firmware version number used: 7.0.4
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: _OTA____
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
20:17:12 MQT: stat/heating/RESULT = {"NAME":"Generic","GPIO":[4,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:17:13 MQT: stat/heating/RESULT = {"Module":"39 (Sonoff Dual R2)"}
20:17:13 MQT: stat/heating/RESULT = {"GPIO0":"4 (DS18x20)","GPIO1":"0 (None)","GPIO3":"0 (None)","GPIO9":"0 (None)"}
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
20:18:02 MQT: stat/heating/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":301,"Rules":"ON DS18B20-2#Temperature>90 do publish hi hi BREAK    ON DS18B20-2#Temperature>10 do publish ha ha ENDON   ON DS18B20-2#Temperature>10 do publish hb hb ENDON   ON DS18B20-2#Temperature>10 do publish hc hc ENDON"}
20:18:02 MQT: stat/heating/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
20:18:02 MQT: stat/heating/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
20:20:33 MQT: stat/heating/STATUS = {"Status":{"Module":39,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"heating","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:20:33 MQT: stat/heating/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/arc/6.6.0/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:59","StartupUTC":"2019-11-09T19:19:34","Sleep":50,"CfgHolder":4617,"BootCount":34,"SaveCount":369,"SaveAddress":"F7000"}}
20:20:33 MQT: stat/heating/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-07-06T13:35:03","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
20:20:33 MQT: stat/heating/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["eightlegs",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000000"]}}
20:20:33 MQT: stat/heating/STATUS4 = {"StatusMEM":{"ProgramSize":523,"Free":480,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
20:20:33 MQT: stat/heating/STATUS5 = {"StatusNET":{"Hostname":"heating","IPAddress":"192.168.86.226","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.86.1","Mac":"C4:4F:33:A2:91:57","Webserver":2,"WifiConfig":2}}
20:20:33 MQT: stat/heating/STATUS6 = {"StatusMQT":{"MqttHost":"simon-pi.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A29157","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:20:34 MQT: stat/heating/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 09 19:20:33 2019","Local":"Sat Nov 09 20:20:33 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:49","Sunset":"17:18"}}
20:20:34 MQT: stat/heating/STATUS10 = {"StatusSNS":{"Time":"2019-11-09T20:20:34","DS18B20-1":{"Id":"051692B925FF","Temperature":56.8},"DS18B20-2":{"Id":"051692B936FF","Temperature":21.2},"TempUnit":"C"}}
20:20:34 MQT: stat/heating/STATUS11 = {"StatusSTS":{"Time":"2019-11-09T20:20:34","Uptime":"0T00:01:00","Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"eightlegs","BSSId":"08:9E:08:FC:E0:79","Channel":6,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:06"}}}
20
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
20:22:10 RUL: DS18B20-2#TEMPERATURE>10 performs "publish ha ha"
20:22:10 MQT: ha = ha
20:22:12 RUL: DS18B20-2#TEMPERATURE>10 performs "publish ha ha"
20:22:12 MQT: ha = ha
20:22:14 RUL: DS18B20-2#TEMPERATURE>10 performs "publish ha ha"
20:22:14 MQT: ha = ha
  ```

### TO REPRODUCE
Add and enable this rule
```
Rule1
  ON DS18B20-2#Temperature>90 do publish hi hi BREAK 
  ON DS18B20-2#Temperature>10 do publish ha ha ENDON
  ON DS18B20-2#Temperature>10 do publish hb hb ENDON
  ON DS18B20-2#Temperature>10 do publish hc hc ENDON
```

### EXPECTED BEHAVIOUR
Expected behaviour (python syntax):
```
If DS18B20-2#Temperature>90:
  publish hi hi

IF DS18B20-2#Temperature>10 && DS18B20-2#Temperature<=90:
  publish ha ha
  publish hb hb
  publish hc hc

IF DS18B20-2#Temperature<=10:
  do nothing
```

### ACTUAL BEHAVIOUR
```
If DS18B20-2#Temperature>90:
  publish hi hi

IF DS18B20-2#Temperature>10 && DS18B20-2#Temperature<=90:
  publish ha ha

IF DS18B20-2#Temperature<=10:
  do nothing
```

### ADDITIONAL CONTEXT
The cause seems to be [here](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_10_rules.ino#L428).  Specifically, the stop_all_rules flag should only be set *if* the rule being processed has a matching trigger.    Ie. the setting of the flag needs to be moved down inside `if (RulesRuleMatch(...) {`

ascillato, 2019-11-09:
Hi,

> Expected behaviour (python syntax):

```
If DS18B20-2#Temperature>90:
  publish hi hi

IF DS18B20-2#Temperature>10 && DS18B20-2#Temperature<=90:
  publish ha ha
  publish hb hb
  publish hc hc

IF DS18B20-2#Temperature<=10:
  do nothing
```

So, for that, your rules syntax should be:

```
Rule1
  ON DS18B20-2#Temperature>90 do publish hi hi BREAK 
  ON DS18B20-2#Temperature>10 do backlog publish ha ha; publish hb hb; publish hc hc ENDON
```

ascillato2, 2019-11-09:
If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-09 by ascillato2

ascillato, 2019-11-09:
From Discord Chat:

if you need this:
```
Power1 = Sensor1<17
Power2 = (Sensor1<17 OR Sensor2<45)
```

the rules should be:

```
rule1
ON DS18B20-1#Temperature<17 do backlog power1 1; power2 1 break
ON DS18B20-2#Temperature<45 do power2 1 break
ON DS18B20-2#Temperature>=45 do power2 0 endon
ON DS18B20-1#Temperature>=17 do power1 0 endon
```

ascillato, 2019-11-09:
So,

* If you want this behaviour:
```
Power1 = Sensor1<17
Power2 = (Sensor1<17 OR Sensor2<45)
```

* Then, we can test the following cases in numbers:

```
A- if Sensor1 is 10, output should be POWER1 1 and POWER2 1
B- if Sensor1 is 20, output should be POWER1 0
C- if Sensor2 is 30, output should be POWER2 1
D- if Sensor2 is 50 AND Sensor1 is 20, output should be POWER2 0
```

* So, proposing the following test rules (var instead of sensor just for testing):

```
Rule1
  on var1#state<17 do backlog power1 1; power2 1 break
  on var1#state>=17 do power1 0 break
  on var2#state<45 do power2 1 endon
  on var2#state>=45 do event testsensor1=%var1% endon
  on event#testsensor1>=17 do power2 0 endon

```

* Rules all together for pasting in the console:

```
rule1 1
rule1 on var1#state<17 do backlog power1 1; power2 1 break on var1#state>=17 do power1 0 break on var2#state<45 do power2 1 endon on var2#state>=45 do event testsensor1=%var1% endon on event#testsensor1>=17 do power2 0 endon
```

* Test Cases and the output of the console:

```
18:26:06 CMD: rule1 on var1#state<17 do backlog power1 1; power2 1 break on var1#state>=17 do power1 0 break on var2#state<45 do power2 1 endon on var2#state>=45 do event testsensor1=%var1% endon on event#testsensor1>=17 do power2 0 endon
18:26:06 MQT: stat/living/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":293,"Rules":"on var1#state<17 do backlog power1 1; power2 1 break on var1#state>=17 do power1 0 break on var2#state<45 do power2 1 endon on var2#state>=45 do event testsensor1=%var1% endon on event#testsensor1>=17 do power2 0 endon"}

-CASE A-

18:28:07 CMD: var1 10
18:28:07 MQT: stat/living/RESULT = {"Var1":"10"}
18:28:07 RUL: VAR1#STATE<17 performs "backlog power1 1; power2 1"
18:28:07 MQT: stat/living/RESULT = {"POWER1":"ON"}
18:28:07 MQT: stat/living/POWER1 = ON
18:28:08 MQT: stat/living/RESULT = {"POWER2":"ON"}
18:28:08 MQT: stat/living/POWER2 = ON

-CASE B-

18:28:53 CMD: var1 20
18:28:53 MQT: stat/living/RESULT = {"Var1":"20"}
18:28:53 RUL: VAR1#STATE>=17 performs "power1 0"
18:28:53 MQT: stat/living/RESULT = {"POWER1":"OFF"}
18:28:53 MQT: stat/living/POWER1 = OFF

-CASE C-

18:29:49 CMD: var2 30
18:29:49 MQT: stat/living/RESULT = {"Var2":"30"}
18:29:49 RUL: VAR2#STATE<45 performs "power2 1"
18:29:49 MQT: stat/living/RESULT = {"POWER2":"ON"}
18:29:49 MQT: stat/living/POWER2 = ON

-CASE D-

18:31:00 CMD: var2 50
18:31:00 MQT: stat/living/RESULT = {"Var2":"50"}
18:31:00 RUL: VAR2#STATE>=45 performs "backlog event testsensor1=20"
18:31:00 MQT: stat/living/RESULT = {"Event":"Done"}
18:31:00 RUL: EVENT#TESTSENSOR1>=17 performs "power2 0"
18:31:00 MQT: stat/living/RESULT = {"POWER2":"OFF"}
18:31:00 MQT: stat/living/POWER2 = OFF
```

So, with this example, your desired behaviour can be achieved without modifying the actual code.

Then, if you need help on using the actual sensor value instead of VAR, you can ask here or in the chat if you want.

Hope this helps.