arendst/Tasmota #6890: DeepSleep mode: no I2C data send to MQTT

guiumb, 2019-11-09:
### BUG DESCRIPTION
Using DeepSleep Mode on MiniD1 borad with I2C sensor no MQTT sensor data was send at wake-up.

I have a Mini D1 board with I2C temperature sensor connected. When I setup my device in order to use deepsleep function (in this test I have used 30 secs) the board goes to sleep, but at wake-up no MQTT sensor data was send.

The system send only INFO1-2-3 MQTT messages but no sensor data was sent.

When I disconnect GPIO16 and I stop deepsleep function I found the GPIO configuration resetted with default values (No config about I2C sensor).

I have already tryed to flash device from scratch and setup device with dynamic sleep set to 0 but the results is the same.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Mini D1 Board
- [x] Tasmota binary firmware version number used: 7.0.0.4 with core 2.6
  - [x] Self-compiled
    - [x] IDE / Compiler used: Arduino IDE 1.8.10
- [ ] Flashing tools used: _____
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:

```
22:35:38 CMD: Backlog Template; Module; GPIO
22:35:38 SRC: WebConsole from 10.0.0.6
22:35:38 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO"
22:35:38 SRC: Backlog
22:35:38 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
22:35:38 MQT: stat/bme_temp_out/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:35:39 SRC: Backlog
22:35:39 CMD: Group 0, Index 1, Command "MODULE", Data ""
22:35:39 MQT: stat/bme_temp_out/RESULT = {"Module":{"18":"Generic"}}
22:35:39 SRC: Backlog
22:35:39 CMD: Group 0, Index 1, Command "GPIO", Data ""
```
 - [x] Provide the output of this command: ``Status 0``:

```
22:37:44 CMD: Status 0
22:37:44 CMD: Group 0, Index 1, Command "STATUS", Data "0"
22:37:44 MQT: stat/bme_temp_out/STATUS = {"Status":{"Module":18,"FriendlyName":["Temperatura Esterna"],"Topic":"bme_temp_out","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:37:44 MQT: stat/bme_temp_out/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:02:50","StartupUTC":"2019-11-09T21:34:54","Sleep":0,"CfgHolder":4617,"BootCount":35,"SaveCount":68,"SaveAddress":"3F8000"}}
22:37:44 MQT: stat/bme_temp_out/STATUS2 = {"StatusFWR":{"Version":"7.0.0.4(tasmota)","BuildDateTime":"2019-11-09T21:37:03","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(5ab15d1)"}}
22:37:44 MQT: stat/bme_temp_out/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["guilan",""],"TelePeriod":10,"Resolution":"A98180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000400","00000000"]}}
22:37:44 MQT: stat/bme_temp_out/STATUS4 = {"StatusMEM":{"ProgramSize":479,"Free":3596,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashMode":2,"Features":["00000809","8FC0A196","00000000","00000405","000000C0","00000800"],"Drivers":"1,2,3,4,8,9,10,12,29","Sensors":"1,2,3,9"}}
22:37:44 MQT: stat/bme_temp_out/STATUS5 = {"StatusNET":{"Hostname":"sensore24","IPAddress":"10.0.0.174","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"84:F3:EB:ED:63:A3","Webserver":2,"WifiConfig":4}}
22:37:44 MQT: stat/bme_temp_out/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.150","MqttPort":1883,"MqttClientMask":"bme_temp_out","MqttClient":"bme_temp_out","MqttUser":"umbyha","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:37:44 MQT: stat/bme_temp_out/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 09 21:37:44 2019","Local":"Sat Nov 09 22:37:44 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:14","Sunset":"17:00"}}
22:37:44 MQT: stat/bme_temp_out/STATUS10 = {"StatusSNS":{"Time":"2019-11-09T22:37:44","ANALOG":{"A0":6},"BMP280":{"Temperature":25.30,"Pressure":960.07},"PressureUnit":"hPa","TempUnit":"C"}}
22:37:44 MQT: stat/bme_temp_out/STATUS11 = {"StatusSTS":{"Time":"2019-11-09T22:37:44","Uptime":"0T00:02:50","UptimeSec":170,"Heap":23,"SleepMode":"Normal","Sleep":0,"LoadAvg":999,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:10"}}}
```

- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_

```
00:00:00 CFG: Loaded from flash at 3F9, Count 51
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project tasmota Temperatura Esterna Version 7.0.0.4(tasmota)-STAGE
00:00:00 I2C: BMP280 found at 0x76
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 guilan in mode 11N as sensore24...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
[... 150 lines cut ...]
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sensore24 with IP address 10.0.0.174
00:00:05 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/bme_temp_out/LWT = Online (retained)
00:00:06 MQT: cmnd/bme_temp_out/POWER = 
00:00:06 MQT: Subscribe to cmnd/bme_temp_out/#
00:00:06 MQT: Subscribe to cmnd/tasmotas/#
00:00:06 MQT: Subscribe to cmnd/bme_temp_out_fb/#
00:00:06 MQT: tele/bme_temp_out/INFO1 = {"Module":"Generic","Version":"7.0.0.4(tasmota)","FallbackTopic":"cmnd/bme_temp_out_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:06 MQT: tele/bme_temp_out/INFO2 = {"WebServerMode":"Admin","Hostname":"sensore24","IPAddress":"10.0.0.174"}
00:00:06 MQT: tele/bme_temp_out/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
00:00:06 QPC: Reset
00:00:08 APP: Boot Count 29
00:00:09 CFG: Saved to flash at 3F9, Count 59, Bytes 4096
```
  ```
There are 3 screenshot:
config.jpg shows GPIO configuration at setup
sensor.jpg shows "normal function" status device
after_reboot.jpg shows GPIO config after deepsleep funtion was disable.

![after_reboot](https://user-images.githubusercontent.com/42201341/68535651-e9e5ce00-0345-11ea-9acc-b13ad619808d.jpg)
![sensor](https://user-images.githubusercontent.com/42201341/68535652-ea7e6480-0345-11ea-9986-9e2a1b8d9062.jpg)
![config](https://user-images.githubusercontent.com/42201341/68535654-ea7e6480-0345-11ea-8be0-bea3943a6954.jpg)

arendst, 2019-11-10:
Verified. Investigating

arendst, 2019-11-10:
Ah! Nice one too. As the deepsleep wake-up time is sometimes shorter than the fast boot cycle time detection it will reset some or all settings resulting in your and mine issue.

As a workaround change the fast boot cycle detection to off using command ``SetOption36 0``.

guiumb, 2019-11-10:
Great ! 
Let me know if I can provide you traces for help in fixing the issue.

Thanks in advance

arendst, 2019-11-10:
Try latest commit. It woks for me.

guiumb, 2019-11-10:
I have falshed from scratch and I have set up SetOption36 to 0 as you suggested.
Immidiatly when I wrote in console DeepSleepTime 30 device shutdown. After 1 min wakeup and sent just onlu INFOS1-2-3 and shutdown again. After that no wakeup action was completed: device remain inactive (no ping and no web response) but blue led on board flash every 30 secs.
I noticed the fixed label: should I download new source code compile it?

arendst, 2019-11-10:
As said. Try latest commit. And yes that means download and recompile.

guiumb, 2019-11-10:
I am sorry for misunderstanding !
I will do immediatly I will keep you updated.

guiumb, 2019-11-10:
I have some problem with my compiler. As soon as I fix it I will provide a feedback.
Many thanks

guiumb, 2019-11-10:
OK. Finally I have solved problem with my compiler !

I have compiled last release, flash from scratch, setoption36 parameter set to 0 and set deepsleep 45.
The device blink led every 45 second but no wakeup is executed: no way to ping it and web interface is down.
I have tyed to recompile and flash again from scratch but the result was the same.

arendst, 2019-11-10:
Make sure a wire is connected between gpio16 and rst

The setoption36 change is not needed anymore. Just leave it set to 1

guiumb, 2019-11-10:
Ok. Connection D0 - RST checked with tester.
Flash from scratch again. No setoption36 modified:

```
00:00:00 CFG: Loaded from flash at 3F4, Count 8
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project tasmota Temperatura Esterna Version 7.0.0.4(tasmota)-STAGE
00:00:00 I2C: BMP280 found at 0x76
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 guilan in mode 11N as bme_temp_out-0931...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:07 RSL: tele/bme_temp_out/STATE = {"Time":"1970-01-01T00:00:07","Uptime":"0T00:00:09","UptimeSec":9,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"FF:FF:FF:FF:FF:FF","Channel":14,"RSSI":100,"LinkCount":0,"Downtime":"0T00:00:00"}}
00:00:07 RSL: tele/bme_temp_out/SENSOR = {"Time":"1970-01-01T00:00:07","ANALOG":{"A0":6},"BMP280":{"Temperature":25.74,"Pressure":964.82},"PressureUnit":"hPa","TempUnit":"C"}
00:00:08 WIF: Checking connection...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on bme_temp_out-0931 with IP address 10.0.0.174
00:00:08 APP: Boot Count 3
00:00:08 CFG: Saved to flash at 3FB, Count 9, Bytes 4096
22:22:19 MQT: Attempting connection...
22:22:19 MQT: Connected
22:22:19 MQT: tele/bme_temp_out/LWT = Online (retained)
22:22:19 MQT: cmnd/bme_temp_out/POWER = 
22:22:19 MQT: Subscribe to cmnd/bme_temp_out/#
22:22:19 MQT: Subscribe to cmnd/tasmotas/#
22:22:19 MQT: Subscribe to cmnd/bme_temp_out_fb/#
22:22:19 MQT: tele/bme_temp_out/INFO1 = {"Module":"Generic","Version":"7.0.0.4(tasmota)","FallbackTopic":"cmnd/bme_temp_out_fb/","GroupTopic":"cmnd/tasmotas/"}
22:22:19 MQT: tele/bme_temp_out/INFO2 = {"WebServerMode":"Admin","Hostname":"bme_temp_out-0931","IPAddress":"10.0.0.174"}
22:22:19 MQT: tele/bme_temp_out/INFO3 = {"RestartReason":"Software/System restart"}
22:22:19 NTP: Drift 0, (UTC) Sun Nov 10 21:22:19 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
22:22:20 HTP: Main Menu
22:22:27 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-10T22:22:27","Uptime":"0T00:00:19","UptimeSec":19,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":42,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:09"}}
22:22:27 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-10T22:22:27","ANALOG":{"A0":6},"BMP280":{"Temperature":25.74,"Pressure":964.80},"PressureUnit":"hPa","TempUnit":"C"}
22:22:37 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-10T22:22:37","Uptime":"0T00:00:29","UptimeSec":29,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":44,"LinkCount":1,"Downtime":"0T00:00:09"}}
22:22:37 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-10T22:22:37","ANALOG":{"A0":6},"BMP280":{"Temperature":25.73,"Pressure":964.81},"PressureUnit":"hPa","TempUnit":"C"}

..........

22:24:02 WIF: Checking connection...
22:24:02 WIF: Connected
22:24:07 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-10T22:24:07","Uptime":"0T00:01:59","UptimeSec":119,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":61,"MqttCount":2,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":50,"LinkCount":2,"Downtime":"0T00:00:13"}}
22:24:07 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-10T22:24:07","ANALOG":{"A0":6},"BMP280":{"Temperature":25.58,"Pressure":964.86},"PressureUnit":"hPa","TempUnit":"C"}
22:24:13 CMD: deepsleeptime 45
22:24:13 SRC: WebConsole from 10.0.0.6
22:24:13 CMD: Group 0, Index 1, Command "DEEPSLEEPTIME", Data "45"
22:24:13 MQT: stat/bme_temp_out/RESULT = {"DeepSleepTime":45}
```
Led blinks but no wakeup. I have reset board with button and the result was:

```
00:00:00 CFG: Loaded from flash at 3FA, Count 10
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project tasmota Temperatura Esterna Version 7.0.0.4(tasmota)-STAGE
00:00:00 I2C: BMP280 found at 0x76
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 guilan in mode 11N as bme_temp_out-0931...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on bme_temp_out-0931 with IP address 10.0.0.174
00:00:06 QPC: Reset
22:29:14 MQT: Attempting connection...
22:29:15 MQT: Connected
22:29:15 MQT: tele/bme_temp_out/LWT = Online (retained)
22:29:15 MQT: cmnd/bme_temp_out/POWER = 
22:29:15 MQT: Subscribe to cmnd/bme_temp_out/#
22:29:15 MQT: Subscribe to cmnd/tasmotas/#
22:29:15 MQT: Subscribe to cmnd/bme_temp_out_fb/#
22:29:15 MQT: tele/bme_temp_out/INFO1 = {"Module":"Generic","Version":"7.0.0.4(tasmota)","FallbackTopic":"cmnd/bme_temp_out_fb/","GroupTopic":"cmnd/tasmotas/"}
22:29:15 MQT: tele/bme_temp_out/INFO2 = {"WebServerMode":"Admin","Hostname":"bme_temp_out-0931","IPAddress":"10.0.0.174"}
22:29:15 MQT: tele/bme_temp_out/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
22:29:15 NTP: Drift 0, (UTC) Sun Nov 10 21:29:15 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
22:29:15 APP: Boot Count 4
22:29:16 CFG: Saved to flash at 3F9, Count 11, Bytes 4096
```
The boot counter was incremented but no sensor MQTT data are present.

arendst, 2019-11-10:
Could you provide the output of command status 2

guiumb, 2019-11-10:
With RST diconnect from GPIO16 and device always on:

`22:46:45 MQT: stat/bme_temp_out/STATUS2 = {"StatusFWR":{"Version":"7.0.0.4(tasmota)","BuildDateTime":"2019-11-10T22:19:39","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(5ab15d1)"}}`

arendst, 2019-11-10:
Ok. I expect an sdk issue. I see you have a sdk commit of 5ab15d1. That is probably the latest stage version.

Pls recompile with option core_pre as that was the one I was using too.

I'm offline now and will investigate tomorrow.

arendst, 2019-11-11:
Using a wemos it all works perfectly for me:
```
10:05:17 CMD: deepsleeptime 0
10:05:17 SRC: Serial
10:05:17 CMD: Group 0, Index 1, Command "DEEPSLEEPTIME", Data "0"
10:05:17 MQT: stat/wemos4/DEEPSLEEPTIME = {"DeepSleepTime":0}
10:05:17 MQT: tele/wemos4/STATE = {"Time":"2019-11-11T10:05:17","Uptime":"0T16:13:41","UptimeSec":58421,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:09"}}
10:05:17 MQT: tele/wemos4/SENSOR = {"Time":"2019-11-11T10:05:17","DS18B20":{"Id":"0000048EC44C","Temperature":20.4},"BME680":{"Temperature":21.5,"Humidity":54.9,"Pressure":1006.4,"Gas":488.29},"PressureUnit":"hPa","TempUnit":"C"}
10:05:25 CMD: deepsleeptime 45
10:05:25 SRC: Serial
10:05:25 CMD: Group 0, Index 1, Command "DEEPSLEEPTIME", Data "45"
10:05:25 MQT: stat/wemos4/DEEPSLEEPTIME = {"DeepSleepTime":45}
10:05:25 MQT: tele/wemos4/STATE = {"Time":"2019-11-11T10:05:25","Uptime":"0T16:13:49","UptimeSec":58429,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:09"}}
10:05:25 MQT: tele/wemos4/SENSOR = {"Time":"2019-11-11T10:05:25","DS18B20":{"Id":"0000048EC44C","Temperature":20.4},"BME680":{"Temperature":21.5,"Humidity":54.9,"Pressure":1006.4,"Gas":488.64},"PressureUnit":"hPa","TempUnit":"C"}
10:05:25 CFG: Saved to flash at F6, Count 351, Bytes 4096
10:05:25 DSL: Reset wrong settings wakeup: 0, slip 9000
10:05:25 MQT: stat/wemos4/STATUS1 = {"DeepSleep":{"Time":"2019-11-11T10:06:00","Epoch":1573463160}}
10:05:25 MQT: tele/wemos4/LWT = Offline (retained)
........
00:00:00 CFG: Loaded from flash at F6, Count 351
00:00:00 SRC: Restart
00:00:00 Project tasmota Wemos4 Version 7.0.0.4(tasmota)-STAGE
00:00:00 DSB: Sensors found 1
00:00:00 I2C: BME680 found at 0x77
00:00:00 CFG: Saved to flash at F5, Count 352, Bytes 4096
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP1, SSId indebuurt1, Channel 1, BSSId 24:D3:F2:97:C0:A1, RSSI -74, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt2, Channel 5, BSSId A0:AB:1B:7D:42:AC, RSSI -38, Encryption 1
00:00:04 WIF: Network 2, AP-, SSId indebuurt3, Channel 12, BSSId 60:E3:27:58:77:E6, RSSI -86, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt2 in mode 11N as wemos4...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on wemos4 with IP address 192.168.2.223
10:06:08 NTP: Drift 0, (UTC) Mon Nov 11 09:06:08 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
10:06:08 APP: Boot Count 198
10:06:08 CFG: Saved to flash at F4, Count 353, Bytes 4096
10:06:09 MQT: Attempting connection...
10:06:09 MQT: Connected
10:06:09 MQT: tele/wemos4/LWT = Online (retained)
10:06:09 MQT: cmnd/wemos4/POWER = 
10:06:09 MQT: Subscribe to cmnd/wemos4/#
10:06:09 MQT: Subscribe to cmnd/sonoffs/#
10:06:09 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
10:06:09 MQT: tele/wemos4/INFO1 = {"Module":"Generic","Version":"7.0.0.4(tasmota)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/sonoffs/"}
10:06:09 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
10:06:09 MQT: tele/wemos4/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
10:06:17 MQT: tele/wemos4/STATE = {"Time":"2019-11-11T10:06:17","Uptime":"0T00:00:19","UptimeSec":19,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:09"}}
10:06:17 MQT: tele/wemos4/SENSOR = {"Time":"2019-11-11T10:06:17","DS18B20":{"Id":"0000048EC44C","Temperature":20.4},"BME680":{"Temperature":21.3,"Humidity":56.1,"Pressure":1006.4,"Gas":423.84},"PressureUnit":"hPa","TempUnit":"C"}
10:06:17 MQT: stat/wemos4/STATUS1 = {"DeepSleep":{"Time":"2019-11-11T10:06:45","Epoch":1573463205}}
10:06:17 MQT: tele/wemos4/LWT = Offline (retained)
```
- I'm wondering what happens after 22:24:13 where I would expect to see teleperiod data and the start of the sleep cycle.
- I ruled out the SDK version. Using your version (NONOSDK22x_191024 = 2.2.2-dev(5ab15d1)) it still works fine with me.
- Make sure you're using the latest dev version. Especially file xdrv_29_deepsleep.ino from 20191110 around noon.

guiumb, 2019-11-11:
Ok. I am out of home.
As soon as I come back I will verify.
Thanks

guiumb, 2019-11-11:
I am frustrated !
No way to have deepsleep: 
I have downloaded last source commit (at 08:00 pm today), I have compiled with 2.6.0 core file dowloaded from [https://github.com/esp8266/Arduino/releases/tag/2.6.0](https://github.com/esp8266/Arduino/releases/tag/2.6.0)
I have used new Mini D1 board and this is the result:
```
00:00:00 CFG: Loaded from flash at 3F5, Count 7
00:00:00 Project tasmota Temperatura Esterna Version 7.0.0.4(tasmota)-STAGE
00:00:00 I2C: BMP280 found at 0x76
00:00:00 WIF: Connecting to AP1 guilan in mode 11N as bme_temp_out-3130...
00:00:07 RSL: tele/bme_temp_out/STATE = {"Time":"1970-01-01T00:00:07","Uptime":"0T00:00:09","UptimeSec":9,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"FF:FF:FF:FF:FF:FF","Channel":14,"RSSI":100,"LinkCount":0,"Downtime":"0T00:00:00"}}
00:00:07 RSL: tele/bme_temp_out/SENSOR = {"Time":"1970-01-01T00:00:07","ANALOG":{"A0":1},"BMP280":{"Temperature":25.82,"Pressure":960.47},"PressureUnit":"hPa","TempUnit":"C"}
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on bme_temp_out-3130 with IP address 10.0.0.174
21:36:40 MQT: Attempting connection...
21:36:41 MQT: Connected
21:36:41 MQT: tele/bme_temp_out/LWT = Online (retained)
21:36:41 MQT: cmnd/bme_temp_out/POWER = 
21:36:41 MQT: tele/bme_temp_out/INFO1 = {"Module":"Generic","Version":"7.0.0.4(tasmota)","FallbackTopic":"cmnd/bme_temp_out_fb/","GroupTopic":"cmnd/tasmotas/"}
21:36:41 MQT: tele/bme_temp_out/INFO2 = {"WebServerMode":"Admin","Hostname":"bme_temp_out-3130","IPAddress":"10.0.0.174"}
21:36:41 MQT: tele/bme_temp_out/INFO3 = {"RestartReason":"External System"}
21:36:48 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T21:36:48","Uptime":"0T00:00:19","UptimeSec":19,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":41,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":46,"LinkCount":1,"Downtime":"0T00:00:09"}}
21:36:48 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T21:36:48","ANALOG":{"A0":1},"BMP280":{"Temperature":25.91,"Pressure":960.44},"PressureUnit":"hPa","TempUnit":"C"}
21:36:58 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T21:36:58","Uptime":"0T00:00:29","UptimeSec":29,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:09"}}
21:36:58 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T21:36:58","ANALOG":{"A0":1},"BMP280":{"Temperature":26.00,"Pressure":960.42},"PressureUnit":"hPa","TempUnit":"C"}
21:37:02 LOG: SerialLog 0, WebLog 4, MqttLog 0, SysLog 0, LogHost , LogPort 514, TelePeriod 10
21:37:02 HTP: Configuration
21:37:03 CFG: Saved to flash at 3FB, Count 9, Bytes 4096
21:37:05 HTP: Main Menu
21:37:08 HTP: Console
21:37:08 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T21:37:08","Uptime":"0T00:00:39","UptimeSec":39,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":61,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:09"}}
21:37:08 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T21:37:08","ANALOG":{"A0":1},"BMP280":{"Temperature":25.99,"Pressure":960.42},"PressureUnit":"hPa","TempUnit":"C"}
21:37:14 CMD: status 2
21:37:14 SRC: WebConsole from 10.0.0.6
21:37:14 CMD: Group 0, Index 1, Command "STATUS", Data "2"
21:37:14 MQT: stat/bme_temp_out/STATUS2 = {"StatusFWR":{"Version":"7.0.0.4(tasmota)","BuildDateTime":"2019-11-11T21:32:30","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(5ab15d1)"}}
21:37:18 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T21:37:18","Uptime":"0T00:00:49","UptimeSec":49,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:09"}}
21:37:18 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T21:37:18","ANALOG":{"A0":1},"BMP280":{"Temperature":26.00,"Pressure":960.46},"PressureUnit":"hPa","TempUnit":"C"}
21:37:21 WIF: Checking connection...
21:37:21 WIF: Connected
.....................
21:40:08 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T21:40:08","Uptime":"0T00:03:39","UptimeSec":219,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":46,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:09"}}
21:40:08 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T21:40:08","ANALOG":{"A0":1},"BMP280":{"Temperature":25.76,"Pressure":960.50},"PressureUnit":"hPa","TempUnit":"C"}
21:40:12 CMD: deepsleeptime 45
21:40:12 SRC: WebConsole from 10.0.0.6
21:40:12 CMD: Group 0, Index 1, Command "DEEPSLEEPTIME", Data "45"
21:40:12 MQT: stat/bme_temp_out/RESULT = {"DeepSleepTime":45}
```
I have noticed that my board goes immidiatly off as soon as I set deepsleeptime in console without sending any other infos (like save, reset, status LWt etc...)

arendst, 2019-11-11:
As you execute from a web console it is expected that you will not see the start of the deep sleep messages.

Once woken the device will go to sleep as soon as teleperiod is executed.  I suggest you change teleperiod to 20 and retry.

guiumb, 2019-11-11:
Teleperiod set to 20:
Deepsleep cycles works perfectly but no MQTT data sent:

```
00:00:00 CFG: Loaded from flash at 3F5, Count 7
00:00:00 Project tasmota Temperatura Esterna Version 7.0.0.4(tasmota)-STAGE
00:00:00 I2C: BMP280 found at 0x76
00:00:00 WIF: Connecting to AP1 guilan in mode 11N as bme_temp_out-3130...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on bme_temp_out-3130 with IP address 10.0.0.174
22:21:34 MQT: Attempting connection...
22:21:34 MQT: Connected
22:21:34 MQT: tele/bme_temp_out/LWT = Online (retained)
22:21:34 MQT: cmnd/bme_temp_out/POWER = 
22:21:34 MQT: tele/bme_temp_out/INFO1 = {"Module":"Generic","Version":"7.0.0.4(tasmota)","FallbackTopic":"cmnd/bme_temp_out_fb/","GroupTopic":"cmnd/tasmotas/"}
22:21:34 MQT: tele/bme_temp_out/INFO2 = {"WebServerMode":"Admin","Hostname":"bme_temp_out-3130","IPAddress":"10.0.0.174"}
22:21:34 MQT: tele/bme_temp_out/INFO3 = {"RestartReason":"Software/System restart"}
22:21:42 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T22:21:42","Uptime":"0T00:00:19","UptimeSec":19,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:09"}}
22:21:42 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T22:21:42","ANALOG":{"A0":1},"BMP280":{"Temperature":24.91,"Pressure":960.06},"PressureUnit":"hPa","TempUnit":"C"}
22:22:02 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T22:22:02","Uptime":"0T00:00:39","UptimeSec":39,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:09"}}
22:22:02 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T22:22:02","ANALOG":{"A0":1},"BMP280":{"Temperature":24.89,"Pressure":960.06},"PressureUnit":"hPa","TempUnit":"C"}
22:22:03 LOG: SerialLog 0, WebLog 4, MqttLog 0, SysLog 0, LogHost , LogPort 514, TelePeriod 20
22:22:03 HTP: Configuration
22:22:03 CFG: Saved to flash at 3FB, Count 9, Bytes 4096
22:22:07 HTP: Main Menu
22:22:12 HTP: Console
22:22:14 WIF: Checking connection...
22:22:14 WIF: Connected
22:22:22 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T22:22:22","Uptime":"0T00:00:59","UptimeSec":59,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":69,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:09"}}
22:22:22 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T22:22:22","ANALOG":{"A0":1},"BMP280":{"Temperature":24.82,"Pressure":960.11},"PressureUnit":"hPa","TempUnit":"C"}
22:22:35 WIF: Checking connection...
22:22:35 WIF: Connected
22:22:42 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T22:22:42","Uptime":"0T00:01:19","UptimeSec":79,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:09"}}
22:22:42 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T22:22:42","ANALOG":{"A0":1},"BMP280":{"Temperature":24.79,"Pressure":960.08},"PressureUnit":"hPa","TempUnit":"C"}
22:22:56 WIF: Checking connection...
22:22:56 WIF: Connected
22:23:02 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T22:23:02","Uptime":"0T00:01:39","UptimeSec":99,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":49,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":52,"LinkCount":1,"Downtime":"0T00:00:09"}}
22:23:02 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T22:23:02","ANALOG":{"A0":1},"BMP280":{"Temperature":24.79,"Pressure":960.08},"PressureUnit":"hPa","TempUnit":"C"}
22:23:07 CMD: deepsleeptime 45
22:23:07 SRC: WebConsole from 10.0.0.6
22:23:07 CMD: Group 0, Index 1, Command "DEEPSLEEPTIME", Data "45"
22:23:07 MQT: stat/bme_temp_out/RESULT = {"DeepSleepTime":45}

00:00:00 CFG: Loaded from flash at 3FA, Count 10
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project tasmota Temperatura Esterna Version 7.0.0.4(tasmota)-STAGE
00:00:00 I2C: BMP280 found at 0x76
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 guilan in mode 11N as bme_temp_out-3130...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on bme_temp_out-3130 with IP address 10.0.0.174
00:00:08 APP: Boot Count 4
00:00:08 CFG: Saved to flash at 3F9, Count 11, Bytes 4096
22:23:24 MQT: Attempting connection...
[... 109 lines cut ...]
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on bme_temp_out-3130 with IP address 10.0.0.174
00:00:08 APP: Boot Count 6
00:00:08 CFG: Saved to flash at 3F7, Count 13, Bytes 4096
22:24:55 MQT: Attempting connection...
22:24:55 MQT: Connected
22:24:55 MQT: tele/bme_temp_out/LWT = Online (retained)
22:24:55 MQT: cmnd/bme_temp_out/POWER = 
22:24:55 MQT: Subscribe to cmnd/bme_temp_out/#
22:24:55 MQT: Subscribe to cmnd/tasmotas/#
[... cut ...]
22:25:47 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-11T22:25:47","Uptime":"0T00:00:18","UptimeSec":18,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":65,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":56,"LinkCount":1,"Downtime":"0T00:00:09"}}
22:25:47 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-11T22:25:47","ANALOG":{"A0":1},"BMP280":{"Temperature":24.74,"Pressure":960.08},"PressureUnit":"hPa","TempUnit":"C"}
22:25:59 WIF: Checking connection...
22:25:59 WIF: Connected
```

arendst, 2019-11-11:
Pls check your mqtt broker as I think the same situation happens as I noted before. The time between sending the teleperiod data and the sleep state is too short for updating the webconsole.

Just verify using the serial console where you must be able to see the mqtt data sending just before the sleep state.

guiumb, 2019-11-11:
Ok. I have to stop my work for today. Tomorrow I will do that.
Thks

guiumb, 2019-11-12:
Hi Matheus,

as you seggested it was a web refresh issue: using serial I can find MQTT messages sent before sleep mode activation:
```
00:00:00 CFG: Loaded from flash at 3F9, Count 27
00:00:00 SRC: Restart
00:00:00 Project tasmota Temperatura Esterna Version 7.0.0.4(tasmota)-STAGE
00:00:00 I2C: BMP280 found at 0x76
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 guilan in mode 11N as bme_temp_out-3130...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on bme_temp_out-3130 with IP address 10.0.0.174
00:00:08 APP: Boot Count 15
00:00:08 CFG: Saved to flash at 3F8, Count 28, Bytes 4096
20:52:41 MQT: Attempting connection...
20:52:41 MQT: Connected
20:52:41 MQT: tele/bme_temp_out/LWT = Online (retained)
20:52:41 MQT: cmnd/bme_temp_out/POWER = 
20:52:41 MQT: Subscribe to cmnd/bme_temp_out/#
20:52:41 MQT: Subscribe to cmnd/tasmotas/#
20:52:41 MQT: Subscribe to cmnd/bme_temp_out_fb/#
20:52:41 MQT: tele/bme_temp_out/INFO1 = {"Module":"Generic","Version":"7.0.0.4(tasmota)","FallbackTopic":"cmnd/bme_temp_out_fb/","GroupTopic":"cmnd/tasmotas/"}
20:52:41 MQT: tele/bme_temp_out/INFO2 = {"WebServerMode":"Admin","Hostname":"bme_temp_out-3130","IPAddress":"10.0.0.174"}
20:52:41 MQT: tele/bme_temp_out/INFO3 = {"RestartReason":"Software/System restart"}
20:52:41 NTP: Drift 0, (UTC) Tue Nov 12 19:52:41 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
20:52:41 HTP: Main Menu
20:52:49 MQT: tele/bme_temp_out/STATE = {"Time":"2019-11-12T20:52:49","Uptime":"0T00:00:19","UptimeSec":19,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":54,"MqttCount":1,"Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":52,"LinkCount":1,"Downtime":"0T00:00:09"}}
20:52:49 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-12T20:52:49","ANALOG":{"A0":1},"BMP280":{"Temperature":25.75,"Pressure":947.38},"PressureUnit":"hPa","TempUnit":"C"}
.....
20:53:22 WIF: Checking connection...
20:53:22 WIF: Connected
20:53:23 CMD: status 2
20:53:23 SRC: WebConsole from 10.0.0.6
20:53:23 CMD: Group 0, Index 1, Command "STATUS", Data "2"
20:53:23 MQT: stat/bme_temp_out/STATUS2 = {"StatusFWR":{"Version":"7.0.0.4(tasmota)","BuildDateTime":"2019-11-11T22:18:54","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(5ab15d1)"}}
20:53:26 CMD: status 0
20:53:26 SRC: WebConsole from 10.0.0.6
20:53:26 CMD: Group 0, Index 1, Command "STATUS", Data "0"
20:53:26 MQT: stat/bme_temp_out/STATUS = {"Status":{"Module":18,"FriendlyName":["Temperatura Esterna"],"Topic":"bme_temp_out","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:53:26 MQT: stat/bme_temp_out/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:56","StartupUTC":"2019-11-12T19:52:30","Sleep":50,"CfgHolder":4617,"BootCount":15,"SaveCount":28,"SaveAddress":"3F8000"}}
20:53:26 MQT: stat/bme_temp_out/STATUS2 = {"StatusFWR":{"Version":"7.0.0.4(tasmota)","BuildDateTime":"2019-11-11T22:18:54","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(5ab15d1)"}}
20:53:26 MQT: stat/bme_temp_out/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["guilan",""],"TelePeriod":45,"Resolution":"A98180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
20:53:26 MQT: stat/bme_temp_out/STATUS4 = {"StatusMEM":{"ProgramSize":482,"Free":3592,"Heap":25,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":2,"Features":["00000809","8FC0A197","00000000","00000405","000000C0","00000801"],"Drivers":"1,2,3,4,8,9,10,12,24,29","Sensors":"1,2,3,9"}}
20:53:26 MQT: stat/bme_temp_out/STATUS5 = {"StatusNET":{"Hostname":"bme_temp_out-3130","IPAddress":"10.0.0.174","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"5C:CF:7F:74:AC:3A","Webserver":2,"WifiConfig":4}}
20:53:26 MQT: stat/bme_temp_out/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.150","MqttPort":1883,"MqttClientMask":"bme_temp_out","MqttClient":"bme_temp_out","MqttUser":"umbyha","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:53:26 MQT: stat/bme_temp_out/STATUS7 = {"StatusTIM":{"UTC":"Tue Nov 12 19:53:26 2019","Local":"Tue Nov 12 20:53:26 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:19","Sunset":"16:56"}}
20:53:26 MQT: stat/bme_temp_out/STATUS10 = {"StatusSNS":{"Time":"2019-11-12T20:53:26","ANALOG":{"A0":1},"BMP280":{"Temperature":25.75,"Pressure":947.36},"PressureUnit":"hPa","TempUnit":"C"}}
[... cut ...] "Downtime":"0T00:00:09"}}
21:02:18 MQT: tele/bme_temp_out/SENSOR = {"Time":"2019-11-12T21:02:18","ANALOG":{"A0":1},"BMP280":{"Temperature":26.44,"Pressure":947.40},"PressureUnit":"hPa","TempUnit":"C"}
21:02:18 MQT: stat/bme_temp_out/RESULT = {"DeepSleep":{"Time":"2019-11-12T21:03:00","Epoch":1573588980}}
21:02:18 MQT: tele/bme_temp_out/LWT = Offline (retained)
```
I would like to thank you for your valuable support !!

Now I can try to delploy my Tasmota devices battery powered project.

Umberto

ascillato2, 2019-11-12:
Great to know that now it is working for you. Closing.

closed 2019-11-12 by ascillato2

arendst, 2019-11-12:
I'm in the process of adding a 4 seconds delay between sending the teleperiod and deepsleep start message and the actual deepsleep. This will alow the system to update the GUI before communication stops

To be released within an hour.

BTW. Thx for the donation.

guiumb, 2019-11-12:
Great !!

I will test it tomorrow on my D1 mini web gui.

It is not a big amount but I hope that's help !

betyar, 2020-09-15:
Hi arendst ,

I came across this issue trying to solve my own DeepSleep  problem. I am using Tasmota 8.5.0.1 I compiled on an ESP8266-12F. Attached is a DHT11 sensor. I have DeepSleep set for 60 seconds and a TelePeriod of 10.

It seems the original problem still persists.  First, I had the problem of some of my configuration settings being reset after a few DeepSleep cycles. This I solved using the SetOption36 0 workaround. Now my problem is similar to issue #949 where the sensor data is not published. What I noticed is that the module would always go into DeepSleep around 7 seconds after waking, not waiting for telemetry data. This happens regardless at how long I set the TelePeriod; currently it is at 10 but I tried it at 20 and even 300, the results were always the same. The module would always go into DeepSleep after about 7 seconds and then wake when it was supposed to. Sometimes telemetry data was received because the module took longer to go to sleep, but this was few and far between. I guess this was because it was waiting to connect to the MQTT server or NTP. But it appears DeepSleep doesn't wait for the Teleperiod at all.

This is the log data when DeepSleep is enabled:

> 00:00:00 CFG: Loaded from flash at F5, Count 135
> 00:00:00 CFG: CR 373/699, Busy 0
> 00:00:00 SRC: Restart
> 00:00:00 Project tasmota-garden Tasmota Version 8.5.0.1(tasmota)-2_7_4_1
> 00:00:00 DHT: (v5) Sensors found 1
> 00:00:00 WIF: Checking connection...
> 00:00:00 WIF: Attempting connection...
> 00:00:00 WIF: Connecting to AP1 betyar0 Channel 6 BSSId 00:23:CD:11:B1:9C in mode 11N as tasmota-garden_42BC3B-7227...
> 00:00:01 WIF: Checking connection...
> 00:00:01 WIF: Connected
> 08:47:05 NTP: UTC 2020-09-15T06:47:04, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
> 08:47:05 HTP: Web server active on tasmota-garden_42BC3B-7227 with IP address 192.168.1.96
> 08:47:06 MQT: Attempting connection...
> 08:47:06 MQT: Connected
> 08:47:06 MQT: tele/tasmota-garden_42BC3B/LWT = Online (retained)
> 08:47:06 MQT: cmnd/tasmota-garden_42BC3B/POWER = 
> 08:47:06 MQT: Subscribe to cmnd/tasmota-garden_42BC3B/#
> 08:47:06 MQT: Subscribe to cmnd/tasmotas/#
> 08:47:06 MQT: Subscribe to cmnd/DVES_42BC3B_fb/#
> 08:47:06 MQT: Unsubscribe from homeassistant/status
> 08:47:06 MQT: Received Topic "cmnd/tasmota-garden_42BC3B/DeepSleepTime", Data Size 2, Data "60"
> 08:47:06 SRC: MQTT
> 08:47:06 CMD: Group 0, Index 1, Command "DEEPSLEEPTIME", Data "60"
> 08:47:06 MQT: stat/tasmota-garden_42BC3B/RESULT = {"DeepSleepTime":60}
> 08:47:06 DSL: Reset wrong settings wakeup: 0, slip 10000
> 08:47:06 MQT: stat/tasmota-garden_42BC3B/RESULT = {"DeepSleep":{"Time":"2020-09-15T08:48:00","Epoch":1600152480}}
> 

And this is the log data when DeepSleep is diabled:

> 00:00:00 CFG: Loaded from flash at F5, Count 135
> 00:00:00 CFG: CR 373/699, Busy 0
> 00:00:00 SRC: Restart
> 00:00:00 Project tasmota-garden Tasmota Version 8.5.0.1(tasmota)-2_7_4_1
> 00:00:00 DHT: (v5) Sensors found 1
> 00:00:00 WIF: Checking connection...
> 00:00:00 WIF: Attempting connection...
> 00:00:00 WIF: Connecting to AP1 betyar0 Channel 6 BSSId 00:23:CD:11:B1:9C in mode 11N as tasmota-garden_42BC3B-7227...
> 00:00:01 WIF: Checking connection...
> 00:00:01 WIF: Connected
> 08:48:05 NTP: UTC 2020-09-15T06:48:04, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
> 08:48:05 HTP: Web server active on tasmota-garden_42BC3B-7227 with IP address 192.168.1.96
> 08:48:06 MQT: Attempting connection...
> 08:48:06 MQT: Connected
> 08:48:06 MQT: tele/tasmota-garden_42BC3B/LWT = Online (retained)
> 08:48:06 MQT: cmnd/tasmota-garden_42BC3B/POWER = 
> 08:48:06 MQT: Subscribe to cmnd/tasmota-garden_42BC3B/#
> 08:48:06 MQT: Subscribe to cmnd/tasmotas/#
> 08:48:06 MQT: Subscribe to cmnd/DVES_42BC3B_fb/#
> 08:48:06 MQT: Unsubscribe from homeassistant/status
> 08:48:06 MQT: Received Topic "cmnd/tasmota-garden_42BC3B/DeepSleepTime", Data Size 1, Data "0"
> 08:48:06 SRC: MQTT
> 08:48:06 CMD: Group 0, Index 1, Command "DEEPSLEEPTIME", Data "0"
> 08:48:06 MQT: stat/tasmota-garden_42BC3B/RESULT = {"DeepSleepTime":0}
> 08:48:06 CFG: Saved to flash at F4, Count 136, Bytes 4096
> 08:48:10 MQT: tele/tasmota-garden_42BC3B/STATE = {"Time":"2020-09-15T08:48:10","Uptime":"0T00:00:09","UptimeSec":9,"Vcc":3.188,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"betyar0","BSSId":"00:23:CD:11:B1:9C","Channel":6,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
> 08:48:10 MQT: tele/tasmota-garden_42BC3B/SENSOR = {"Time":"2020-09-15T08:48:10","DHT11":{"Temperature":34.8,"Humidity":32.0,"DewPoint":15.6},"TempUnit":"C"}
> 08:48:11 APP: Boot Count 49
> 08:48:11 CFG: Saved to flash at FB, Count 137, Bytes 4096
> 08:48:20 MQT: tele/tasmota-garden_42BC3B/STATE = {"Time":"2020-09-15T08:48:20","Uptime":"0T00:00:19","UptimeSec":19,"Vcc":3.187,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"betyar0","BSSId":"00:23:CD:11:B1:9C","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
> 08:48:20 MQT: tele/tasmota-garden_42BC3B/SENSOR = {"Time":"2020-09-15T08:48:20","DHT11":{"Temperature":34.9,"Humidity":32.0,"DewPoint":15.7},"TempUnit":"C"}
> 08:48:25 WIF: Checking connection...
> 08:48:30 MQT: tele/tasmota-garden_42BC3B/STATE = {"Time":"2020-09-15T08:48:30","Uptime":"0T00:00:29","UptimeSec":29,"Vcc":3.200,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"betyar0","BSSId":"00:23:CD:11:B1:9C","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
> 08:48:30 MQT: tele/tasmota-garden_42BC3B/SENSOR = {"Time":"2020-09-15T08:48:30","DHT11":{"Temperature":34.9,"Humidity":31.0,"DewPoint":15.2},"TempUnit":"C"}
> 
> 

Any thoughts on this?