arendst/Tasmota #6891: Include/add  timers informations to STATUS command managment

E-NINA, 2019-11-09:
Hello, I'm setting timmers on a sonoff 4Ch, direclty from Home assistant, and realized that seems isn't possible get back the actual status of timmers.
 It will be possible include information of all timers on "status" command.

Thanks 
ENina

meingraham, 2019-11-10:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

You can query the timer status using the `Timers` command or any specific timer by issuing `Timer<x>`. You can find the details of these commands in the wiki.

Please close this issue and address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

E-NINA, 2019-11-10:
Hello, sorry for the question. It was not clear for me from wiki that I could issue that command.
I understood that was to delete or copy, not also to report out values.
Thanks for replay

closed 2019-11-10 by E-NINA