arendst/Tasmota #6892: Shutterposition with no parameter close shutter

calcipher49, 2019-11-10:
#6586 # BUG DESCRIPTION

When Shutter mode is enabled, any command Shutterposition<X> with no parameter will close correspondig shutter instead returning current status.

### REQUESTED INFORMATION
Tasmota version: 7.0.0.3
Compiled  with #define USE_SHUTTER enabled
On any device with 2 relays, wired with: relay1 open shutter and relay2 close shutter

### TO REPRODUCE
Enable shutter mode:
Backlog Interlock 1 ; Interlock 1,2 ; Setoption80 1 ; ShutterRelay1 1 ; Restart 1
Then open shutter: Shutterposition1 100
 Then run: Shutterposition1 
This close the shutter (order interpred as Shutterposition1 0)

### EXPECTED BEHAVIOUR
Shutterposition<X> with no parameter will returns current status (shutter position) with no change on it, as for any other Tasmota command

ascillato2, 2019-11-12:
Hi,

Please, can you test latest Tasmota version (7.0.0.4) ?

ihavenonick, 2019-11-12:
same with Tasmota 7.0.0.4

ascillato, 2019-11-12:
@stefanbode Do you have any idea about this?

ihavenonick, 2019-11-12:
`19:30:47 MQT: tele/sonoff/SENSOR = {"Time":"2019-11-12T19:30:47","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2019-10-06T16:51:16","Total":1.153,"Yesterday":0.041,"Today":0.034,"Period":0,"Power":2,"ApparentPower":6,"ReactivePower":6,"Factor":0.33,"Frequency":50,"Voltage":231,"Current":0.027},"Shutter1":{"Position":100,"direction":0}}
19:34:52 CMD: Shutterposition1
19:34:52 SHT: Start shutter in direction -1
19:34:52 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
19:34:52 MQT: stat/sonoff/POWER2 = ON
19:34:52 MQT: stat/sonoff/RESULT = {"ShutterPosition1":0}
19:34:53 SHT: Shutter 0: Real Pos: 30279, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 0.3  [s]
19:34:54 SHT: Shutter 0: Real Pos: 28322, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 1.3  [s]
19:34:55 SHT: Shutter 0: Real Pos: 25644, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 2.6  [s]
19:34:56 SHT: Shutter 0: Real Pos: 23996, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 3.4  [s]
19:34:57 SHT: Shutter 0: Real Pos: 22348, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 4.2  [s]
19:34:58 SHT: Shutter 0: Real Pos: 20185, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 5.2  [s]
19:34:59 SHT: Shutter 0: Real Pos: 18125, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 6.2  [s]
19:35:00 SHT: Shutter 0: Real Pos: 15962, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 7.3  [s]
19:35:01 SHT: Shutter 0: Real Pos: 14005, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 8.3  [s]
19:35:02 SHT: Shutter 0: Real Pos: 11945, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 9.3  [s]
19:35:03 SHT: Shutter 0: Real Pos: 9885, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 10.3  [s]
19:35:04 SHT: Shutter 0: Real Pos: 7722, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 11.3  [s]
19:35:05 SHT: Shutter 0: Real Pos: 5662, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 12.3  [s]
19:35:06 SHT: Shutter 0: Real Pos: 3602, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 13.3  [s]
19:35:07 SHT: Shutter 0: Real Pos: 1542, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 14.3  [s]
19:35:08 MQT: stat/sonoff/SHUTTER1 = 0
19:35:08 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
19:35:08 MQT: stat/sonoff/POWER2 = OFF
19:35:08 MQT: tele/sonoff/RESULT = {"Shutter1":{"Position":0,"direction":0}}`

calcipher49, 2019-11-12:
Hi,
Same issue with 7.0.0.4
In the past, this command worked properly (with Stefandbode fork on 6.5.0 version).
Seems this bug appeared with Shutter integration in main Tasmota.

stefanbode, 2019-11-13:
Let me debug today and provide a fix. Have you seen since yesterday you can also send shutterposition up/down/stop to be compatible with openHAB and similar.

closed 2019-11-13 by arendst

localhost61, 2019-11-13:
I confirm the issue here. If position is 0, then it reports the position. If not, then it moves to position 0.

commit 4bc52c5b (2019-11-13): Merge pull request #6919 from stefanbode/patch-8

fix #6892
commit 888bcb4c (2019-11-13): fix #6892

#6892 shutterposition not reported