arendst/Tasmota #6895: 6.6.0 to 6.7.1 WiFi Dropping

krushmike, 2019-11-10:
I upgraded Sonoff S31 from 6.6.0 to 6.7.1 this morning and noticed WiFi was all of a sudden incredibly unstable.  Pinging the IP address of the device would frequently drop.  Thought maybe it was always this way, so moved the plug closer to my AP, same issue.  Rolled back to 6.6.0 and the issue is persisting.  Is there something that would have persisted in 6.7.1 which is preventing a full rollback to 6.7.1 so I can stabalize WiFi.

Noticed this in the 6.6.0 notes and thought this may be the issue:

This release is based on ESP8266/Arduino library core 2.3.0 as some people encountered wifi related issues on core 2.4.2 and 2.5.2.

ascillato2, 2019-11-10:
Hi, 

This has been already addressed. 

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-10 by ascillato2

krushmike, 2019-11-10:
Could you at least point me to where this has been addressed.  I am on the chat now and searching around a bit.

Jason2866, 2019-11-10:
Update to latest development version from http://thehackbox.org/tasmota/tasmota.bin
Uses latest new core 2.6.0. Do a `reset 5` before