arendst/Tasmota #6896: When changing the amount of Timers, SaveData is not working.

Einstein67, 2019-11-10:
### BUG DESCRIPTION
When changing the amount of Timers in tahmota.h (Line 59) SaveData is not working and Dump-Files are corrupted. 

REQUESTED INFORMATION
Tasmota version: 7.0.0.4
self compiled with "Core pre"

TO REPRODUCE
Change the Amount of Timer to e.g. 32, compile and flash.
Change any parameter like "setoption3 0" and reboot.
After restart setoption3 ist still ON

EXPECTED BEHAVIOUR
Settings should be stored at reboot (As it is the case with 16 timers). 

ADDITIONAL INFORMATION
Weblog 4 shows a different Flash-Byte size (4160 vs. 4196). Maybe this is a problem?

`with 16 Timer --> Savedata and Dump-File is OK
16:44:09 CFG: in Flash gespeichert am F7, zählen 5, Bytes 4096

with 32 Timer --> Savedata and Dump-File Error
16:41:19 CFG: in Flash gespeichert am F5, zählen 7, Bytes 4160`

arendst, 2019-11-10:
Einstein what's in a name.

You did not seem to have read the text above the changes you have made:
```
/*********************************************************************************************\
 * Constants
\*********************************************************************************************/

// Changes to the following MAX_ defines will impact settings layout
```

Any changes in this area will cross the 4k boundary of the Settings area corrupting flash. If you want more timers you will have to re-arrage the contents of the 4k Settings area.

Einstein67, 2019-11-10:
Ahhh OK, Then the behavior is normal and more than 16 timers are no longer supported!?! 
I'm almost sure that used to work.

Thx for your answer

closed 2019-11-10 by Einstein67