arendst/Tasmota #6901: MHZ19B soft UART setup crashes device

borpin, 2019-11-10:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

When using the MHZ19B configuration, the device crashes arefter a short while 

[image]

[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Generic (18)
- [x] Tasmota binary firmware version number used: _____ 6.6.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: _____ esphome_flasher
- [x ] Provide the output of command: ``Backlog Template; Module; GPIO``:

  Configuration output here:
```
20:34:34 MQT: stat/sonoffCO2_5/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:34:34 MQT: stat/sonoffCO2_5/RESULT = {"Module":"18 (Generic)"}
20:34:34 MQT: stat/sonoffCO2_5/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"61 (MHZ Rx)","GPIO5":"60 (MHZ Tx)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
  ```
- [x ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  ```
- [x ] Provide the output of this command: ``Status 0``:

```
20:35:23 CMD: status 0
20:35:23 MQT: stat/sonoffCO2_5/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff Master CO2"],"Topic":"sonoffCO2_5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:35:23 MQT: stat/sonoffCO2_5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Exception","Uptime":"0T00:04:47","StartupUTC":"2019-11-10T19:30:36","Sleep":50,"CfgHolder":4617,"BootCount":219,"SaveCount":231,"SaveAddress":"FA000"}}
20:35:23 MQT: stat/sonoffCO2_5/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sensors)","BuildDateTime":"2019-07-06T13:12:39","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:35:23 MQT: stat/sonoffCO2_5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["home123",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
20:35:23 MQT: stat/sonoffCO2_5/STATUS4 = {"StatusMEM":{"ProgramSize":545,"Free":456,"Heap":12,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDEE396","0017A3A4","B7FFBFCD","01DABBC0"]}}
20:35:23 MQT: stat/sonoffCO2_5/STATUS5 = {"StatusNET":{"Hostname":"sonoffCO2_5-0156","IPAddress":"192.168.7.126","Gateway":"192.168.7.25","Subnetmask":"255.255.255.0","DNSServer":"192.168.7.25","Mac":"2C:F4:32:4A:A0:9C","Webserver":2,"WifiConfig":4}}
20:35:23 MQT: stat/sonoffCO2_5/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.7.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4AA09C","MqttUser":"posting","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:35:23 MQT: stat/sonoffCO2_5/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 10 19:35:23 2019","Local":"Sun Nov 10 20:35:23 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:51","Sunset":"17:17"}}
20:35:23 MQT: stat/sonoffCO2_5/STATUS10 = {"StatusSNS":{"Time":"2019-11-10T20:35:23","MHZ19B":{"Model":"B","CarbonDioxide":445,"Temperature":21.0},"TempUnit":"C"}}
20:35:23 MQT: stat/sonoffCO2_5/STATUS11 = {"StatusSTS":{"Time":"2019-11-10T20:35:23","Uptime":"0T00:04:47","Heap":12,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"home123","BSSId":"B4:FB:E4:97:8D:5F","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
```
00:00:00 CFG: Loaded from flash at FA, Count 239
[... 51 lines cut ...]
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

No steps

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Detect CO2 without crashing

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Included above

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

This was working on ESPHome (but WiFi issues made me try Tasmota.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-11-10:
Known issue, software serial on core 2.3.0 doesnt work with MHZ 19
Use hardware serial or actual build

borpin, 2019-11-10:
Ok thanks - sorry my searching didn't turn it up as a known issue.

Can it be fixed?

ascillato2, 2019-11-10:
Yes, use hardware serial (use the tx and rx pins only) or just updated your Tasmota to latest and use core 2.6.0

Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-10 by ascillato2

borpin, 2019-11-10:
No bother. Thanks.