arendst/Tasmota #6904: Voltage is 0 despite setting

robin13, 2019-11-11:
### BUG DESCRIPTION

Using a Gosund SP111 socket I have tried with Sonoff-Tasmota 6.7.1 and Tasmota 7.0.0.3 but I always get `"Voltage":0`

I am using a 60W light bulb as a load. 

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund SP111
- [x] Tasmota binary firmware version number used: 6.7.1 and 7.0.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:
```
09:23:47 MQT: stat/sp111-01/RESULT = {"NAME":"Gosund SP111","GPIO":[56,0,57,0,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":18}
09:23:47 MQT: stat/sp111-01/RESULT = {"Module":{"0":"Gosund SP111"}}
09:23:47 MQT: stat/sp111-01/RESULT = {"GPIO":"Not supported"}
```

- [x] Provide the output of this command: ``Status 0``:
```
09:24:36 MQT: stat/sp111-01/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"sp111-01","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:24:36 MQT: stat/sp111-01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:02","StartupUTC":"2019-11-11T08:16:34","Sleep":50,"CfgHolder":4617,"BootCount":17,"SaveCount":63,"SaveAddress":"F8000"}}
09:24:36 MQT: stat/sp111-01/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
09:24:36 MQT: stat/sp111-01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["cundr_heim",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00208009","2805C8000100060000005A00000000000000","00000000"]}}
09:24:36 MQT: stat/sp111-01/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
09:24:36 MQT: stat/sp111-01/STATUS5 = {"StatusNET":{"Hostname":"sp111-01-6485","IPAddress":"XX-redacted-XX","Gateway":"XX-redacted-XX","Subnetmask":"255.255.255.0","DNSServer":"10.10.10.1","Mac":"XX-redacted-XX","Webserver":2,"WifiConfig":2}}
09:24:36 MQT: stat/sp111-01/STATUS6 = {"StatusMQT":{"MqttHost":"hassio.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A4D955","MqttUser":"hassio","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
09:24:36 MQT: stat/sp111-01/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 11 08:24:36 2019","Local":"Mon Nov 11 09:24:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:52","Sunset":"17:15"}}
09:24:36 MQT: stat/sp111-01/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:24:36 MQT: stat/sp111-01/STATUS10 = {"StatusSNS":{"Time":"2019-11-11T09:24:36","ENERGY":{"TotalStartTime":"2019-11-08T16:48:08","Total":0.015,"Yesterday":0.000,"Today":0.007,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
09:24:36 MQT: stat/sp111-01/STATUS11 = {"StatusSTS":{"Time":"2019-11-11T09:24:36","Uptime":"0T00:08:02","UptimeSec":482,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"XX-redacted-XX","BSSId":"XX-redacted-XX","Channel":11,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
```
09:26:40 MQT: tele/sp111-01/STATE = {"Time":"2019-11-11T09:26:40","Uptime":"0T00:10:06","UptimeSec":606,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"cundr_heim","BSSId":"F0:9F:C2:F7:C1:63","Channel":11,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}
09:26:40 MQT: tele/sp111-01/SENSOR = {"Time":"2019-11-11T09:26:40","ENERGY":{"TotalStartTime":"2019-11-08T16:48:08","Total":0.018,"Yesterday":0.000,"Today":0.009,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
```

### TO REPRODUCE

```
VoltageSet 230
PowerSet 60.0
CurrentSet 260.86
```

### EXPECTED BEHAVIOUR
I would expect voltage and power to be shown.  Interestingly the `Total` is shown and seems to be increasing with time.

Jason2866, 2019-11-11:
PZEM devices can NOT be calibrated
https://github.com/arendst/Tasmota/wiki/PZEM-0XX#note-on-calibration

Please close. Thx

andrethomas2, 2019-11-11:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-11 by andrethomas2

robin13, 2019-11-12:
In the end I found out that I have a SP111 v2 (I thought it was the v1) power monitoring plug:
https://blakadder.github.io/templates/gosund_SP111_v2.html

Using this template it worked fine and voltage was displayed, but was wildly inaccurate.
@Jason2866 - [power monitoring calibration](https://github.com/arendst/Tasmota/wiki/Power-Monitoring-Calibration) was necessary for it to give accurate readings.