arendst/Tasmota #6905: Reassign capacitive touch button

glennmurley15, 2019-11-11:
Is there a way of running a tasmota command/rule (eg to reassign a button’s action) that will NOT persist after a reboot of the device?

I see in the docs that all rules persist a reboot as they are stored in the flash memory.

Basically I need to send a command to a tasmota device (sonoff touch), assigning the button that activates the relay to an unused GPIO pin, to disable relay state change (essentially locking the relay in position and ignoring any further button presses)

However, if my device was somehow to lose wifi capability, I would need a way of reactivating the button, so it will toggle the relay again - my plan is to have this reactivation happen on restart.

I’m thinking about setting a rule by a command over wifi, that can be removed over wifi under intended use,

And then a rule on boot to remove the rule if power loss occurs - my failsafe incase wifi/mqtt loss so physical switch doesnt become redundant and need flashing 🙄

glennmurley15, 2019-11-11:
Ideally there would be:

Boot: assign touch button to default relay
MQTT: assign touch button to empty GPIO
MQTT: assign touch button to default relay

meingraham, 2019-11-11:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Rule with Wifi#Disconnected as a trigger?

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-11 by ascillato2