arendst/Tasmota #6907: Turn on,off POWER when Switch2 is toggle if mossquitto is down

au190, 2019-11-11:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

Backlog Module 18; SSID1 wifiid; Password1 pww; MqttHost 192.168.1.11; MqttUser usr; MqttPassword pw1; FriendlyName1 Ent_x1; MqttClient Ent_x1; Topic x1; Setoption13 1; PowerOnState 0; PowerRetain 1; SwitchMode2 1; GPIO5 21; GPIO14 10; SwitchTopic x1_Md;

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

Good result and logs:
16:14:07 MQT: cmnd/x1_Md/POWER2 = OFF
16:14:07 MQT: cmnd/x1_Md/POWER2 = ON
16:14:08 MQT: cmnd/x1_Md/POWER2 = OFF
16:14:08 MQT: cmnd/x1_Md/POWER2 = ON
16:14:09 MQT: cmnd/x1_Md/POWER2 = OFF
16:14:09 MQT: cmnd/x1_Md/POWER2 = ON
16:14:09 MQT: cmnd/x1_Md/POWER2 = OFF
16:14:09 MQT: cmnd/x1_Md/POWER2 = ON
16:14:16 MQT: cmnd/x1_Md/POWER2 = OFF

Error it comes when Mosquito is down.

16:09:45 RSL: stat/x1/POWER = OFF
16:09:45 RSL: stat/x1/RESULT = {"POWER":"ON"}
16:09:45 RSL: stat/x1/POWER = ON
16:09:46 RSL: stat/x1/RESULT = {"POWER":"OFF"}

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
When I press the switch on power2 I get message "cmnd/x1_Md/POWER2"
and the output "stat/x1/POWER " is not touggle.
If the Mosquitto is down when  I press the switch the output is toggleing, insted of getting message 
stat/x1/POWER 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
 To be the same massge when the mossguito is down "cmnd/x1_Md/POWER2" and the stat/x1/POWER not toggle.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-11-11:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-11 by ascillato2

au190, 2019-11-11:
But is this not a bug ?
Working the same feature in different mode if MQTT broker is available and if not?

ascillato, 2019-11-11:
Not a bug. A fallback feature. See switchtopic in the wiki.

au190, 2019-11-11:
A red the fallback feature. Is that good if the MQTT broker is available, and switch2 fallback to realy1 ?
Because in my case when I toggle switch2 it toggle realy1.

ascillato, 2019-11-11:
Yes, that is the intended behavior.

As explained in the wiki, you can modify this behavior by the use of rules.

au190, 2019-11-11:
Strange but I accept. Thx.