arendst/Tasmota #6908: iFan03 System#Boot Trigger not firing for Rule with Subscribe

cbkcbkcbk, 2019-11-11:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

System#Boot is not working as a rule trigger with iFan03.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [Y ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [Y ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [Y ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ Y] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [Y ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ Y] Device used (e.g., Sonoff Basic): Sonoff iFan03
- [ ] Tasmota binary firmware version number used: 6.7.1
  - [ ] Pre-compiled
  - [X ] Self-compiled
    - [X ] IDE / Compiler used: Gitpod
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```

  Configuration output here:
16:17:55 CMD: Backlog Template
16:17:55 MQT: stat/SFan01/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}

16:18:56 CMD: Module
16:18:56 MQT: stat/SFan01/RESULT = {"Module":{"71":"Sonoff iFan03"}}

16:19:48 CMD: GPIOS
16:19:48 MQT: stat/SFan01/RESULT = {"GPIOs1":{"0":"None","18":"Button2","91":"Button2n","123":"Button2i","127":"Button2in","19":"Button3","92":"Button3n","124":"Button3i","128":"Button3in","20":"Button4","93":"Button4n","125":"Button4i","129":"Button4in","9":"Switch1","82":"Switch1n","10":"Switch2","83":"Switch2n","11":"Switch3","84":"Switch3n","12":"Switch4","85":"Switch4n","13":"Switch5","86":"Switch5n","14":"Switch6","87":"Switch6n","15":"Switch7","88":"Switch7n"}}
16:19:48 MQT: stat/SFan01/RESULT = {"GPIOs2":{"16":"Switch8","89":"Switch8n","25":"Relay5","33":"Relay5i","26":"Relay6","34":"Relay6i","27":"Relay7","35":"Relay7i","28":"Relay8","36":"Relay8i","53":"Led2","57":"Led2i","54":"Led3","58":"Led3i","55":"Led4","59":"Led4i","157":"LedLink","158":"LedLinki","37":"PWM1","46":"PWM1i","38":"PWM2","47":"PWM2i","39":"PWM3","48":"PWM3i","40":"PWM4","49":"PWM4i","41":"PWM5","50":"PWM5i","42":"Counter1","94":"Counter1n"}}
16:19:48 MQT: stat/SFan01/RESULT = {"GPIOs3":{"43":"Counter2","95":"Counter2n","44":"Counter3","96":"Counter3n","45":"Counter4","97":"Counter4n","160":"Buzzer","5":"I2C SCL","6":"I2C SDA","1":"DHT11","2":"AM2301","3":"SI7021","4":"DS18x20","7":"WS2812","147":"ALux IrRcv","159":"ALux IrSel","143":"MY92x1 DI","144":"MY92x1 DCKI","140":"SM16716 CLK","141":"SM16716 DAT","142":"SM16716 PWR","180":"SM2135 Clk","181":"SM2135 Dat","107":"Tuya Tx","108":"Tuya Rx"}}
16:19:48 MQT: stat/SFan01/RESULT = {"GPIOs4":{"183":"EXS Enable","8":"IRsend","51":"IRrecv","73":"SR04 Tri","74":"SR04 Ech","102":"HX711 SCK","103":"HX711 DAT","130":"HLWBL SEL","131":"HLWBL SELi","132":"HLWBL CF1","133":"HLW8012 CF","134":"BL0937 CF","156":"ADE7953 IRQ","145":"CSE7766 Tx","146":"CSE7766 Rx","135":"MCP39F5 Tx","136":"MCP39F5 Rx","137":"MCP39F5 Rst","62":"PZEM0XX Tx","63":"PZEM004 Rx","98":"PZEM016 Rx","99":"PZEM017 Rx","71":"SerBr Tx","72":"SerBr Rx"}}
16:19:48 MQT: stat/SFan01/RESULT = {"GPIOs5":{"60":"MHZ Tx","61":"MHZ Rx","64":"SAir Tx","65":"SAir Rx","101":"SDS0X1 Tx","70":"SDS0X1 Rx","69":"PMS5003"}}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
17:13:46 CMD: Backlog Rule1
17:13:46 MQT: stat/SFan01/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":268,"Rules":"ON  system#boot  DO Subscribe Fpwr, custom/SFan01/Fpwr ENDON ON Fanspeed#Data DO publish custom/SFan01/FanSpeed %value% ENDON ON Fanspeed#Data>0 DO Mem1 %value% ENDON ON Event#Fpwr=1 DO FanSpeed %Mem1% ENDON ON Event#Fpwr=0 DO FanSpeed 0 ENDON"}
17:14:35 CMD: Backlog Rule2
17:14:35 MQT: stat/SFan01/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
17:14:42 CMD: Backlog Rule3
17:14:42 MQT: stat/SFan01/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

17:15:09 CMD: Status 0
17:15:09 MQT: stat/SFan01/STATUS = {"Status":{"Module":71,"FriendlyName":["Sonoff"],"Topic":"SFan01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
[... 57 lines cut ...]
2.  Enable Rule1

3.  Reboot iFan03

### EXPECTED BEHAVIOUR

After the reboot, the rule should be triggered to Subscribe to "custom/SFan01/Fpwr".

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

Command "Rule1" indicates that the rule was accepted.  The other triggers are working just fine (Fanspeed#Data, and Event#Fpwr=1, etc.).  I am able to manually subscribe to "custom/SFan01/Fpwr" via a terminal command.  It just won't happen on System#Boot trigger.

My two guesses are 1)  I have something set wrong, or 2) it is related to something different in the microprocessor as described by arendst during the process of incorporating iFan03.

"iFan03 info so far:

Where the ifan02 was controlled by the second microcontroller all control for the ifan03 has been moved to the esp8285; the microcontroller is only used for RF decoding.The microcontroller interfaces with esp8285 using the serial interface 9600 bps 8N1. I managed to trace all codes from the remote with handshaking between the two.As the esp8285 does the fan relay sequencing I have to implement this in Tasmota too.The buzzer is now controlled by the esp8285 where it can easily be enabled/disabled (GPIO10)So it will take some time before I have it all put together but at least I think I have the info I need.Stay tuned."

https://github.com/arendst/Tasmota/issues/5988#issuecomment-511115954
--

**(Please, remember to close the issue when the problem has been addressed)**

cbkcbkcbk, 2019-11-11:
Forgot the mention that I added the following lines in "user_config_override.h"

#undef USE_EMULATION_HUE                      // Enable Hue Bridge emulation for Alexa (+14k code, +2k mem common)
#undef USE_EMULATION_WEMO                     // Enable Belkin WeMo emulation for Alexa (+6k code, +2k mem common)

#define USE_EXPRESSION                         // Add support for expression evaluation in rules (+3k2 code, +64 bytes mem)
#define SUPPORT_IF_STATEMENT                 // Add support for IF statement in rules (+4k2 code, -332 bytes mem)
#define SUPPORT_MQTT_EVENT                     // Support trigger event with MQTT subscriptions (+3k5 code)

Jason2866, 2019-11-12:
You need to compare fanspeed with a value like in this example
`on fanspeed#data=3 do buzzer 3,3 endon`

cbkcbkcbk, 2019-11-12:
> You need to compare fanspeed with a value like in this example
> `on fanspeed#data=3 do buzzer 3,3 endon`

My rules using fanspeed seem to be working as I wanted as written.  Hoping for a fix to the problem of the System#Boot trigger not working (the first line of my rule).

Jason2866, 2019-11-12:
`ON  system#boot  DO Subscribe Fpwr, custom/SFan01/Fpwr ENDON`
is not a valid rule expression. Try  `ON  different trigger DO Subscribe Fpwr, custom/SFan01/Fpwr ENDON` and it will not work too.

Jason2866, 2019-11-12:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2019-11-12:
Hi,

The rule:

```
ON  system#boot  DO Subscribe Fpwr, custom/SFan01/Fpwr ENDON
```

Is not valid because on SYSTEM BOOT the MQTT was not initialized, so the subscribe command is ignored.

You have to use MQTT#INITIALIZED instead.

Thanks

closed 2019-11-12 by ascillato2

cbkcbkcbk, 2019-11-12:
I tried and now agree that it won’t work with a different trigger, but I don’t see how this is an invalid expression. I still suspect this is a bug.  If you insist it’s an invalid expression, I will go to the chat.  But please at least consider this:

1.   I followed exactly the given example in the wiki.

		https://github.com/arendst/Tasmota/wiki/Subscribe-&-Unsubscribe

		“Examples:
		Rule1
     			 ON System#Boot DO Subscribe BkLight, stat/%topic%/POWER ENDON
      			ON Event#BkLight=ON DO <command> ENDON”
  2.   I can copy  the Subscribe command direct from Rule1 output in the console, paste it as a new command in the console, and it works fine:

16:07:22 CMD: *******Demonstration
16:07:27 CMD: Subscribe
16:07:27 MQT: stat/SFan01/RESULT = {"Subscribe":""}
16:07:31 CMD: Rule1
16:07:31 MQT: stat/SFan01/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":268,"Rules":"ON  system#boot  DO Subscribe Fpwr, custom/SFan01/Fpwr ENDON ON Fanspeed#Data DO publish custom/SFan01/FanSpeed %value% ENDON ON Fanspeed#Data>0 DO Mem1 %value% ENDON ON Event#Fpwr=1 DO FanSpeed %Mem1% ENDON ON Event#Fpwr=0 DO FanSpeed 0 ENDON"}
16:07:45 CMD: Subscribe Fpwr, custom/SFan01/Fpwr
16:07:45 MQT: stat/SFan01/RESULT = {"Subscribe":"FPWR,custom/SFan01/Fpwr/#"}
16:07:50 CMD: Subscribe
16:07:50 MQT: stat/SFan01/RESULT = {"Subscribe":"FPWR,custom/SFan01/Fpwr; "}

I totally acknowledge I can be missing something stupid here.

ascillato, 2019-11-12:
Hi,

Again:

The rule:

```
ON  system#boot  DO Subscribe Fpwr, custom/SFan01/Fpwr ENDON
```

Is not valid because on SYSTEM#BOOT the MQTT was not initialized, so the subscribe command is ignored, because THERE IS NO MQTT CONNECTION !

You have to use MQTT#INITIALIZED instead.

The wiki article was wrong. Fixing...