arendst/Tasmota #6914: template for shelly dimmer

bloodhound83, 2019-11-12:
Hello, 

I would like to order the new "Shelly Dimmer" and currently do not see a template. Has anyone ever tested the device or even run it with a different setting from another device?

If the request is wrong, please move or close it.

ascillato, 2019-11-12:
Shelly didn't start shipping those devices. No one have them, so we don't know if Tasmota can be flashed on them. Anyway, The stock firmware of Shelly supports MQTT, so if Tasmota don't work, you can still use them with your Home Automation Software.

bloodhound83, 2019-11-12:
@ascillato thank you for your quick answer. 
I received my shipping message yesterday, so I assume that some devices have been shipped. But i guess you are right, i have only to wait a couple of days / weeks to get an answer...

good hint regarding the mqtt support from the stock firmware....

jj-uk, 2019-11-25:
My shelly dimmers arrived today. Have you had any time to see if Tasmota can be flashed on to them? Sorry to bother you.

ascillato, 2019-11-25:
Hi,

Mine have arrived too. It has an ESP8266, so it might be possible to use Tasmota on it, but will require some more research due to it uses several other chips for controlling the dimming.

So, please be patient ;)

meingraham, 2019-11-25:
Some high res photos provided by @Dino on Discord - for future reference:
https://onedrive.live.com/?authkey=%21AEmy6Ok03JXOMbk&id=CA999BBC242363E5%21115610&cid=CA999BBC242363E5

jj-uk, 2019-11-25:
Maybe an email to their support dept. Asking for a simple sketch to demo dimming? Can't hurt to ask...
They'll sell more once it's been tasmotafied. :D

djhk739, 2019-11-26:
I had two delivered over the weekend along with a Shelly 1PM. Got tasmota successfully flashed on the 1PM, but no joy with the dimmers. I was using the diagram from the Shelly site for the header pin-outs but the programmer could not connect. This evening I wired up both dimmers and set them up with the Shelly software and new firmware was available, so have updated them but not tried to re-flash yet. D

SwedishChef, 2019-11-26:
I successfully flashed the dimmer! 
I also had the problem that the programmer could not connect, so I opened the housing and connected again. What I did ' wrong' at this time was that I connected all the wires in mirrored order by accident - and flashing suddenly succeeded. So my guess is that the documentation for the pin-out is wrong (mirrored).
I already requested the GPIO mapping to configure Tasmota but did not get a reply so far...

djhk739, 2019-11-26:
Great news SwedishChef, when you say ‘mirrored’ do you mean they’ve got ‘top to bottom’ mixed up? D

SwedishChef, 2019-11-27:
The problem is that after opening the housing and flashing I recognized that the device was turned around 180 degrees compared to the pin-out picture on shell home page.
As I still connected the needed wires as described, it was most probably 'top-to-bottom mixed up'. Give it a try :-)
I already sent a message to Shelly to get this clarified.

djhk739, 2019-11-27:
Yup, got mine flashed as well, now moving on to the GPIO assignments.

SwedishChef, 2019-11-28:
@djhk739 please share your results for GPIO assignment here - many thanks.

PS: still no answer from shell.org regarding this ;-(

kovaga, 2019-11-29:
I've sent them an email re. tasmota template support 2 days ago as well, but no replay so  far.
Usually they are pretty fast on replying.

phatina, 2019-11-30:
> I've sent them an email re. tasmota template support 2 days ago as well, but no replay so far.
> Usually they are pretty fast on replying.

Where can I get this template? What's the current status, please?

ascillato, 2019-11-30:
There is no template for shelly dimmer. It requires more code modifications and reverse engineering. It will take some time on being developed.

Tim3100, 2019-12-03:
Thanks a lot for your efforts! Highly appreciated.

jj-uk, 2019-12-05:
Anyone heard back from Shelly yet? I had a reply to my request for the source code and/or schematics as the product is sold as "open-source" (written on the box packaging).

Turns out it's not open-source at all and they're not going to give me the code :/

So I've asked for an example sketch to highlight how to perform dimming instead....

SwedishChef, 2019-12-05:
Just for me to understand the problem: 
Why configuring function PWMx to the corresponding GPIO pin via tasmota template does not work in case of Shelly dimmer? Do they use some different (non-standard) approach to dim the output?

ascillato, 2019-12-05:
As explained in previous comments. Shelly dimmer has other mcu to control dimming. It requires a new driver to be coded and a lot of reverse engineering. Tasmota is not going to work as it is now.

phatina, 2019-12-05:
> Anyone heard back from Shelly yet? I had a reply to my request for the source code and/or schematics as the product is sold as "open-source" (written on the box packaging).
>
> Turns out it's not [...]

Aaah, I asked them the same question but with no answer.

mgoeller, 2019-12-23:
Indeed, the pinout reference on the official Shelly website is wrong. I figured out most of the connections of the second (outer) row of the header on the PCB. The right (inner) row corresponds to the flashing connections of the ESP8266, the outer row for the STM32F031K6.

Did someone figure out the communication between the ESP and the MCU? Serial? Why did Shelly decide to breakout GPIO2 (U1TX) of the ESP on the header?
[image]
[image]

joba-1, 2019-12-23:
an STM32 with all flash/debug pins broken out ... interesting!

mgerald21, 2019-12-27:
I followed the esp8266 rx/tx traces and both are connect via 200R resistors to the stm32 PA9/PA10.
So I'm pretty sure both MCUs use the uart for communication.

Did someone manage to dump the flash of the esp? I'm unable to connect with esptool.py.

edge90, 2020-01-02:
I'm gonna give this some time soon. The stm32 is hopefully not read/write protected.

jj-uk, 2020-01-02:
Snooping the esp gpio whilst sending dim commands would be a good start. My logic analyser is not mains rated so I can't do this.
Anyone have a scope that can record ?

mgerald21, 2020-01-02:
I did make some progress ...
I was able to read the esp dimmer firmware with esptool, but stm32 need to be held in reset state by keeping NRST on ground.

Digging into firmware shows, that shelly dimmer is based on rBoot with dual image support and 256k spiffs. spiffs holds an stm.bin files which is very likely an stm32 firmware image!? Maybe shelly dimmer also support OTA update of the stm32 MCU?

Akeid, 2020-01-05:
I really like these dimmers. On a side note, I have a total of 300W worth of LEDs that need to be dimmed on a single circuit. The shelly dimmer is rated for 200-220W max. It looks like the FET doing the dimming is the [OSG65R200J](http://www.orientalsemi.com/storage/OSG65R200J.pdf) and there are two of them on the board. Does anyone have a suggestion for another mosfet in a similar form factor that is rated for higher power? Would a placing a heatsink on top of the existing mosfets help disspate the extra 100W that I need? The datasheet for the OSG65R200J shows a max power dissipation value of 151W for each mosfet.

ascillato2, 2020-01-05:
@Akeid offtopic, please address those questions to the Tasmota support chat.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

emilhetty, 2020-01-07:
The ones of you that have managed to flash the Shelly Dimmer, did you do any trick to flash it? Are you sure that the GPIO is correct?
I can't connect to mine, have 6 of them and can't connect to anyone of them.
I've flashed atleast 20 other Shelly devices, so I know how to flash them.

jan-koch, 2020-01-08:
> Edit: Found it out, VCC has to be powered with 5V and not 3.3V as usual for ESP8266.
> Then it flashed as it should.

And then you can simply control the dimmer via Tasmota as you do with regular ESP8266 devices? Sorry if that's a stupid question, I'm quite new to this.

edge90, 2020-01-08:
No, there's still stuff to work out.

edge90, 2020-01-08:
[Flash from the shelly dimmer](https://github.com/edge90/shelly_dimmer/raw/master/esp8266ex_flash.img)

**To connect to the device:**
Hookup the device as @mgoeller pictured. Connect GPIO0 to GND then plug the usb device in. Wait a second or so and then disconnect GPIO0 from GND.

**Read the flash**
`esptool.py --port /dev/ttyUSB0 --baud 115200 read_flash 0x00000 0x400000 backup.img`

emilhetty, 2020-01-09:
> > Edit: Found it out, VCC has to be powered with 5V and not 3.3V as usual for ESP8266.
> > Then it flashed as it should.
> 
> And then you can simply control the dimmer via Tasmota as you do with regular ESP8266 devices? Sorry if that's a stupid question, I'm quite new to this.

No, you could not, the Shelly Dimmer is not supported yet.
The Shelly Dimmer is much more complicated than the other Shelly devices since it has a second mcu.

emilhetty, 2020-01-09:
[Link to the Shelly firmwares ](https://api.shelly.cloud/files/firmware/)
[And the Shelly Dimmer firmware file](http://api.shelly.cloud/firmware/SHDM-1_build.zip)

edge90, 2020-01-09:
The BOOT0 pin on the stm32 seems like it's connected to pin 17 on the esp8266.

mgerald21, 2020-01-09:
@edge90: are sure about the BOOT pin?
Stm32f031:
BOOT0=0 BOOT1=x - main flash memory boot
BOOT0=1 BOOT1=0 - system memory boot (boot loader)
BOOT0=1 BOOT1=1 - RAM memory

BOOT0=1 would always start the bootloader and esp8266 needs to transfer the firmware using UART. Well there is an stm.bin file in the SPIFFS of the esp so this sounds reasonable.

edge90, 2020-01-10:
Pretty sure, followed the traces and verified with a ohmmeter. Will check again tonight.

edge90, 2020-01-10:
[image]
[image]

wichers, 2020-01-15:
Reversing the stm.bin blob confirms GPIOA PIN9, PIN10 are used for uart communication and it is set to 115200. 

The UART IRQ handler looks relatively simple and straightforward (protocol should be simple), needs some more reversing though..

Here is my current ida db:
https://pastebin.com/xYuGs481

And the spiffs extracted stm.bin:
https://pastebin.com/Wtr5GAAT

KoljaWindeler, 2020-01-15:
Just a short question ... has anyone considered to ask shelly for information about the communication protocol? They've shown a fair amount of community support in the past .. 

edit: I've send them a mail. Will get back to you asap if I hear back from them

ascillato, 2020-01-15:
@KoljaWindeler 

> Just a short question ... has anyone considered to ask shelly for information about the communication protocol? They've shown a fair amount of community support in the past ..

yes, multiple times in this thread. See previous comments:

https://github.com/arendst/Tasmota/issues/6914#issuecomment-558392370

https://github.com/arendst/Tasmota/issues/6914#issuecomment-562120992

https://github.com/arendst/Tasmota/issues/6914#issuecomment-558959497

https://github.com/arendst/Tasmota/issues/6914#issuecomment-559497593

mgerald21, 2020-01-15:
Can one of the devs reopen this issue. There is already same progress and further investigation is ongoing.

wichers, 2020-01-17:
Still need to do more reversing.. UART packet structure consists at least of something like:

```
uint16_t packet_checksum(uint8_t *buf, int len)
{
  int counter;
  uint16_t ret;

  counter = 0;
  ret = 0;
  while ( counter < len )
    ret += buf[counter++];
  return ret;
}

struct packet {
 //uint8_t unknown_0; // do not yet know if this is transmitted, on TX always 1
 uint8_t unknown_1;
 uint8_t id; // should be (1, 4), 2, 3, 16, 17, 32 or 48 
 uint8_t size;
 uint8_t buf[xxx]; // buf is optional
 uint16_t checksum;
 uint8_t unknown_2; // on TX always 4?
}  __attribute__((packed));

// checksum is calculated like this:
packet->checksum = packet_checksum(&packet->unknown_1, packet->size + 3);
```
Other GPIO pins used:
GPIOA PIN 8, 11
GPIOB PIN 2

mgerald21, 2020-01-17:
PA8(18) and PA11(20) are probably driving the two power mosfets.

Energy monitoring is done via a HLW8012
STM ----- HLW8012
PA0(6)  =   CF1(7)
PA1(7)   =  CF(6)

edge90, 2020-01-17:
By looking at the images is seems like PA11 (which is pin 21 not 20) goes to the one of the mosfets.

mgerald21, 2020-01-18:
You're right - corrected that.

mgerald21, 2020-01-18:
Alright, I hooked up an opto-isolator to the TX line ....
Serial output starts with some garbage and rBoot startup message.

> rBoot v1.2.1 - richardaburton@gmail.com
> Flash Size:   16 Mbit
> Flash Mode:   DIO
> Flash Speed:  40 MHz
> rBoot Option: Big flash
> 
> Booting rom 0 (0x8000).

After boot there is binary communication between the MCUs ([capture.txt](https://github.com/arendst/Tasmota/files/4080693/capture.txt))

@wichers: your packet definition looks pretty right:

01 01 20 06 00 00 00 00 0f 00 00 36 04 - Start
01 02 10 00 00 12 04 - Polling
01 03 10 00 00 13 04 - Polling
01 04 10 00 00 14 04 - Polling
01 05 10 00 00 15 04 - Polling
01 06 10 00 00 16 04 - Polling
01 07 10 00 00 17 04 - Polling
01 08 10 00 00 18 04 - Polling
01 09 10 00 00 19 04 - Polling
01 0a 10 00 00 1a 04 - Polling
01 0b 02 06 fe 01 ff 00 00 00 02 11 04  - Dimming
01 0c 10 00 00 1c 04 - Polling
01 0d 01 02 e8 03 00 fb 04 - ??
01 0e 10 00 00 1e 04 - Polling
01 0f 02 06 00 00 f4 01 00 00 01 0c 04 - Dimming
01 10 10 00 00 20 04 - Polling
01 11 10 00 00 21 04 - Polling
01 12 02 06 e8 03 f4 01 00 00 01 fa 04 - Dimming
01 13 10 00 00 23 04 - Polling
01 14 02 06 00 00 f4 01 00 00 01 11 04 - Dimming
01 15 10 00 00 25 04 - Polling
01 16 10 00 00 26 04 - Polling
01 17 10 00 00 27 04 - Polling            

- Byte 0 - always 0x01
- Byte 1 - packet counter
- Byte 2 - command
   * 0x01 = ??
   * 0x02 = dimming
   * 0x10 = polling
   * 0x20 = startup
- Byte3 - size
- Byte[s] - payload (at least one byte)
- ByteN-2 = checksum
- ByteN-1 = always 0x04

Next I'll try to capture RX line in parallel.

edge90, 2020-01-19:
Don't know if this helps anyone but here's a dump of the stm32f031k6: https://github.com/edge90/shelly_dimmer/raw/master/stm32f031k6.bin

edge90, 2020-01-20:
Got some more time to dig around.

# From 100% to 0%
01 2a 20 06 00 00 00 00 0f 00 00 5f 04 (from 100% to 0% while off \#1)
01 4f 20 06 00 00 00 00 0f 00 00 84 04 (from 100% to 0% while off \#2)
01 e0 20 06 00 00 00 00 0f 00 01 15 04 (from 100% to 0% while off \#3)

01 8a 20 06 0a 00 00 00 0f 00 00 c9 04 (from 100% to 0% while on \#1)
01 a9 20 06 0a 00 00 00 0f 00 00 e8 04 (from 100% to 0% while on \#2)
01 53 20 06 0a 00 00 00 0f 00 00 92 04 (from 100% to 0% while on \#3)

# From 0% to 100%
01 c3 20 06 00 00 00 00 0f 00 00 f8 04 (from 0% to 100% while off \#1)
01 fd 20 06 00 00 00 00 0f 00 01 32 04 (from 0% to 100% while off \#2)
01 24 20 06 00 00 00 00 0f 00 00 59 04 (from 0% to 100% while off \#3)

01 89 20 06 e8 03 00 00 0f 00 01 a9 04 (from 0% to 100% while on \#1)
01 b1 20 06 e8 03 00 00 0f 00 01 d1 04 (from 0% to 100% while on \#2)
01 d9 20 06 e8 03 00 00 0f 00 01 f9 04 (from 0% to 100% while on \#3)

# Toggle switch at 100%
01 f4 02 06 00 00 f4 01 00 00 01 f1 04 (turn off while at 100% \#1)
01 a0 02 06 00 00 f4 01 00 00 01 9d 04 (turn off while at 100% \#2)
01 bb 02 06 00 00 f4 01 00 00 01 b8 04 (turn off while at 100% \#3)

01 04 02 06 e8 03 f4 01 00 00 01 ec 04 (turn on while at 100% \#1)
01 2e 02 06 e8 03 f4 01 00 00 02 16 04 (turn on while at 100% \#2)
01 75 02 06 e8 03 f4 01 00 00 02 5d 04 (turn on while at 100% \#3)

# Toggle switch at 0%
01 07 20 06 0a 00 00 00 0f 00 00 46 04 (turn on while at 0% \#1)
01 3b 20 06 0a 00 00 00 0f 00 00 7a 04 (turn on while at 0% \#2)
01 74 20 06 0a 00 00 00 0f 00 00 b3 04 (turn on while at 0% \#3)

01 23 20 06 00 00 00 00 0f 00 00 58 04 (turn off while at 0% \#1)
01 5d 20 06 00 00 00 00 0f 00 00 92 04 (turn off while at 0% \#2)
01 a5 20 06 00 00 00 00 0f 00 00 da 04 (turn off while at 0% \#3)

" Settings
All of the following sequences are repeated 4 times for some reason.

## Set pulse mode to "trailing edge"

01 88 30 48 00 00 00 00 00 00 00 00 00 00 00 00 (repeated 4 times)
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 01 00 04

01 89 20 06 00 00 02 00 0f 00 00 c0 04 (followed by this 4 times)

## Set pulse mode to "leading edge"
01 e5 30 48 00 00 00 00 00 00 00 00 00 00 00 00 (repeated 4 times)
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 01 5d 04

01 e6 20 06 00 00 01 00 0f 00 01 1c 04 (followed by this 4 times)

## Start calibration
01 32 30 48 00 00 00 00 00 00 00 00 00 00 00 00 (repeated 4 times)
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 aa 04

01 33 04 02 00 00 00 39 04 (followed by this 4 times)

## Buttons fade rate

### x1
01 fc 20 06 00 00 01 00 05 00 01 28 04 (repeated 4 times)

### x2
01 5c 20 06 00 00 01 00 0a 00 00 8d 04 (repeated 4 times)

### x3
01 75 20 06 00 00 01 00 0f 00 00 ab 04 (repeated 4 times)

### x4
01 cf 20 06 00 00 01 00 14 00 01 0a 04 (repeated 4 times)

#### Full example for x4
01 28 20 06 00 00 01 00 14 00 00 63 04
01 28 20 06 00 00 01 00 14 00 00 63 04
01 28 20 06 00 00 01 00 14 00 00 63 04
01 28 20 06 00 00 01 00 14 00 00 63 04

01 29 10 00 00 39 04

01 2a 20 06 00 00 01 00 14 00 00 65 04
01 2a 20 06 00 00 01 00 14 00 00 65 04
01 2a 20 06 00 00 01 00 14 00 00 65 04
01 2a 20 06 00 00 01 00 14 00 00 65 04

### x5
01 f0 20 06 00 00 01 00 19 00 01 30 04 (repeated 4 times)

# Additional stuff
For the following tests I connected the usb serial rx to rx on the on board.
I'm guessing that this is the tx-data from the stm32.

## Set "pulse mode" (leading/trailing edge)
01 02 30 01 01 00 34 04 (set trailing edge \#1)
01 03 20 01 01 00 25 04
01 55 30 01 01 00 87 04 (set trailing edge \#2)
01 56 20 01 01 00 78 04

## Got these two when setting "pulse mode" to leading edge.
01 6f 30 01 01 00 a1 04 (set leading edge \#1)
01 70 20 01 01 00 92 04
01 2a 30 01 01 00 5c 04 (set leading edge \#2)
01 2c 20 01 01 00 4e 04

## Buttons fade rate

### x1
01 b0 20 01 01 00 d2 04

### x2
01 d8 20 01 01 00 fa 04

KoljaWindeler, 2020-01-20:
Hi, Ive also done some digging, **setting the brightness** seams to be very straight forward:
the 2 byte payload is birghtness*10 in 2-byte little endian 

e.g. 18% is 180 in dec, 00B4 in hex so the payload is B4 00.
So the complete message turn out as 
01 9C 01  02   B4 00   01 53   04 = 0x01, id, cmd, len, payload, check, 0x04 

That works for all brightness values, even "off" is simply 00 00 as brightness.

One thing to figure out is the response to the cmd 0x10.
Is see something like this 
01 45 10 0C 01 00 B4 00 00 00 00 00 0F 00 00 00 01 25 04

I'd guess power consumption?

wichers, 2020-01-20:
@edge90 thank you!

> One thing to figure out is the response to the cmd 0x10.

It consists of 2 x uint16_t followed by 2 x uint32_t. The source of the second uint16_t and the last uint32_t are known to me, the others two values aren't, at the moment.. 

Does anyone know of a working stm32 uart bootloader implementation? Or do we need to rewrite stm32flash (https://sourceforge.net/projects/stm32flash/) to arduino?

KoljaWindeler, 2020-01-20:
> The source of the second uint16_t and the last uint32_t are known to me, the others two values aren't, at the moment

would you share your knowledge?

wichers, 2020-01-20:
second uint16_t is a copy from brightness value set in 0x01 or 0x20 packet

last uint32_t is a copy from the last uint16_t 0x20 packet, but is only copied from 0x20 packet handler when value < 100, this seems to be the fade rate as edge90 found out..

edge90, 2020-01-20:
@KoljaWindeler I had just realized that there was something wrong with my original attempt. Updated answer should be more accurate.

mgerald21, 2020-01-21:
What about starting a Wiki entry for the Shelly dimmer to collect all the current information?

mgerald21, 2020-01-21:
@edge90: absolutely no, pin headers on Shellies (1, 2, dimmer, etc ) are not isolated from main!!!! So GND on the header is somehow connect to mains and you must not use GPIO while connected to mains. That’s why I hooked up an optocoupler to sniff tax line!

KoljaWindeler, 2020-01-21:
Just as a precautions: You can power the device totally fine by applying 3.3V to VCC on the header. That way you can easily sniff and test without any risk. I'll try to intercept the communication tonight. 
My plan is to connect the esp8266 tx/rx lines from the dimmer to another esp8266 (emulation device) and to keep the stm in reset so it won't disturb the communication.

This way I'll make sure that my parser and message coder works correct and I'll be able to fake some 0x10 messages. I'll try to add some values into the first 32bit field and see if that changes the display of the wattage output (on the dimmer website).

I guess we can remove the original firmware once that's working, or is there anything that we don't know by now?

edge90, 2020-01-21:
@KoljaWindeler that's how i got my data. However you won't get any power consuption reports with a value other than zero. That will require a load which in turn requires mains power.

KoljaWindeler, 2020-01-21:
> That will require a load which in turn requires mains power.

true, OR by intercepting it :)  I really don't like open 220VAC

edge90, 2020-01-21:
@KoljaWindeler regarding the ack. I didn't read them at the same time so the ack is a response from later "request".

[203 more replies]

PR #9905 (2020-11-18): Disable SHELLY_VOLTAGE_MON parameter
## Description:

The `SHELLY_VOLTAGE_MON` parameter should only be defined if a HW patch has been done which enables voltage and current monitoring described here: https://github.com/arendst/Tasmota/issues/6914#issuecomment-691668208

**Related issue (if applicable):** fixes #6914

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V. [... cut]
closed 2020-11-16 by ascillato2