arendst/Tasmota #6920: 6.7.1 Wi-Fi Disconnects on S20 Sockets

rcarmo, 2019-11-13:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

Version 6.7.1 S20 Sockets disconnect from Wi-Fi and enter a WiFiManager loop.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): S20
- [ ] Tasmota binary firmware version number used: 6.7.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

10:59:21 CMD: Backlog Template; Module; GPIO
10:59:21 MQT: stat/socket0/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
10:59:21 MQT: stat/socket0/RESULT = {"Module":{"8":"Sonoff S2X"}}
10:59:22 MQT: stat/socket0/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"}}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

10:59:47 CMD: status 0
10:59:47 MQT: stat/socket0/STATUS = {"Status":{"Module":8,"FriendlyName":["Socket 0"],"Topic":"socket0","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:59:47 MQT: stat/socket0/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.105:8000/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"1T22:04:48","StartupUTC":"2019-11-11T11:54:59","Sleep":50,"CfgHolder":4617,"BootCount":95,"SaveCount":4954,"SaveAddress":"F4000"}}
10:59:47 MQT: stat/socket0/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":6,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
10:59:47 MQT: stat/socket0/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"home.lan","LogPort":514,"SSId":["7","7"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000009","2805C8000100060000005AFF000000000000","00000001"]}}
10:59:47 MQT: stat/socket0/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
10:59:47 MQT: stat/socket0/STATUS5 = {"StatusNET":{"Hostname":"socket0","IPAddress":"192.168.1.93","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:11:42:4B","Webserver":2,"WifiConfig":4}}
10:59:47 MQT: stat/socket0/STATUS6 = {"StatusMQT":{"MqttHost":"home.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_11424B","MqttUser":"DVES_USER","MqttCount":7,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
10:59:47 MQT: stat/socket0/STATUS7 = {"StatusTIM":{"UTC":"Wed Nov 13 09:59:47 2019","Local":"Wed Nov 13 10:59:47 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:55","Sunset":"17:13"}}
10:59:47 MQT: stat/socket0/STATUS10 = {"StatusSNS":{"Time":"2019-11-13T10:59:47"}}
10:59:47 MQT: stat/socket0/STATUS11 = {"StatusSTS":{"Time":"2019-11-13T10:59:47","Uptime":"1T22:04:48","UptimeSec":165888,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":7,"POWER":"OFF","Wifi":{"AP":2,"SSId":"7","BSSId":"00:24:36:AC:BB:51","Channel":13,"RSSI":100,"LinkCount":3,"Downtime":"0T00:11:44"}}}
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

There is no console log - I also have syslog enabled and see no strange entries, the devices just drop out of Wi-Fi
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Reflash S20 sockets from 5.11.1 to 6.7.1 via FTDI. 
Plug them in
Wait

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expected the devices to stay as reliable as they once were

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-11-13:
Can not reproduce. All my S20 do work fine with v.6.7.1
Erased flash? Used decent USB-serial adapter?
See https://github.com/arendst/Tasmota/wiki/Prerequisites

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

rcarmo, 2019-11-13:
Hi there. Yes, I have decent hardware and know what I am doing. I did erase flash before upgrading because I saw that 5.14 was the last version that could be upgraded and I was using 5.11.

Chat came up blank. Going around doing `reset 5` to force recalibration and reconfiguring all the devices also didn't help (forgot to add that).

0n3-70uch, 2019-11-13:
Hi, cannot confirm. Using 7.0.0.4 [3c89ded](https://github.com/arendst/Tasmota/tree/3c89dedb92de224cbc3c1444f1092f770fed1fa4)
and Core 2.6.0. I have three S20 devices and everything fine.

Jason2866, 2019-11-13:
Try latest development version. It uses a different SDK and sets wifi TX power to 17 DBm.
http://thehackbox.org/tasmota/tasmota.bin.

Please erase flash with esptool.py and after this was successful flash tasmota with esptool.py
Dont restore a backup file. Can have weird settings

rcarmo, 2019-11-13:
I see that 7.1 is being worked on (with a commit mentioning preparation for release). Would that be within the next couple of days?

blakadder, 2019-11-13:
My S20 with an added sensor works even more reliably after an upgrade to the development binary.

You can either wait for 7.1 (next few days) or download a dev version from thehackbox which is equal to the future 7.1 release anyway

closed 2019-11-13 by blakadder