arendst/Tasmota #6921: Channel<1-3> resets Channel4 value to 0

Th0maz, 2019-11-13:
With my RGB+White light I can set every single channel by command "channel<1-4> value".
When the white channel4 is set to any value and I update RGB channels 1-3, channel4 is set back to zero.
I Could not work around this by SetOptions or anything.

This occurred on an AILight with Tasmota 6.6.0.

ascillato2, 2019-11-13:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-13 by ascillato2

s-hadinger, 2019-11-13:
Use `SetOption37 128`

Th0maz, 2019-11-13:
> Use `SetOption37 128`

Thanks, that solved my problem. Thought I tried that before, but obviously I did not.