arendst/Tasmota #6922: New Device: revolt smart home thermostat with TYWE3S

sven5, 2019-11-13:
Hi,

Just to inform you about a possibly new device.
I bought a smart home WIFI controllable thermostat from pearl in Germany. 
It's called [revolt WLAN-Thermostat](https://www.pearl.de/a-NX4608-3103.shtml)
It connects to a cloud system called [ELESION](https://www.elesion.com/)
Sadly, they do not offer any SDK or source code to create own projects with this device. So one is tied to the ELESION cloud. 

However, inside the device works a TYWE3S controller. 
I wonder if it would be possible to flash a new firmware on this device.

[image]

Jason2866, 2019-11-13:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-13 by ascillato2

hagenbuch, 2022-12-14:
I would like to use this, too! No one tried it so far?

sfromis, 2022-12-14:
Device looks a lot like everything is controlled by "the other" MCU, with the TYWE3S being incidental, and most likely merely used to provide Wifi (board even says RF-WIFI-TYWE3S). Not a "typical" case for Tasmota, to make it work you'd have to "hack" the communication to the main MCU. 

It might be TuyaMCU protocol, in which case Tasmota has some support, but my guess would be not. Meaning that hacking it looks like hard work :grimacing:

hagenbuch, 2022-12-14:
Ah Ok thanks, you're right. I overlooked the other CPU :)