arendst/Tasmota #6923: Consolidate Hardware & Software Serial Commands

meingraham, 2019-11-13:
**Have you looked for this feature in other issues and in the wiki?**  
N/A

**Is your feature request related to a problem? Please describe.**  
N/A

**Describe the solution you'd like**  
Now that TasmotaSerial has been enhanced to make software serial very reliable, is there an opportunity to consolidate the hardware serial and software serial driver into a single driver which is "GPIO selection" aware. In essence, if the serial bridge is configured on IO1/IO3 or IO13/IO15 it uses hardware serial. Otherwise it uses software serial. In this way, the user can use a single set of commands and does not have to know whether to use the hardware set or the software set. In addition to UX improvement, this could potentially reduce program size by consolidating the drivers. The `SSerialSend` command could be retained for backward compatibility, but it would just be an alias to the single `SerialSend` command.

Additionally, would it be possible to add a command or SetOption to "configure" the serial communications mode rather than having to have a System#Boot rule to set this up each time the device restarts.

**Describe alternatives you've considered**  
Leave it as is.

**Additional context**  
Discussed with @s-hadinger on Discord.

ascillato2, 2019-11-13:
Consolidating the commands should be possible, but consolidating the drivers is not possible. Software Serial and hardware serial requires different code. The hardware serial uses the internal electronics of the chip for all the protocol handling, leaving the CPU free for other Tasks. In the Software serial, everything is managed by the CPU.

meingraham, 2019-11-13:
Yes, I understand that there is different code for software vs. hardware, but "packaged" so the logic figures out which "path" to execute based on the selected GPIO rather than the user having to select one command vs. another.

blakadder, 2019-11-13:
There's been many cases where people forgot its SSerialSend (myself included) and its not immediately obvious

teixeluis, 2019-11-28:
Is there currently a way to configure more than just the baud rate for the hardware serial port, i.e. the number of bits, the parity and the stop bits?

I'm working on a slave MCU that talks 9600 8E1

Thanks

Cheers,

Luis Teixeira

ascillato2, 2019-11-28:
@teixeluis 

please check:

https://github.com/arendst/Tasmota/wiki/TasmotaSlave

_(I'm hiding your comment due to OFF TOPIC, if you have more questions, please address them to the Tasmota Support Chat. Thanks)_

teixeluis, 2019-11-28:
Thanks Adrian,

I've raised the topic in the chat meanwhile.In my case it is not a Tasmota
slave device, but a different type of chip which receives commands via the
serial port.
It's serial port settings are slightly unusual - 8 data bits, even parity,
1 stop bit.

Thanks for your suggestion but it doesn't quite fit in this scenario.

Cheers,

Luis Teixeira

arendst, 2019-12-08:
Agree. I have to make sure there is no use case for both.

On the list.

meingraham, 2020-01-02:
How to make feature requests under review not go stale nor auto-closed?

ascillato, 2020-01-02:
It goes STALE if there is no activity. Now the bot won't do anything until more 25 days.

meingraham, 2020-01-27:
Still a desired enhancement.

closed 2020-02-27 by stale[bot]