arendst/Tasmota #6924: Zigbee IEEE address reset to "0000000000000000" after reboot

ma261065, 2019-11-14:
### BUG DESCRIPTION
Zigbee IEEE address reset to "0000000000000000" after reboot

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU v3
- [x] Tasmota binary firmware version number used: 7.0.0.4
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VS Code/PIO
- [ ] Flashing tools used: VSCode/PIO
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
11:38:55 CMD: Backlog Template; Module; GPIO
11:38:56 MQT: stat/miketasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:38:56 MQT: stat/miketasmota/RESULT = {"Module":{"18":"Generic"}}
11:38:56 MQT: stat/miketasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"166":"Zigbee Rx"},"GPIO14":{"0":"None"},"GPIO15":{"165":"Zigbee Tx"},"GPIO16":{"0":"None"}}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
11:39:28 CMD: status 0
11:39:28 MQT: stat/miketasmota/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"miketasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:39:28 MQT: stat/miketasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:17:30","StartupUTC":"2019-11-14T00:21:58","Sleep":50,"CfgHolder":4617,"BootCount":15,"SaveCount":26,"SaveAddress":"FA000"}}
11:39:28 MQT: stat/miketasmota/STATUS2 = {"StatusFWR":{"Version":"7.0.0.4(tasmota)","BuildDateTime":"2019-11-10T23:18:32","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
11:39:28 MQT: stat/miketasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DODO-31BF",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
11:39:28 MQT: stat/miketasmota/STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":412,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","047683A1","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,23,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
11:39:28 MQT: stat/miketasmota/STATUS5 = {"StatusNET":{"Hostname":"miketasmota-5694","IPAddress":"192.168.1.14","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:3C:96:3E","Webserver":2,"WifiConfig":2}}
11:39:28 MQT: stat/miketasmota/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.eclipse.org","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3C963E","MqttUser":"mikealex","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:39:28 MQT: stat/miketasmota/STATUS7 = {"StatusTIM":{"UTC":"Thu Nov 14 00:39:28 2019","Local":"Thu Nov 14 11:39:28 2019","StartDST":"Sun Oct 06 03:00:00 2019","EndDST":"Sun Apr 07 03:00:00 2019","Timezone":99,"Sunrise":"17:57","Sunset":"03:11"}}
11:39:28 MQT: stat/miketasmota/STATUS10 = {"StatusSNS":{"Time":"2019-11-14T11:39:28"}}
11:39:28 MQT: stat/miketasmota/STATUS11 = {"StatusSTS":{"Time":"2019-11-14T11:39:28","Uptime":"0T00:17:30","UptimeSec":1050,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"DODO-31BF","BSSId":"D0:6F:82:43:31:C8","Channel":3,"RSSI":46,"LinkCount":1,"Downtime":"0T00:00:06"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
00:00:00 CFG: Loaded from flash at F4, Count 24
00:00:00 Project tasmota Tasmota Version 7.0.0.4(tasmota)-STAGE
00:00:00 WIF: Connecting to AP1 DODO-31BF in mode 11N as miketasmota-5694...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on miketasmota-5694 with IP address 192.168.1.14
11:17:49 MQT: Attempting connection...
11:17:49 MQT: Connected
11:17:49 MQT: tele/miketasmota/LWT = Online (retained)
11:17:49 MQT: cmnd/miketasmota/POWER = 
11:17:49 MQT: tele/miketasmota/INFO1 = {"Module":"Generic","Version":"7.0.0.4(tasmota)","FallbackTopic":"cmnd/DVES_3C963E_fb/","GroupTopic":"cmnd/tasmotas/"}
11:17:49 MQT: tele/miketasmota/INFO2 = {"WebServerMode":"Admin","Hostname":"miketasmota-5694","IPAddress":"192.168.1.14"}
11:17:50 MQT: tele/miketasmota/INFO3 = {"RestartReason":"Software/System restart"}
11:17:53 MQT: tele/miketasmota/RESULT = {"ZigbeeZNPSent":"410000"}
11:17:54 MQT: tele/miketasmota/RESULT = {"ZigbeeZNPReceived":"4180020200020603"}
11:17:54 MQT: tele/miketasmota/RESULT = {"ZigbeeState":{"Status":1,"Message":"CC2530 booted","RestartReason":"Watchdog","MajorRel":2,"MinorRel":6}}
11:17:55 ZIG: checking device configuration
11:17:55 MQT: tele/miketasmota/RESULT = {"ZigbeeZNPSent":"2108000F00"}
[... 220 lines cut ...]

### TO REPRODUCE
* Reset Zigbee with zigbeereset 1
* Allow new device to join with zigbeepermitjoin 1
* Device joins successfully
* Check status with zigbeestatus2
* Device has full info as below:
   11:19:37 MQT: stat/miketasmota/RESULT = {"ZigbeeStatus-99":[{"Device":"0x5406","IEEEAddr":"00158D0003F114DB","ModelId":"lumi.weather","Manufacturer":"LUMI"}]}

* Reboot ESP82xx by pressing reset button on the board. Power remain to CC2530 board
* Wait for device to send data
* Check status with zigbeestatus2 again
* Device has full info as below, however IEEE address is now 0000000000000000, and ModelID and Manufacturer are missing
11:32:44 MQT: stat/miketasmota/RESULT = {"ZigbeeStatus-99":[{"Device":"0x4343","IEEEAddr":"0000000000000000"}]}

### EXPECTED BEHAVIOUR
IEEE address, ModelID and Manufacturer should persist reboot. Note that I saw another variation that I cannot repro, but I also saw similar behaviour after leaving Z2T running for a few days - the IEEE address, ModelID and Manufacturer slowly dropped off over time. Some devices would have them and others not until after a few days none of the 3 connected devices had them.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ma261065, 2019-11-14:
I note that the ShortAddress of the sensor changes above. 0x4343 was the ID the device was previously joined as before the zigbeereset 1. So it looks like zigbeereset 1 doesn't clear the info for previously joined devices. Since posting the dump above two more previously joined devices are now also sending data, even though I never explicitly allowed them to re-join the network.

So we could be looking at a combination of issues here:
1) zigbeereset 1 does not appear to correctly perform a factory reset and reconfiguration of the CC2530 chip. I say this because previously joined sensors are apparently rejoining the network even though the Z2T is not in zigbeepermitjoin mode.
2) Devices at some point are losing their IEEE address and ModelID and Manufacturer info

s-hadinger, 2019-11-14:
Thanks for your feedback.

Currently Z2T does not persist device information like IEEEAddress, ModelID, ManufID. When you reboot Tasmota, it also reboots the CC2530, and it needs to rediscover IEEEAddress from the network. The only moments when a device sends its IEEEAddress is: when it first joins (pairing), when it rejoins after a long moment, or when you ask for it (`ZigbeeProbe`).

This behavior is unfortunate but it's the way Zigbee and CC2530 work. When the Zigbee device is AC powered (not on battery) you can probe it with `ZigbeeProbe`.

In the mid-term I plan to add persistence in Flash for IEEEAddress, clusters and endpoints. They will then survive a reboot.

`ZigbeeReset` does a factory reset of the CC2530 and flushes its internal memory. I'm also surprised that devices automatically reconnects, but I think this is due to the fact the the PanID and Encryption keys are unchanged. Since the CC2530 does not remember the device, it stills accepts it and assigns it a new short address. I'm not sure it's a strong issue nor I can change this behavior.

closed 2019-11-14 by ascillato2