arendst/Tasmota #6927: Sending data to InfluxDB

Derkades, 2019-11-14:
I'd like a feature to send data from sensors (e.g. Sonoff POW or DHT12) to InfluxDB.

**Why?**
It is more convenient than the setup most people have right now, which is `Tasmota -> MQTT -> NodeRED -> InfluxDB` 

**How?**
There is an existing ESP8266 library that can do this (~2KiB) https://github.com/tobiasschuerg/ESP8266_Influx_DB

I might create a PR myself if I can figure out how Tasmota works

Jason2866, 2019-11-15:
The only chance, getting considered to be included,  when a PR is submitted.

ascillato, 2019-11-15:
Hi,

Nice idea, but now, in the most used home automation softwares like Home Assistant and OpenHab, InfluxDB has become a very common Add On. So, setting up the InfluxDB integration on them is really fast and easy for long term data storage. 

In my case, I have `Tasmota -> MQTT -> HomeAssistant -> InfluxDB`

And as everything happening in HA goes to InfluxDB, also goes all the data coming from Tasmotas. So, it is easier for me to just configure InfluxDB on HA, rather than configuring it on every Tasmota.

Anyway, the label feature request was added in order to see if there is someone interested on working on this to cover your usecase. 

Thanks for sharing your ideas.

jziolkowski, 2019-11-17:
With the plethora of available options to send Tasmota telemetry to InfluxDB (I do it myself using mqtt_consumer in Telegraf), I agree that the code space for this feature could be used in a better way. 
Or, if a PR is provided, it would be able as an option for self-compilation.

All in all, if you use InfluxDB then you already have some software solutions along with it to feed/read the datapoints. Having to add a small configuration to bridge telemetry from MQTT to InfluxDB is IMO the better way.

ascillato2, 2019-11-17:
Closing this request as no one is working on this.

Anyway, thanks for sharing your ideas. Added a label for future reference.

closed 2019-11-17 by ascillato2

Makr91, 2020-04-11:
Hey guys, this may help, however it is in  Docker form:

http://nilhcem.com/iot/home-monitoring-with-mqtt-influxdb-grafana