arendst/Tasmota #6929: Description field in "Other parameters" config

moelski, 2019-11-15:
Hi, 
I use 12 Smart Socket within my house. If you connect to a device you only have the IP address to guess what the device is used for. 
So what do you think about adding a simple description field to tasmota? Just an input field in the "Other parameters" configuration where you can place a small device description like "Washing machine". 
This information could be part of the info topic in MQTT and displayed on the index page. 
Could be very handy in most cases. 

Example for MQTT Message: 
```
tele/sonoff/INFO1 
{"Module":"Gosund SP1  v23","Version":"6.7.1(sonoff)","FallbackTopic":"cmnd/IOT-Steckdose-01_fb/","GroupTopic":"sonoffs"**,"Description":"Washing machine"**}
```

blakadder, 2019-11-15:
or you can use:
- topic
- friendly name
- custom template name which becomes module name when applied

moelski, 2019-11-15:
That's possible. 
But a description field would be easier to handle. 
And you can change the description without creating new topic in MQTT.

Jason2866, 2019-11-15:
This additional info would cost extra flash space.

arendst, 2019-11-15:
As suggested before use the Friendly name field for your request.

moelski, 2019-11-15:
Thx for that. 
But Friendly name is not part of the MQTT massage, is'n it?

arendst, 2019-11-15:
It's part of ``status``

moelski, 2019-11-15:
Ok. But STATUS will not be send to the MQTT server automatically?

Jason2866, 2019-11-15:
A small rule will do the job

andrethomas2, 2019-11-15:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-15 by andrethomas2