arendst/Tasmota #6930: FR: Setoption to disable boot count for deep sleeping devices

andrethomas, 2019-11-15:
Would like a setoption to disable the boot count for special cases like deepsleep.

When teleperiod is 10 it does not appear to save the boot count to flash.

My observation is that when teleperiod > 14 the device will output the following on the seriallog

```
11:04:52 APP: Boot Count 87
11:04:52 CFG: Saved to flash at F5, Count 137, Bytes 4096
```

Although this is useful for diagnostic purposes it would be nice to conditionally disable this specific write if deepsleep is enabled by either a setoption or compiler directive.

This is to avoid an unwanted number of flash cycles when deep sleep is being used.

**Have you looked for this feature in other issues and in the wiki?**  

**Is your feature request related to a problem? Please describe.**  

Unwanted flash cycles.

**Describe the solution you'd like**  

Be able to set this using a setoption.

**Describe alternatives you've considered**  

Using teleperiod of < 11 but then the device does not always get NTP time.

arendst, 2019-11-15:
Did you try the latest deepsleep code? It postpones teleperiod unitl wifi is connected. Then enables the user configured teleperiod, which is at least 10 seconds. Then once MQTT is connected it set teleperiod to 5 seconds and reports teleperiod data after which it goes to sleep.

As bootcount is saved 10 seconds after restart it should be saved always UNLESS your wifi AND MQTT connection are settled within 4 seconds. Are they? Mine take at least 8 seconds although I always do wifi scan on restart which can take some seconds too. Considering this it may indeed be faster than 4 to 5 seconds skipping bootcount write.

(I tend to come to conclusions while I write. Not sure if that's a good thing though...)

Any way. I'll add a setoption for this. Default no bootcount. Bootcount only on setoption is 1.

andrethomas, 2019-11-15:
Here is what I get on teleperiod 10
```
00:00:00 CFG: Loaded from flash at F5, Count 249
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 7.0.0.4(tasmota)-2_6_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as tasmota-7632...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-7632 with IP address 192.168.42.13
12:57:50 NTP: Drift 0, (UTC) Fri Nov 15 10:57:50 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
12:57:50 QPC: Reset
12:57:51 RSL: tele/tasmota/STATE = {"Time":"2019-11-15T12:57:51","Uptime":"0T00:00:09","UptimeSec":9,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Wireless","BSSId":"E4:8D:8C:B9:6A:38","Channel":3,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
12:57:51 RSL: stat/tasmota/RESULT = {"DeepSleep":{"Time":"2019-11-15T12:58:44","Epoch":1573815524}}
12:57:51 RSL: tele/tasmota/LWT = Offline
```

And this is what I get on teleperiod 15

```
00:00:00 CFG: Loaded from flash at FB, Count 251
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 7.0.0.4(tasmota)-2_6_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as tasmota-7632...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-7632 with IP address 192.168.42.13
12:59:51 NTP: Drift 0, (UTC) Fri Nov 15 10:59:51 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
12:59:51 QPC: Reset
12:59:53 APP: Boot Count 200
12:59:53 CFG: Saved to flash at FA, Count 252, Bytes 4096
12:59:57 RSL: tele/tasmota/STATE = {"Time":"2019-11-15T12:59:57","Uptime":"0T00:00:14","UptimeSec":14,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Wireless","BSSId":"E4:8D:8C:B9:6A:38","Channel":3,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
12:59:57 RSL: stat/tasmota/RESULT = {"DeepSleep":{"Time":"2019-11-15T13:00:44","Epoch":1573815644}}
12:59:57 RSL: tele/tasmota/LWT = Offline
```

I don't think the sleep code was changed after my pull:

```
commit 7b10bec8fe63bcfa464bd221c5eeb4fae0d03a98 (HEAD -> development, origin/development, origin/HEAD)
Author: Theo Arends <11044339+arendst@users.noreply.github.com>
Date:   Thu Nov 14 15:57:52 2019 +0100
```

andrethomas, 2019-11-15:
And yes, the 42.13 in the ip address is intentional :)

arendst, 2019-11-15:
That's strange. It seems indeed 10 seconds of uptime will send your teleperiod message. That shouldn't happen with my latest changes where teleperiod is set to 9999 after restart and once wifi is connected it is set to 10 in your case. It then should be initiated at 15 seconds uptime as expected.

Investigating

andrethomas, 2019-11-15:
That was before your last commit so let me try to reproduce it now with the SO76

arendst, 2019-11-15:
Bugger.

I guess the change slipped true while making other changes. Pls try the one from today which does contain the correct changes.

andrethomas, 2019-11-15:
I think the issue was on my side... perhaps flashed a stale bin which was solved by removing all the platform and package folders to force migration to 2.6.1

In my pastes above it always ended with `RSL: tele/tasmota/LWT = Offline` but now it ends with `13:34:27 APP: Sleeping`

arendst, 2019-11-15:
De-bugger. The fix was in indeed. Rest my mind.

So now it should work as expected independant of minimal teleperiod as it will always take more than 10 seconds before the initial teleperiod kicks in.

andrethomas, 2019-11-15:
So now with SO76 enabled and teleperiod 15 it still writes to flash

```
00:00:00 CFG: Loaded from flash at F6, Count 277
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 7.0.0.4(tasmota)-2_6_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as tasmota-7632...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-7632 with IP address 192.168.42.13
13:41:07 NTP: Drift 0, (UTC) Fri Nov 15 11:41:07 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
13:41:07 QPC: Reset
13:41:09 APP: Boot Count 223
13:41:09 CFG: Saved to flash at F5, Count 278, Bytes 4096
13:41:21 RSL: tele/tasmota/STATE = {"Time":"2019-11-15T13:41:21","Uptime":"0T00:00:22","UptimeSec":22,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Wireless","BSSId":"E4:8D:8C:B9:6A:38","Channel":3,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
13:41:21 RSL: stat/tasmota/RESULT = {"DeepSleep":{"Time":"2019-11-15T13:42:00","Epoch":1573818120}}
13:41:24 APP: Sleeping
```

But with SO76 disabled it works as expected not writing to flash

```
00:00:00 CFG: Loaded from flash at FB, Count 280
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 7.0.0.4(tasmota)-2_6_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as tasmota-7632...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-7632 with IP address 192.168.42.13
13:44:07 NTP: Drift 0, (UTC) Fri Nov 15 11:44:07 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
13:44:07 QPC: Reset
13:44:21 RSL: tele/tasmota/STATE = {"Time":"2019-11-15T13:44:21","Uptime":"0T00:00:22","UptimeSec":22,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Wireless","BSSId":"E4:8D:8C:B9:6A:38","Channel":3,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
13:44:21 RSL: stat/tasmota/RESULT = {"DeepSleep":{"Time":"2019-11-15T13:45:00","Epoch":1573818300}}
13:44:24 APP: Sleeping
```

andrethomas, 2019-11-15:
Same with teleperiod 10 - Only SO76 enabled results in flash write for boot count :)

I think that solves this one since it is disabled by default.

Thanks!

closed 2019-11-15 by andrethomas

arendst, 2019-11-15:
Correct.