arendst/Tasmota #6931: Slider on  SHUTTER   does not reflect/refresh current position when moving from buttons or web TOGGLE switches

imtragsl, 2019-11-15:
Currently using 7.0.0.4

**Have you looked for this feature in other issues and in the wiki?**  
  Yes
**Is your feature request related to a problem? Please describe.**  
  No,  just a feature

**Describe the solution you'd like**  
 I wouldl like to see the slider move, (reflect current position)  when  shutther is activated via external buttons or by Tasmota web Interface  Toggle controls.

**(Please, remember to close the issue when the problem has been addressed)**

imtragsl, 2019-11-15:
I've realized that pressinfg F5  position is reflected, ....

Can an autorefresh be sent  (similar to the one that reflects AM2301 values ?

arendst, 2019-11-15:
Duplicate. See other issues. In short, No.

imtragsl, 2019-11-15:
Sorry  & Thanks..

closed 2019-11-15 by imtragsl