arendst/Tasmota #6932: Arduino IDE 1.8.10 compile erros

mike2nl, 2019-11-15:
Error message:
sonoff_sc_182:1669:15: error: expected unqualified-id before string constant
 static void BacklogLoop()
               ^

- Used Cores are 2.6.0 and 2.6.1 and IDE 1.8.10
- Used Tasmota version are 7.0.0.2 and 7.0.0.4 from today

Setup 1 - NO errors:
IDE 1.8.10, core 2.6.0 and Tasmota 7.0.0.2
NO compile errors and works 100% on hardware

Setup 2 - NO errors:
IDE 1.8.10, core 2.6.1 and Tasmota 7.0.0.2
NO compile errors and works 100% on hardware

Setup 3 - errors:
IDE 1.8.10, core 2.6.0 and Tasmota 7.0.0.4 from today
Error message see top of the issue

Setup 4 - errors:
IDE 1.8.10, core 2.6.1 and Tasmota 7.0.0.4 from today
Error message see top of the issue

There must be changes in 7.0.0.4 which gives IDE 1.8.10 a hard time. At the end @Jason2866 (Thanks) told me that @arendst has possible dropped the IDE support. That sounds not really good because i know a lot of users on discord which use IDE. 

But we have still the information on the github page that we support IDE and all the Wiki information around IDE. For the future i'm willing to test new versions for Tasmota with the IDE setup or we must delete all information about IDE on the github main page and the Wiki.

arendst, 2019-11-15:
Let's first decide that IDE is still supported.

Then considering your error message there is no change around Backlogloop() between 7.0.0.2 and 7.0.0.4 so I suggest you start debugging using your IDE and find out what's wrong.

mike2nl, 2019-11-15:
@arendst 
have you tested 7.0.0.4 in IDE, or only in VSC?
BTW: @Jason2866 had the same error after updating his IDE. He told me that on discord

I can do more debugging but much deeper i can't go. I had setup the lost deep debug info.
i can upload all the output for you if you wish so.

arendst, 2019-11-15:
I did not test with IDE. In fact I stopped using the IDE a long time ago.

I just did a test and see it fails misserably with very descriptive error messages where a normal person cannot make any sense out of it. Perhaps you can ;-)

mike2nl, 2019-11-15:
@arendst 
Thanks for the feedback!
Then i get a question on my inner eye:
- that means we no longer support Arduino IDE then?

Oops, a lot of work is then waiting in the wiki.

arendst, 2019-11-15:
How about you fix this IDE compilation. Can't be difficult.

mike2nl, 2019-11-15:
It's an idea and i will look into it. Found already that esp8266/arduino has changed a lot of libraries. I saw this in the afternoon while i was testing. So we can't not really trust that side in this core. I mean the libraries which came with core 2.6.1. Jesus

arendst, 2019-11-15:
It also fails on IDE 1.8.9 and core 2.5.2 so it's not core related.

Must be something with source file naming as the IDE is very picky with sequencing filenames...

mike2nl, 2019-11-16:
stopped debugging

closed 2019-11-16 by mike2nl

bovo-fabiano, 2019-11-16:
Same problem here.