arendst/Tasmota #6935: ADC1115 added automatically, but non connected

qualitywins, 2019-11-15:
Hi,

I have a WemosD1 with 2 Devices connected a BME280 and a TSL2561. However I can also see data for a ADS1115 being submitted, though I have non connected.

20:07:49 CMD: status 8
20:07:49 MQT: stat/topic/STATUS8 = {"StatusSNS":{"Time":"2019-11-15T20:07:49","Switch1":"ON","BME280":{"Temperature":22.0,"Humidity":68.5,"Pressure":994.1},"ADS1115":{"A0":-8446,"A1":-8446,"A2":-8446,"A3":-8446},"TSL2561":{"Illuminance":28.650},"PressureUnit":"hPa","TempUnit":"C"}}

I use

Program Version | 6.7.1(sensors)
2019-10-26T13:21:02
STAGE/2.2.2-dev(38a443e)

Kind regards,

Nikolas

ascillato2, 2019-11-15:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Also, please try with latest version 7.0.0.5

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

qualitywins, 2019-11-16:
Solved in 7.0.0.4

ADC1115 can be disabled with I2CDriver13 0. 

Thanks for the advise.

closed 2019-11-16 by qualitywins