arendst/Tasmota #6937: Support rule with rfbridge with portisch with rfraw 177 data.

niekvlessert, 2019-11-15:
**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  
Well, what I would like to know is if can create a rule that publish things on mosquitto running on my pi if my rfbridge with portisch receives  {"RfRaw":{"Data":"AA B1 03 042E 017C 16DA 281818180909181809091818090918181818 55"}}. The data that I would like to match to is this part: 281818180909181809091818090918181818, it's always the same when pressing the bell. This data only comes in when activating bucket sniffing with rfraw 177, because the doorbell is not supported by Portisch. I could try to add it (if I understand how this works, the documentation is not the greatest), but if I could match on lines like this I could match on any rfdata coming in, even when it's not support by Portisch.

nagyrobi, 2019-12-09:
check this out:
https://community.home-assistant.io/t/sonoff-rf-bridge-tasmota-with-portisch-firmware-rf-doorbell-sort-of-working/138571/6

The idea is that with rfraw 177 you sniff all the time and you get all the received data continuously in MQTT. Then, you do the processing on your PI to match the proper code and perform actions on it, not on Tasmota.

niekvlessert, 2019-12-09:
@nagyrobi Well, that is indeed precisely what I suggested. I don't know a lot about home-assistant but it seems to me the suggested config is for that. My question is how to get this rfraw 177 data onto the configured mqtt server, in this thread it seems they have managed that: 'With rfraw 177 i got it to work, but the outcome was different so automation was complex (because i put al the different outcome in it).'

nagyrobi, 2019-12-10:
I don't know anything about your setup. But in my case, if I set "rfraw 177" in my Sonoff Bridge Module, whenever I press a button on my RF remote, I get something like this in my MQTT server, under the **tele/rf-bridge-1/RESULT** topic: `{"Time":"2019-12-10T09:37:46","RfRaw":{"Data":"AA B1 04 01CC 07BC 03D4 0776 3818182828181818181918 55"}}`
So rfraw 177 not only displays the received messages in the console, but also publishes them all to MQTT.

niekvlessert, 2019-12-10:
Wow, I didn't even consider that and up until this point I've been only sending data to the rf bridge, I didn't even know which topic to subscribe to for data coming in. But you're absolutely right, the events already come in, great. Thanks!

closed 2019-12-10 by niekvlessert