arendst/Tasmota #6940: Provide support for half-duplex Modbus ICs like MAX3485

mclane, 2019-11-15:
Hello,
I have designed Modbus to MQTT interface which uses  an ESP8266 module linked directly to a MAX3485 interface chip. 
[Modbus2Wifi.pdf](https://github.com/arendst/Tasmota/files/3853462/Modbus2Wifi.pdf)
This chip needs apart form the normal TX /RX pins an additional R/W pin (in my case connected to GPIO4). So before a transmission on the serial line  the GPIO needs to be switched HIGH and after back to LOW. So far, I have implemented the firmware using https://github.com/homieiot/homie-esp8266. together with the ModbusMaster library. However, I would like to switch to Tasmota since all my other ESP8266 devices use this nice framework now. I would like to use the smart meter interface https://github.com/arendst/Tasmota/wiki/Smart-Meter-Interface. Would it be possible to integrate this feature? Maybe someone can point me to the place where the modbus communication is implemented so that I can add this little feature myself. Thanks so far for the huge work!

ascillato, 2019-11-15:
Smart meters and modbus are already implemented.

ascillato2, 2019-11-15:
If you need assistance on how to use Tasmota, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-15 by ascillato2