arendst/Tasmota #6941: Update sensor value if change is greater than...

lindehoff, 2019-11-16:
I have tried to look for this feature but I can't find it anywhere.

If iÍ understand correctly sensors are updated with a time interval and that can be changed with `TelePeriod`. This is fine for most sensors like temperature, but I have a light sensor, TSL2561, and want to be updated immediately if the change is greater the a specific value. I think this could be very nice for other sensors as well, like humidity. Let's say you take a shower and want to trigger a fan to turn on if the humidity quickly rises. I know that you can lower the TelePeriod to like 10 sec but I think it would be much better if you just could define a threshold value when you want to trigger an update.

I have tried to look at the Rules functionality but can't figure out how to configure it for this scenario.

There might already be a feature for this but I can't seam to find it.

Jason2866, 2019-11-16:
It is already possible. Check wiki rules.
A example for from rules cookbook. It reacts immediately 
https://github.com/arendst/Tasmota/wiki/Rule-Cookbook#prevent-wemos-d1-mini-load-overcurrent

blakadder, 2019-11-16:
[image]
This is the relevant part for this question

Closing since answered

closed 2019-11-16 by blakadder

lalo-uy, 2019-11-18:
None of this option watch for delta ,
DS18B20#Tempreature react with a tiny change, 0.064 if 12bits.
DS18B20#Tempreature > 20 is absolute value.
It will be possible to assign DS18B20#Tempreature to a variable oldTemp,
and do
abs( DS18B20#Tempreature - oldTemp ) > 2

ascillato, 2019-11-18:
Please, check the delta example of the wiki at rules cookbook. Thanks.

Alxspb, 2021-03-19:
> Please, check the delta example of the wiki at rules cookbook. Thanks.
Delta example is about trigger when current value is in delta (ts is asking for change delta). Its quite easy and not related to initial question.
Why this issue is closed if it's still not answered?

Jason2866, 2021-03-19:
Since github is for issues (= bugs) in the software.
The topic is a question and it is answered how it can be done (rules).
Case closed

ascillato2, 2021-03-19:
Please, check the delta example of the wiki at rules cookbook. Thanks.