arendst/Tasmota #6947: MQTT publish issue (Wemos D1 mini)

HaViGit, 2019-11-16:
After flashing Tasmota on a Wemos D1 Mini with two relays connected all looks okay and I can toggle both relays from the web interface. Within Tasmota I have activated MQTT which publish in the topic Sonoff213/stat/POWER1 and  Sonoff213/stat/POWER2 for the two relays.

When toggling from the web interface the MQTT Explorer program shows:

– Sonoff213
  – stat
       RESULT = {POWER1”:”OFF”}
       POWER1 = OFF

So far so good but now I am struggling to toggle by publishing the toggle command via MQTT. 

Within MQTT Explorer (or any other program) I publish ON or OFF to the topic Sonoff213/stat/POWER1 but the switch does not toggle.

Someone more experienced and willing to help me with the right way to publish ON and OFF?

ascillato2, 2019-11-16:
How to use MQTT is explained in the wiki at: https://github.com/arendst/Tasmota/wiki/Commands

sending to stat is not correct. stat is status. it is read only
commands are sent to cmnd

topic Sonoff213/cmnd/POWER1
payload 1

the payload can be:
1 for ON
ON for ON
0 for OFF
OFF for OFF
2 for toggle
TOGGLE for toggle
and others.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-16 by ascillato2

HaViGit, 2019-11-16:
Will keep in mind. Anyway topic Sonoff213/**cmnd/**POWER1 works, thanks!