arendst/Tasmota #6949: Random reboots on efun plugs sh331w.

ecrode, 2019-11-17:
## ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
I've just bought a EFUN plug from amazon, and successfully flashed it OTA using tuya-convert. After that I've reset it using "Reset 5" and updated to the latest dev build (7.0.0.4(0962562-tasmota)) thanks to advice I got from discord while trying to troubleshoot this issue. The problem is that after couple of hours issues with random reboots have resumed. I've setup node red to log all of the reboot reasons from four units plugged into the same power strip. I would love to get to the bottom of this and determine if there is a hardware incompatibility in play that deters this unit operating with custom firmware. Right now random reboots make these units unusable for my needs.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): EFUN SH331W 
- [X] Tasmota binary firmware version number used: 7.0.0.4(0962562-tasmota)
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: _____
- [X] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
06:07:49 MQT: stat/firecam/RESULT = {"NAME":"Efun-Plug","GPIO":[56,0,158,0,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":18}
06:07:50 MQT: stat/firecam/RESULT = {"Module":{"0":"Efun-Plug"}}
06:07:50 MQT: stat/firecam/RESULT = {"GPIO":"Not supported"}

  ```
- [X] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
06:08:25 MQT: stat/firecam/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
06:08:26 MQT: stat/firecam/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
06:08:26 MQT: stat/firecam/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
  ```
- [X] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
06:09:01 MQT: stat/firecam/STATUS = {"Status":{"Module":0,"FriendlyName":["Fire Cam"],"Topic":"firecam","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:09:01 MQT: stat/firecam/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Power on","Uptime":"0T07:11:46","StartupUTC":"2019-11-16T21:57:15","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":36,"SaveAddress":"F4000"}}
06:09:01 MQT: stat/firecam/STATUS2 = {"StatusFWR":{"Version":"7.0.0.4(0962562-tasmota)","BuildDateTime":"2019-11-16T15:52:54","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(bb83b9b)","Hardware":"ESP8266EX"}}
06:09:01 MQT: stat/firecam/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Biot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
06:09:01 MQT: stat/firecam/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
06:09:01 MQT: stat/firecam/STATUS5 = {"StatusNET":{"Hostname":"swfirecam","IPAddress":"10.0.20.134","Gateway":"10.0.20.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.20.1","Mac":"84:F3:EB:72:4D:E5","Webserver":2,"WifiConfig":2}}
06:09:01 MQT: stat/firecam/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.20.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_724DE5","MqttUser":"iot","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
06:09:01 MQT: stat/firecam/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 17 05:09:01 2019","Local":"Sun Nov 17 06:09:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:02","Sunset":"17:08"}}
06:09:01 MQT: stat/firecam/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
06:09:01 MQT: stat/firecam/STATUS10 = {"StatusSNS":{"Time":"2019-11-17T06:09:01","ENERGY":{"TotalStartTime":"2019-11-16T17:33:54","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":128,"Current":0.000}}}
06:09:01 MQT: stat/firecam/STATUS11 = {"StatusSTS":{"Time":"2019-11-17T06:09:01","Uptime":"0T07:11:46","UptimeSec":25906,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Biot","BSSId":"BE:4E:26:D0:FF:52","Channel":6,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:07"}}}

  ```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here: (please disregard node red wrapper from MQTT logging)
{"time":"2019/11/16 21:57:24Z","level":"INFO","var":"msg","message":{"topic":"tele/firecam/INFO3","payload":{"RestartReason":"Power on"},"qos":0,"retain":false,"_msgid":"aeeeebd9.f12998"}}
{"time":"2019/11/16 22:09:29Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Blocked Loop"},"qos":0,"retain":false,"_msgid":"c291ac53.d04dc"}}
{"time":"2019/11/16 22:14:59Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Hardware Watchdog"},"qos":0,"retain":false,"_msgid":"d5458d24.22a08"}}
{"time":"2019/11/16 22:15:20Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Blocked Loop"},"qos":0,"retain":false,"_msgid":"bf1089e3.d54108"}}
{"time":"2019/11/16 22:22:53Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Blocked Loop"},"qos":0,"retain":false,"_msgid":"2830dd74.977e82"}}
{"time":"2019/11/16 22:23:07Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x40244cf6 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"},"qos":0,"retain":false,"_msgid":"ee75111e.1a5fc"}}
{"time":"2019/11/16 22:23:30Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Power on"},"qos":0,"retain":false,"_msgid":"49479351.9f972c"}}
{"time":"2019/11/16 22:27:20Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Hardware Watchdog"},"qos":0,"retain":false,"_msgid":"e585f83b.31f7e8"}}
{"time":"2019/11/16 22:27:37Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Blocked Loop"},"qos":0,"retain":false,"_msgid":"6fbbcd53.108624"}}
{"time":"2019/11/16 22:52:47Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Blocked Loop"},"qos":0,"retain":false,"_msgid":"aa9bd4fa.808ba8"}}
{"time":"2019/11/16 22:58:11Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Software/System restart"},"qos":0,"retain":false,"_msgid":"520bf7dd.d58378"}}
{"time":"2019/11/16 23:06:41Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Software/System restart"},"qos":0,"retain":false,"_msgid":"f2566e8c.ebd38"}}
{"time":"2019/11/16 23:10:26Z","level":"INFO","var":"msg","message":{"topic":"tele/officedisp/INFO3","payload":{"RestartReason":"Hardware Watchdog"},"qos":0,"retain":false,"_msgid":"b2df79f0.8c5238"}}
[... cut ...]
{"time":"2019/11/17 02:43:16Z","level":"INFO","var":"msg","message":{"topic":"tele/officespk/INFO3","payload":{"RestartReason":"Software/System restart"},"qos":0,"retain":false,"_msgid":"92406b52.d9e138"}}
{"time":"2019/11/17 03:32:32Z","level":"INFO","var":"msg","message":{"topic":"tele/modem/INFO3","payload":{"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4022653f epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000044 depc:0x00000000"},"qos":0,"retain":false,"_msgid":"8be11771.730bd8"}}

  ```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-11-17:
Maybe the GPIO config is not correct. Do a `reset 3`, power cycle and leave it alone.
Does it still have Exceptions?

arendst, 2019-11-17:
Four devices in close vicinity like all in one power strip are bound to fail.

ecrode, 2019-11-17:
@Jason2866 got the template from [blackadder (here)](https://blakadder.github.io/templates/EFUN_SH331W.html), I've reset one of the devices and I'll report if it starts working better. Any reason why Reset 3 is better than 5? 
@arendst thank you, I'll sped them out for continued observation, just to clarify the power strip they are plugged into is a KMC 12 outlet workbench power strips so they were not as close as they could have been. Also the reason I've started monitoring more units and logging reset reasons was that I encountered these problems with the first unit I've tried to use. It behaved in a similar matter with the others still in the box.

arendst, 2019-11-17:
Thx. Problem is you have to many **different** issues like watchdogs, blocked loops and exceptions. 

Most of the times these are not tasmota issues but more a tight power supply issue caused by extra current needed for wifi calibration or wifi transmissions.

arendst, 2019-11-17:
BTW Reset three erases the SDK wifi parameter area AND starting from the latest dev versions als erases any flash above this area too.

Reset five only erases Tasmota settings area which is situated below the SDK wifi parameter area.

ecrode, 2019-11-17:
Thanks Theo, because I was not seeing a flood of posts about similar issues to mine, I figured that the problems are likely hardware related, can't expect a big power budget on a $8/part delivered.
By the way, I have the four units now plugged into separate wall outlets with at least 10ft between them, since then I had one unit reboot due to "Fatal exception:9 flag:2 (EXCEPTION) epc1:0x40221f01 epc2:0x00000000 epc3:0x00000000 excvaddr:0x000013a7 depc:0x00000000", it currently has RSSI of (68%).

ascillato, 2019-11-17:
A smaller version of Tasmota is Tasmota_basic.bin

ecrode, 2019-11-17:
Update on the unit running tasmota_basic, it just rebooted due to "Hardware Watchdog".
I've ordered four more switches from a different manufacturer (BN-LINK WiFi), will report if those are any better (fingers crossed).

ascillato2, 2019-11-17:
> Update on the unit running tasmota_basic, it just rebooted due to "Hardware Watchdog".

Closing this issue as it is hardware related. Sorry.

closed 2019-11-17 by ascillato2

ecrode, 2019-11-21:
Update: efun plugs are running much better on firmware 7.0.0.5 whereas there were many errors on 7.0.0.4 and 7.0.0.3. 
BNC-60/U133TJ are working much better than efun plugs, but it is unclear if this is due to difference in hardware or firmware. 
Out of everything running 7.0.0.5, it was the BNC-60/U133TJ module that restarted due to:
Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x400043f0 excvaddr:0x00000000 depc:0x00000000
So far this is the only reason for failure for either efun or bn-link plugs running on 7.0.0.5 during the last 24 hours. I'm testing 8 plugs (four of each) with at least 10 ft between any two of them.

arendst, 2019-11-21:
You might want to try to change the SDK version used during compiling. If so pls change in platformio.ini the following lines:
```
; NONOSDK221
;                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK221
; NONOSDK22x_190313
;                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190313
; NONOSDK22x_190703 = 2.2.2-dev(38a443e)
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703
; NONOSDK22x_191024 = 2.2.2-dev(5ab15d1)
;                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191024
; NONOSDK22x_191105 = 2.2.2-dev(bb83b9b) (Tasmota default)
;                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191105
```
This will disable SDK22x_191105 and enable SDK22x_190703

ptr727, 2020-09-08:
FYI, SH331W running ESPHome also occasionally reboots, no other WiFi devices nearby.