arendst/Tasmota #6950: Default state on startup for lights (not previous state)

jhndrx, 2019-11-17:
**Have you looked for this feature in other issues and in the wiki?**  
Yes, the answers didn't fully solve this request.

**Is your feature request related to a problem? Please describe.**  
New feature. Startup using default RGBW values stored in non-volatile memory.

**Describe the solution you'd like**  
A SetOption that sets a RGBW value for the device on initial startup (i.e. flipping the physical light switch from off to on). This would be used in tandem with "PowerOnState=1" to turn relays on after power up. 

This feature should not impede on "PowerOnState=3" to return relays to their last saved state. The default value should be stored separately from the stored previous state.

**Describe alternatives you've considered**  
The current available settings can return the light values to the last state it was in when power is removed via a physical switch. Not ideal if the light was in a non-preferred state.

**Additional context**  
Kids, visitors, or non-technical people may not know how or may not have access to modify a smart bulb. If the bulb is in a non-preferred state (such as dim color), it would be useful to be able to flip the physical switch off then on and return it to a "default" state.

s-hadinger, 2019-11-17:
You can use rule to do it:

```
PowerOnState 0
Rule1 On Power1#Boot Do Color 00FF000000 Endon
Rule1 1
``` 

(Of course change it with the desired color)

With this rule it is better to have `PowerOnState 0` otherwise you will have a short flash with the old color and the new color appearing. With `PowerOnState 0` Tasmota starts with the light off, and the `Color` command turns it on with the desired color without any flash.

jhndrx, 2019-11-17:
I missed rules in my research. Thanks.

closed 2019-11-17 by jhndrx