arendst/Tasmota #6954: Please help with my LSC Gu10 RBB Bulb

nozamo, 2019-11-17:
Hello im pretty new here on the form, and also sorry for my english but ill do my best.
I recently bought a gu10 bulb from the action (dollar store in the Netherlands).
i flashed it successfully with tasmota (using tuya converter) but i could not get the light working does anyone have a configuration file/settings or something else to make this bulb work with tasmota.
thanks already
Nemo
the Netherlands :)

Jason2866, 2019-11-17:
Should be this https://blakadder.github.io/templates/lsc_smart_connect_multicolor_spot.html

nozamo, 2019-11-17:
> Should be this https://blakadder.github.io/templates/lsc_smart_connect_multicolor_spot.html

Thanks for your help :)
now i can go to work with implenmenting this stuff

closed 2019-11-17 by nozamo